facebookresearch/habitat-sim #181: Saving RGB frames from habitat-sim

BurakHocaoglu, 2019-09-01:
Hello,

I have several questions on generating a synthetic image dataset with RGB sensors of habitat-sim and the scene datasets like Replica as 3D background. I can spawn a custom 3D mesh into a scene dynamically thanks to the bullet physics integration. I can set a random transformation for the last imported mesh for attitude variance. What I'd like to know:

1) Can I save RGB images on demand e.g. Viewer's key press event?

2) It seems semantic objects belong to the scene data only and do not consider the imported meshes when trying to get their bounding boxes. Therefore; do I have to compute the camera transformations to generate 2D positions of bounding boxes? Also, when 3D rotations are involved, I believe I need to rotate the original bounding box (with no rotations at all, e.g. about the gaze axis) with respect to the rotation that I will give with ```setTransformation``` call and compute new corners for the bounding box. Then, project that bounding box onto 2D image plane, which will change the bounding box again. Is there an easier way to achieve this or else?

Hope I am clear.

Kind regards,

Burak

aclegg3, 2019-09-03:
Hey Burak,

Happy to hear the object instancing is going well.

Regarding your image saving question:

- We provide an example of saving frames to disk with PIL in python which can be found in "save_color_observation" and like functions in examples/demo_runner.py.

- To save the current framebuffer in C++ you can use the [Magnum::DebugTools::screenshot ](https://doc.magnum.graphics/magnum/namespaceMagnum_1_1DebugTools.html#a66cd81ac668af7f8dffcd27ee47b2019)function. I've added an example of this to Viewer.cpp in master branch with the "I" key.

Regarding instanced object semantics: this feature is not yet complete. Presently, as you say, only semantics of the scene object are considered. We do plan to support this use case and I'll look into it in the near future. If you have a work-around I suggest using it until then. Is it the case that you are loading an object with a semantic texture or are you hoping to set the semantic value of the object programmatically?

BurakHocaoglu, 2019-09-04:
Something similar, but without any semantics. My plan is that I have a few meshes (with textures that have small diffuse/specular coefficients or no contribution at all, so they have a dark look) that I can convert into glTF format with the help of assimp package and import them into a scene dynamically, give them random transformations based on the position of the agent in the scene/world. Then, I will use the camera matrix in order to get the pixel coordinates of the bounding boxes of objects that I have imported earlier. I have the scene/world coordinates of imported objects (I assume), then I will use the camera matrix to project those objects onto the image and while doing that I hope to compute their bounding boxes as well.

So, according to issue #80 , I can get the intrinsic parameters of the camera (talking with respect to the video here https://www.youtube.com/watch?v=fVJeJMWZcq8&t=700s about image plane geometry), the camera matrix is the product of the intrinsic parameters matrix with the random transformation I give with ```setTransformation()``` call. This will be my workaround I suppose.

BurakHocaoglu, 2019-11-07:
Hello @aclegg3 ,

I have a question. I am trying to apply the following transformation scheme to the 3D bounding boxes in habitat, to project them into the screenshot I am going to take (example transformation, bb is of type ```Magnum::Range3D```):
```
bb = existingObjects_[physObjectID]->getCumulativeBB()
_trans = existingObjects_[physObjectID]->MagnumObject::transformationMatrix()

vertex = bb.backBottomLeft()
cam_matrix = renderCamera->getCameraMatrix()
projection_matrix = renderCamera->getProjectionMatrix()
projected_vertex = viewport_matrix * projection_matrix * cam_matrix * _trans * vertex
```
I apply the above scheme to all 8 vertices of bb in ```Magnum::Range3D```; where viewport matrix (the image is 800 x 600 by default) is the following:
```
400   0       0       400
0     300     0       300
0     0       0.5     0.5
0     0       0       1
```

However, I could not achieve a reasonable bounding box in 2D (not centered correctly, not in correct width and height, got negative values for x and y components of the projected center). Something is off and I can't see that. Can you help?

Kind regards,

Burak

aclegg3, 2022-03-01:
Follow-up here seems like a separate issue, please open a new issue if still relevant.

*NOTE*: python utilities for processing visual sensor observations to generate images and video can be found in the [viz_utils](https://aihabitat.org/docs/habitat-sim/habitat_sim.utils.viz_utils.html) module.

closed 2022-03-01 by aclegg3