facebookresearch/habitat-sim #182: Runtime Segmentation fault (core dumped)

SenZHANG-GitHub, 2019-09-02:
## Steps to reproduce

1. CentOS release 6.9 (Linux version 2.6.32-696.16.1.el6.x86_64 (gcc version 4.4.7 20120313 (Red Hat 4.4.7-18) (GCC) )
2. Complier: gcc 7.2.0
3. Compilation status: success 
4. Python version: 3.6.5 

## Observed Results
"python src/habitat-sim/examples/example.py --scene ~/data/habitat/scene_datasets/habitat-test-scenes/skokloster-castle.glb" leads to:

## Expected Results

sim_cfg.physics_config_file = ./data/default.phys_scene_config.json
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
Segmentation fault (core dumped)

## Possible Reason

1. Incorrect EGL version? I locally installed the following two packages by "rpm2cpio XX | cpio -idv" : mesa-libEGL-11.0.7-4.el6.x86_64.rpm  & mesa-libEGL-devel-11.0.7-4.el6.x86_64.rpm. Then I directly replace the header files by EGL 1.5 to include necessary DEFINEs as suggested in https://github.com/facebookresearch/habitat-sim/issues/145#issuecomment-521318477

Is the version of EGL .so file enough for habitat? (v11.0.7 for centos 6)

2. Incorrect CUDA or GPU version? I loaded cuda 7.5 with Tesla K40 but still not working

3. Other missing or incorrect dependencies?

-

erikwijmans, 2019-09-02:
CUDA 9.2 and 10.0 on a V100 should work.  In order for EGL to work correctly, habitat-sim needs to pick up Nvidia's version of EGL at runtime.  If it doesn't by default, it is likely that your system has a non-standard Nvidia driver install and you need to add the path to Nvidia's libEGL to `LD_LIBRARY_PATH`.

SenZHANG-GitHub, 2019-09-03:
> CUDA 9.2 and 10.0 on a V100 should work. In order for EGL to work correctly, habitat-sim needs to pick up Nvidia's version of EGL at runtime. If it doesn't by default, it is likely that your system [...]

Hit the point! Thanks. Actually I noticed this answer https://github.com/facebookresearch/habitat-sim/issues/18#issuecomment-483930596 which should also be the right answer here. But I misunderstood and thought libglvnd is one of the dependencies of EGL, trying to fix by updating EGL to latest mesa version and installing libglvnd. Now figure out that mesa-egl and libglvnd-egl are two distributions. "python example/example.py --scene ...." seems working now:

[image]

To those who might be interested, the final solution is straight-forward (on CentOS6 without root): 
(1) locally install (rpm2cpio) libglvnd-1.0.1-0.1.git5baa1e5.el6.x86_64.rpm, libglvnd-core-devel-1.0.1-0.1.git5baa1e5.el6.x86_64.rpm, libglvnd-devel-1.0.1-0.1.git5baa1e5.el6.x86_64.rpm, libglvnd-egl-1.0.1-0.1.git5baa1e5.el6.x86_64.rpm, libglvnd-gles-1.0.1-0.1.git5baa1e5.el6.x86_64.rpm, libglvnd-glx-1.0.1-0.1.git5baa1e5.el6.x86_64.rpm, libglvnd-opengl-1.0.1-0.1.git5baa1e5.el6.x86_64.rpm, mesa-libEGL-18.0.5-4.el7_6.x86_64.rpm, mesa-libEGL-devel-18.0.5-4.el7_6.x86_64.rpm, mesa-libGL-18.0.5-4.el7_6.x86_64.rpm, mesa-libGL-devel-18.0.5-4.el7_6.x86_64.rpm (install mesa version to get the header files for compilation)
(2) add "-DEGL_LIBRARY=/path/to/libglvnd/libEGL.so" and "-DEGL_INCLUDE_DIR=/path/to/include" to setup.py
(3) modify some system variables (gcc version should be >= 7.1.0
export CC=/path/to/gcc/7.2.0/bin/gcc
export CXX=/path/to/gcc/7.2.0/bin/g++

# WARNING: playing with LD_PRELOAD in ~/.bashrc could be super-dangerous (unexpected missing dependencies can crash everything and hard to fix w.o. root). DO export necessary LD_LIBRARY_PATH at the same time or leave some bashes running either on tmux or other terminals to remedy the system whenever needed.
export LD_PRELOAD=/path/to/gcc/7.2.0/lib64/libstdc++.so.6:/path/to/gcc/7.2.0/lib64/libgcc_s.so.1
export LD_PRELOAD=$LD_PRELOAD:/path/to/libglvnd/libEGL.so
export LD_PRELOAD=$LD_PRELOAD:/path/to/libglvnd/libGL.so

# lib64 below is the library path for your local installation
export LD_LIBRARY_PATH=/path/to/lib64:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=/path/to/lib64/libglvnd:$LD_LIBRARY_PATH
(4) "python setup.py install --headless"

closed 2019-09-03 by SenZHANG-GitHub

erikwijmans, 2019-09-03:
Just for future reference,

> (2) Add "-DEGL_LIBRARY=/path/to/libglvnd/libEGL.so" and "-DEGL_INCLUDE_DIR=/path/to/include" to setup.py

is likely unneeded as using whatever EGL gets picked up at compile time is totally fine, its just at run-time that the correct EGL needs to be picked up.

SenZHANG-GitHub, 2019-09-03:
> Just for future reference,
> 
> > (2) Add "-DEGL_LIBRARY=/path/to/libglvnd/libEGL.so" and "-DEGL_INCLUDE_DIR=/path/to/include" to setup.py
> 
> is likely unneeded as using whatever EGL gets picked up at compile time is totally fine, its just at run-time that the correct EGL needs to be picked up.

Yes the version of libEGL.so does not matter for compilation. Adding the path of whatever libEGL.so to LD_LIBRARY_PATH should also work. But the EGL header files should be up-to-date to include some necessary variables for compilation.

erikwijmans, 2019-09-03:
Ah yes, they should be :)