facebookresearch/habitat-sim #183: Problem with Setting Up Habitat-Sim

zNotAgain, 2019-09-03:
## Things I'm using
 1. `Ubuntu 16.04.1`
 2. `Miniconda`
 3. `Habitat-Sim already cloned`

## Steps to reproduce

  1. `conda create -n habitat python=3.6 cmake=3.14.0`
  2. `conda activate habitat`
  3. `cd habitat-sim`
  4. `pip install -r requirements.txt`
  5. `python setup.py install --headless`

## Observed Results
```
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
[0/1] Re-running CMake...
CMake Warning at deps/glog/CMakeLists.txt:49 (find_package):
  By not providing "Findgflags.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gflags", but
  CMake did not find one.

  Could not find a package configuration file provided by "gflags" (requested
  version 2.2.0) with any of the following names:

    gflagsConfig.cmake
    gflags-config.cmake

  Add the installation prefix of "gflags" to CMAKE_PREFIX_PATH or set
  "gflags_DIR" to a directory containing one of the above files.  If "gflags"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Shared libraries disabled
-- Looking for ZLIB...
-- Checking for module 'zzip-zlib-config'
--   No package 'zzip-zlib-config' found
-- Found ZLIB: optimized;/usr/lib/x86_64-linux-gnu/libz.so;debug;/usr/lib/x86_64-linux-gnu/libz.so
-- Checking for module 'minizip'
--   No package 'minizip' found
-- Enabled formats: AMF 3DS AC ASE ASSBIN ASSXML B3D BVH COLLADA DXF CSM HMP IRRMESH IRR LWO LWS MD2 MD3 MD5 MDC MDL NFF NDO OFF OBJ OGRE OPENGEX PLY MS3D COB BLEND IFC XGL FBX Q3D Q3BSP RAW SIB SMD STL TERRAGEN 3D X X3D GLTF 3MF MMD
-- Disabled formats:
-- Found PythonInterp: /home/zye009/miniconda3/envs/habitat/bin/python (found suitable version "3.6.9", minimum required is "3.6") 
-- Bindings being generated for python at /home/zye009/miniconda3/envs/habitat/bin/python
-- Found PythonInterp: /home/zye009/miniconda3/envs/habitat/bin/python (found version "3.6.9") 
-- pybind11 v2.3.0
-- Could NOT find EGL (missing: EGL_LIBRARY EGL_INCLUDE_DIR) 
CMake Error at deps/magnum/src/Magnum/Platform/CMakeLists.txt:796 (message):
  EGL library, required by some contexts, was not found.  Set
  WITH_*APPLICATION and/or WITH_EGLCONTEXT to OFF to skip building them.

-- Configuring incomplete, errors occurred!
See also "/home/zye009/habitat-sim/build/CMakeFiles/CMakeOutput.log".
See also "/home/zye009/habitat-sim/build/CMakeFiles/CMakeError.log".
FAILED: build.ninja 
/home/zye009/miniconda3/envs/habitat/bin/cmake -S/home/zye009/habitat-sim/src -B/home/zye009/habitat-sim/build
ninja: error: rebuilding 'build.ninja': subcommand failed
Traceback (most recent call last):
  File "setup.py", line 368, in <module>
    zip_safe=False,
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 195, in run
    self.build_extension(ext)
  File "setup.py", line 263, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--']' returned non-zero exit status 1.
```
I was thinking it has something with `WITH_EGLCONTEXT`, but I'm not sure. Any advice? 
## Expected Results

 Build Successful. 

## Relevant Code

```
// TODO(you): code here to reproduce the problem
```

erikwijmans, 2019-09-03:
Installing `libegl1-mesa-dev` should fix that error at least.  Keep in mind that `--headless` requires an Nvidia GPU with ~modern drivers.

zNotAgain, 2019-09-04:
I'm assuming I just have to run `apt-get install libegl1-mesa-dev`. But the catch is that I don't have root access 😢. Is there a way around this without root access? Or a way to set `WITH_EGLCONTEXT` to OFF?

erikwijmans, 2019-09-04:
EGL is required for `--headless`, so if you need that, there unfortunately isn't anything we can do.

zNotAgain, 2019-09-04:
I see. I tried with `python setup.py install` too. But now I have this error.
```
[20/219] Building CXX object deps/magnum/src/Magnum/Shaders/CMakeFiles/MagnumShaders.dir/Flat.cpp.o
[21/219] Building CXX object deps/tinyply/CMakeFiles/example.dir/source/example.cpp.o
[22/219] Building CXX object deps/magnum/src/Magnum/Platform/CMakeFiles/MagnumWindowlessGlxApplication.dir/WindowlessGlxApplication.cpp.o
FAILED: deps/magnum/src/Magnum/Platform/CMakeFiles/MagnumWindowlessGlxApplication.dir/WindowlessGlxApplication.cpp.o 
ccache /usr/bin/c++   -I/home/zye009/habitat-sim/src/deps -I/home/zye009/habitat-sim/src/deps/magnum/src/MagnumExternal/OpenGL -I/home/zye009/habitat-sim/src/deps/magnum/src -Ideps/magnum/src -I/home/zye009/habitat-sim/src/deps/corrade/src -Ideps/corrade/src -isystem /home/zye009/habitat-sim/src/cmake/../deps/eigen-git-mirror -isystem /home/zye009/habitat-sim/src/cmake/../deps/Sophus -isystem /home/zye009/habitat-sim/src/cmake/../deps/rapidjson/include -DVERSION_INFO=\"0.1.2\" -fvisibility=hidden -O2 -g -DNDEBUG -fPIC   -Wall -Wextra -Wold-style-cast -Winit-self -Werror=return-type -Wmissing-declarations -pedantic -fvisibility=hidden -fvisibility-inlines-hidden -Wzero-as-null-pointer-constant -Wdouble-promotion -fno-strict-aliasing -Wno-old-style-cast -std=gnu++14 -MD -MT deps/magnum/src/Magnum/Platform/CMakeFiles/MagnumWindowlessGlxApplication.dir/WindowlessGlxApplication.cpp.o -MF deps/magnum/src/Magnum/Platform/CMakeFiles/MagnumWindowlessGlxApplication.dir/WindowlessGlxApplication.cpp.o.d -o deps/magnum/src/Magnum/Platform/CMakeFiles/MagnumWindowlessGlxApplication.dir/WindowlessGlxApplication.cpp.o -c /home/zye009/habitat-sim/src/deps/magnum/src/Magnum/Platform/WindowlessGlxApplication.cpp
In file included from /home/zye009/habitat-sim/src/deps/magnum/src/Magnum/Platform/WindowlessGlxApplication.cpp:26:0:
/home/zye009/habitat-sim/src/deps/magnum/src/Magnum/Platform/WindowlessGlxApplication.h:167:33: error: ‘GLX_CONTEXT_FORWARD_COMPATIBLE_BIT_ARB’ was not declared in this scope
             ForwardCompatible = GLX_CONTEXT_FORWARD_COMPATIBLE_BIT_ARB,
                                 ^
/home/zye009/habitat-sim/src/deps/magnum/src/Magnum/Platform/WindowlessGlxApplication.h:175:21: error: ‘GLX_CONTEXT_DEBUG_BIT_ARB’ was not declared in this scope
             Debug = GLX_CONTEXT_DEBUG_BIT_ARB
                     ^
.
.
.
[32/219] Building CXX object deps/tinyply/CMakeFiles/tinyply.dir/source/tinyply.cpp.o
[33/219] Building CXX object deps/magnum-bindings/src/python/magnum/CMakeFiles/magnum.dir/platform/glx.cpp.o
FAILED: deps/magnum-bindings/src/python/magnum/CMakeFiles/magnum.dir/platform/glx.cpp.o 
ccache /usr/bin/c++  -Dmagnum_EXPORTS -I/home/zye009/habitat-sim/src/deps -I/home/zye009/habitat-sim/src/deps/magnum-bindings/src -I/home/zye009/habitat-sim/src/deps/magnum-bindings/src/python -Ideps/magnum-bindings/src/python -I/home/zye009/habitat-sim/src/deps/magnum/src -Ideps/magnum/src -I/home/zye009/habitat-sim/src/deps/corrade/src -Ideps/corrade/src -I/home/zye009/habitat-sim/src/deps/magnum/src/MagnumExternal/OpenGL -I/home/zye009/habitat-sim/src/deps/glfw/include -isystem /home/zye009/habitat-sim/src/cmake/../deps/eigen-git-mirror -isystem /home/zye009/habitat-sim/src/cmake/../deps/Sophus -isystem /home/zye009/habitat-sim/src/cmake/../deps/rapidjson/include -isystem /home/zye009/habitat-sim/src/deps/pybind11/include -isystem /home/zye009/miniconda3/envs/habitat/include/python3.6m -DVERSION_INFO=\"0.1.2\" -fvisibility=hidden -O2 -g -DNDEBUG -fPIC -fvisibility=hidden   -Wall -Wextra -Wold-style-cast -Winit-self -Werror=return-type -Wmissing-declarations -pedantic -fvisibility=hidden -fvisibility-inlines-hidden -Wzero-as-null-pointer-constant -Wdouble-promotion -flto -fno-fat-lto-objects -std=gnu++14 -MD -MT deps/magnum-bindings/src/python/magnum/CMakeFiles/magnum.dir/platform/glx.cpp.o -MF deps/magnum-bindings/src/python/magnum/CMakeFiles/magnum.dir/platform/glx.cpp.o.d -o deps/magnum-bindings/src/python/magnum/CMakeFiles/magnum.dir/platform/glx.cpp.o -c /home/zye009/habitat-sim/src/deps/magnum-bindings/src/python/magnum/platform/glx.cpp
In file included from /home/zye009/habitat-sim/src/deps/magnum-bindings/src/python/magnum/platform/glx.cpp:27:0:
/home/zye009/habitat-sim/src/deps/magnum/src/Magnum/Platform/WindowlessGlxApplication.h:167:33: error: ‘GLX_CONTEXT_FORWARD_COMPATIBLE_BIT_ARB’ was not declared in this scope
             ForwardCompatible = GLX_CONTEXT_FORWARD_COMPATIBLE_BIT_ARB,
                                 ^
/home/zye009/habitat-sim/src/deps/magnum/src/Magnum/Platform/WindowlessGlxApplication.h:175:21: error: ‘GLX_CONTEXT_DEBUG_BIT_ARB’ was not declared in this scope
             Debug = GLX_CONTEXT_DEBUG_BIT_ARB
                     ^
.
.
.
[53/219] Building CXX object esp/geo/CMakeFiles/geo.dir/OBB.cpp.o
[54/219] Building CXX object deps/magnum-bindings/src/python/magnum/CMakeFiles/magnum.dir/math.matrixfloat.cpp.o
[55/219] Building CXX object deps/magnum-bindings/src/python/magnum/CMakeFiles/magnum.dir/math.matrixdouble.cpp.o
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "setup.py", line 376, in <module>
    zip_safe=False,
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
[... cut ...]
  File "setup.py", line 273, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/home/zye009/miniconda3/envs/habitat/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--']' returned non-zero exit status 1.
```
This is the full log  --> [build.log](https://github.com/facebookresearch/habitat-sim/files/3573188/build.log).

erikwijmans, 2019-09-05:
Hmmm.  That not building suggests that this is a headless server (no display attached), so you'd definitely need to build with `--headless`.  Can you ask the system administrator to install `libegl1-mesa-dev`?

zNotAgain, 2019-09-07:
Thanks for your quick replies! They're now trying to give me a GUI server, will take about 2 days. I'll update again if there are any issues :)

zNotAgain, 2019-09-09:
Due to privilege issues, I simply did `apt-get download libegl1-mesa-dev`. How do I specify the cmake file to use this custom path for libegl? 

```
If your machine has a custom installation location for the nvidia OpenGL and EGL drivers, you may need to manually provide the EGL_LIBRARY path to cmake as follows. Add -DEGL_LIBRARY=/usr/lib/x86_64-linux-gnu/nvidia-opengl/libEGL.so to the build.sh command line invoking cmake. When running any executable adjust the environment as follows: LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/nvidia-opengl:${LD_LIBRARY_PATH} examples/example.py.
```
The above portion is from the habitat-sim repository description concerning common errors.

Where and how should I modify the `build.sh` file?

mathfac, 2019-09-23:
@zNotAgain any help needed here?

zNotAgain, 2019-09-25:
I'm now using a Linux Machine with GUI. When running `python setup.py install` on `habitat-sim`, I get this error:
```
Could NOT find Vulkan (missing: VULKAN_LIBRARY VULKAN_INCLUDE_DIR)
Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
Using X11 for window creation
CMake Error at deps/glfw/CMakeLists.txt:228 (message):
  The RandR headers were not found
```

erikwijmans, 2019-09-25:
GLFW needs some from `xorg-dev libglu1-mesa-dev`, i.e. `sudo apt install xorg-dev libglu1-mesa-dev`.

zNotAgain, 2019-09-26:
Thanks! It installs successfully now. I'm facing an issue when running `python examples/example.py` on `habitat-api`. It says,
```
2019-09-26 15:29:18,559 initializing sim Sim-v0
Renderer: llvmpipe (LLVM 8.0, 256 bits) by VMware, Inc.
OpenGL version: 3.3 (Core Profile) Mesa 19.0.8
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    mesa-forward-compatible-line-width-range
I0926 15:29:19.739832 23532 simulator.py:112] Loaded navmesh data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
Shader::compile(): compilation of vertex shader 1 failed with the following message:
0:1(10): error: GLSL 4.10 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.50, 3.30, 1.00 ES, and 3.00 ES
Shader::compile(): compilation of fragment shader 2 failed with the following message:
0:1(10): error: GLSL 4.10 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.50, 3.30, 1.00 ES, and 3.00 ES
Assertion Mn::GL::Shader::compile({vert, frag}) failed in /home/zyeo009/Desktop/habitat-sim/src/esp/gfx/DepthUnprojection.cpp on line 57
```
Do I have to install another version of GLSL?

zNotAgain, 2019-10-06:
This is a snippet of `/habitat-sim/src/esp/gfx/DepthUnprojection.cpp`

[image]

Do I have to change anything here?

erikwijmans, 2019-10-07:
I am not sure if we currently support GL < 4.1.  We were relying on a feature from 4.1, but we may longer need that.  If you can install a more recently of OpenGL, that would be preferable.

zNotAgain, 2019-10-08:
Hmm, so I tried `sudo apt-get update` and it seems the issues are resolved.

Here is the output when I run `python examples/example.py`
```
2019-10-08 10:18:45,208 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1008 10:18:45.211654 24751 WindowlessContext.cpp:99] [EGL] Detected 3 EGL devices
I1008 10:18:45.222843 24751 WindowlessContext.cpp:120] [EGL] Selected EGL device 0 for CUDA device 0
I1008 10:18:45.227041 24751 WindowlessContext.cpp:134] [EGL] Version: 1.5
I1008 10:18:45.227052 24751 WindowlessContext.cpp:135] [EGL] Vendor: NVIDIA
Renderer: GeForce GTX 1080/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 430.50
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
I1008 10:18:46.169758 24751 simulator.py:124] Loaded navmesh data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
2019-10-08 10:18:46,170 initializing task Nav-v0
Environment creation successful
Agent stepping around inside environment.
Episode finished after 3 steps.
I1008 10:18:46.209467 24751 Simulator.cpp:35] Deconstructing Simulator
I1008 10:18:46.209506 24751 SemanticScene.h:40] Deconstructing SemanticScene
I1008 10:18:46.209509 24751 SceneManager.h:24] Deconstructing SceneManager
I1008 10:18:46.209512 24751 SceneGraph.h:20] Deconstructing SceneGraph
I1008 10:18:46.209522 24751 Sensor.h:82] Deconstructing Sensor
I1008 10:18:46.209525 24751 RenderTarget.h:51] Deconstructing RenderTarget
I1008 10:18:46.222141 24751 Renderer.cpp:33] Deconstructing Renderer
I1008 10:18:46.222182 24751 WindowlessContext.h:16] Deconstructing WindowlessContext
I1008 10:18:46.222199 24751 WindowlessContext.cpp:245] Deconstructing GL context
I1008 10:18:46.222260 24751 PathFinder.h:113] Deconstructing PathFinder
```

Is this the expected response? 🤔

erikwijmans, 2019-10-08:
Yep!  That looks fine!

zNotAgain, 2019-10-08:
Great! Thanks for all the help :)

closed 2019-10-08 by erikwijmans

bigbike, 2020-01-14:
On fedora, one can do `sudo yum install mesa-libGL-devel`