facebookresearch/habitat-sim #199: How to set camera angle?

placeforyiming, 2019-09-09:
Play around habitat-sim for a while, and using the function below to set the sensors. Now, I want to set a camera pitch angle instead of forward facing. I checked the code, but didn't find any clue, so how could I do that?

def make_cfg(settings):
    sim_cfg = habitat_sim.SimulatorConfiguration()
    sim_cfg.gpu_device_id = 0
    sim_cfg.scene.id = settings["scene"]
    
    # Note: all sensors must have the same resolution
    sensors = {
        "color_sensor": {
            "sensor_type": habitat_sim.SensorType.COLOR,
            "resolution": [settings["height"], settings["width"]],
            "position": [0.0, settings["sensor_height"], 0.0],
        },
        "depth_sensor": {
            "sensor_type": habitat_sim.SensorType.DEPTH,
            "resolution": [settings["height"], settings["width"]],
            "position": [0.0, settings["sensor_height"], 0.0],
        },
        "semantic_sensor": {
            "sensor_type": habitat_sim.SensorType.SEMANTIC,
            "resolution": [settings["height"], settings["width"]],
            "position": [0.0, settings["sensor_height"], 0.0],
        },  
    }
    
    sensor_specs = []
    for sensor_uuid, sensor_params in sensors.items():
        if settings[sensor_uuid]:
            sensor_spec = habitat_sim.SensorSpec()
            sensor_spec.uuid = sensor_uuid
            sensor_spec.sensor_type = sensor_params["sensor_type"]
            sensor_spec.resolution = sensor_params["resolution"]
            sensor_spec.position = sensor_params["position"]

            sensor_specs.append(sensor_spec)
            
    # Here you can specify the amount of displacement in a forward action and the turn angle
    agent_cfg = habitat_sim.agent.AgentConfiguration()
    agent_cfg.sensor_specifications = sensor_specs
    agent_cfg.action_space = {
        "move_forward": habitat_sim.agent.ActionSpec(
            "move_forward", habitat_sim.agent.ActuationSpec(amount=0.1)
        ),
        "turn_left": habitat_sim.agent.ActionSpec(
            "turn_left", habitat_sim.agent.ActuationSpec(amount=20.0)
        ),
        "turn_right": habitat_sim.agent.ActionSpec(
            "turn_right", habitat_sim.agent.ActuationSpec(amount=20.0)
        ),
    }
    
    return habitat_sim.Configuration(sim_cfg, [agent_cfg])

aclegg3, 2019-09-09:
The agent_state also includes the sensor states. You can set the position of each sensor and rotation (quaternion) through agent.set_state() as well:

(code from [demo_runner.py - init_physics_test_scene() )](https://github.com/facebookresearch/habitat-sim/blob/69a469a514543888e3ebaac837311c9c97400145/examples/demo_runner.py#L99)
```
for sensor in agent_state.sensor_states:
     agent_state.sensor_states[sensor].rotation = agent_state.rotation
     agent_state.sensor_states[sensor].position = agent_state.position + np.array([0, 1.5, 0])
self._sim.get_agent(0).set_state(agent_state)
```

placeforyiming, 2019-09-09:
Can't make it work, I got  agent_state.sensor_states={}, here is my code:

`
random.seed(sim_settings["seed"])
sim.seed(sim_settings["seed"])

# Set agent state
agent = sim.initialize_agent(sim_settings["default_agent"])
agent_state = habitat_sim.AgentState()
agent_state.position = np.array([3.0, 0.072447, 0.0])
agent_state.rotation = np.quaternion(1, 0.0, 0.0, 0.0)

print (agent_state.sensor_states)

`

placeforyiming, 2019-09-09:
Problem solved!

Just found the attribute name of sensor is 

`sensor_spec.orientation =[0.34,0,0]`

not rotation

closed 2019-09-09 by placeforyiming