facebookresearch/habitat-sim #201: About the coordinate system

placeforyiming, 2019-09-10:
Is agent_state.position return (x,y,z) in world coordinate system?
Is agent_state.rotation return (QW,QX,QY,QZ) ?

I set 
```
agent_state.position = np.array([3.0, 0.072447, 0.0])
agent_state.rotation = np.quaternion(1, 0.0, 0.0, 0.0)
```
Then, found the second value in agent_state.position keep constant, so what does that mean?
After making the camera face upward a little bit
`sensor_spec.orientation =[0.34,0,0]`
Found all three values in  agent_state.position keep changing.

The action is among [move forawrd, turn left, turn right]

placeforyiming, 2019-09-10:
Ok, I think I got it, the default system is the camera coordinate in first frame, that's why if I add a pitch, there is a change in y direction.

If I'm right, please close this issue

erikwijmans, 2019-09-10:
Two things
1. `agent_state.position` is the position of the agent (which is always on the floor), not the position of the camera.  `agent_state.sensor_states` is the states of the sensors (in world coordinates).
1.  Doing `agent_state.position = ...` will not be propagated to the agent.  You need to do `agent.set_state(agent_state)` for the changes to be reflected on the agent. 

Also, in most cases, you should also do `agent_state.sensor_states = {}` beforehand.  If you do
```
agent_state = agent.state
agent_state.position = ...
agent.set_state(agent_state)
```
Then it won't look like anything changed as the sensors will still be in their previous positions.
```
agent_state = agent.state
agent_state.position = ...
agent_state.sensor_states = {}
agent.set_state(agent_state)
```
will cause the sensors to reset to their default orientation (which is defined relative to the agents new position).

placeforyiming, 2019-09-10:
> agent_state.sensor_states

This is what I got by printing agent pose and camera pose, it looks good, since I set the sensor height by:
`sim_settings = {
    "width": 256,  # Spatial resolution of the observations    
    "height": 256,
    "scene": test_scene,  # Scene path
    "default_agent": 0,  
    "sensor_height": 0.2,  # Height of sensors in meters
    "color_sensor": True,  # RGB sensor
    "semantic_sensor": True,  # Semantic sensor
    "depth_sensor": True,  # Depth sensor
    "seed": 1,
}`

and I set the pitch angle by
` sensor_specs = []
    for sensor_uuid, sensor_params in sensors.items():
        if settings[sensor_uuid]:
            sensor_spec = habitat_sim.SensorSpec()
            sensor_spec.uuid = sensor_uuid
            sensor_spec.sensor_type = sensor_params["sensor_type"]
            sensor_spec.resolution = sensor_params["resolution"]
            sensor_spec.position = sensor_params["position"]
            
            sensor_spec.orientation =[0.34,0,0]
            
            
            sensor_specs.append(sensor_spec)`

[image]

However, when I apply the actions and print out the pose of agent, I got this 
`[3.       0.072447 0.      ]
quaternion(1, 0, 0, 0)
action turn_left
[ 2.5829377  -2.3081307  -0.09995258]
quaternion(0.984807789325714, 0, 0.173648193478584, 0)
action turn_left
[ 2.5829377  -2.3081307  -0.09995258]
quaternion(0.939692616462708, 0, 0.342020153999329, 0)
action move_forward
[ 2.5186589  -2.3081307  -0.17655703]
quaternion(0.939692616462708, 0, 0.342020153999329, 0)
action move_forward
[ 2.45438   -2.3081307 -0.2531615]
quaternion(0.939692616462708, 0, 0.342020153999329, 0)
action move_forward
[ 2.3901012  -2.3081307  -0.32976595]
quaternion(0.939692616462708, 0, 0.342020153999329, 0)
action move_forward
[ 2.3258224 -2.2799525 -0.4063704]
quaternion(0.939692616462708, 0, 0.342020153999329, 0)
action move_forward
[ 2.2615435  -2.240534   -0.48297486]
quaternion(0.939692616462708, 0, 0.342020153999329, 0)
action turn_left
[ 2.2615435  -2.240534   -0.48297486]
quaternion(0.866025388240814, 0, 0.500000059604645, 0)
action turn_left
[ 2.2615435  -2.240534   -0.48297486]`

All the other poses look good except the first one

erikwijmans, 2019-09-10:
Is `np.array([3.0, 0.072447, 0.0])` navigable?  If you set the agent to a non-navigable position and then take an action, weird things can happen.

placeforyiming, 2019-09-10:
> Is `np.array([3.0, 0.072447, 0.0])` navigable? If you set the agent to a non-navigable position and then take an action, weird things can happen.

Then, how should I say it's navigable? Or maybe another question, should I trust the agent.state or should I trust the action? If weird things happen, the agent/camera pose should be always accurate, right? I  mean, for most cases, if the pose is accurate, although it didn't follow the action, algorithm still can work.

And personally I'm trying to do 3D reconstruction for replica, could I ask your opinion if this synthetic data is accurate enough to be reconstructed? For 3D reconstruction, I can blindly give the data and rely on bundle adjustment to get the pose, but feed the pose for each frames, is it accurate enough to reconstruct the whole scene?

closed 2019-09-10 by placeforyiming