facebookresearch/habitat-sim #205: Error during install: recompile with -fPIC

Danmou, 2019-09-11:
## Steps to reproduce
On Ubuntu 16.04.4 (headless) with miniconda installed:
```bash
sudo apt update && sudo apt install libjpeg-dev libpng-dev libglfw3-dev libglm-dev libx11-dev libomp-dev libegl1-mesa-dev
conda create -n habitat python=3.6 cmake=3.14
conda activate habitat
git clone https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim
pip install -r requirements.txt
python setup.py install --headless
```

## Observed Results
It fails with the following error:
```
Linking CXX shared module ../../lib.linux-x86_64-3.6/habitat_sim/_ext/habitat_sim_bindings.cpython-36m-x86_64-linux-gnu.so
/usr/bin/ld: /home/rpg_students/mouritzen/.local/lib/libgflags.a(gflags.cc.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/home/rpg_students/mouritzen/.local/lib/libgflags.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
```
<details><summary>Full output:</summary><p>

```
$ python setup.py install --headless
running install
running bdist_egg
running egg_info
creating habitat_sim.egg-info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-3.6
creating build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/geo.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/logging.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/utils.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/sensor.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/gfx.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/errors.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/simulator.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/scene.py -> build/lib.linux-x86_64-3.6/habitat_sim
creating build/lib.linux-x86_64-3.6/habitat_sim/nav
copying habitat_sim/nav/greedy_geodesic_follower.py -> build/lib.linux-x86_64-3.6/habitat_sim/nav
copying habitat_sim/nav/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/nav
creating build/lib.linux-x86_64-3.6/habitat_sim/sensors
copying habitat_sim/sensors/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors
copying habitat_sim/sensors/sensor_suite.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors
creating build/lib.linux-x86_64-3.6/habitat_sim/agent
copying habitat_sim/agent/agent.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent
copying habitat_sim/agent/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent
creating build/lib.linux-x86_64-3.6/habitat_sim/bindings
copying habitat_sim/bindings/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/bindings
creating build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/default_controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/pyrobot_noisy_controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
running build_ext
Submodule 'docs/m.css' (https://github.com/mosra/m.css) registered for path 'docs/m.css'
Submodule 'src/deps/Sophus' (https://github.com/strasdat/Sophus) registered for path 'src/deps/Sophus'
Submodule 'src/deps/assimp' (https://github.com/assimp/assimp) registered for path 'src/deps/assimp'
[... 1398 lines cut ...]
  File "/home/rpg_students/mouritzen/.conda/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/rpg_students/mouritzen/.conda/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/rpg_students/mouritzen/.conda/envs/habitat/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/rpg_students/mouritzen/.conda/envs/habitat/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/home/rpg_students/mouritzen/.conda/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/rpg_students/mouritzen/.conda/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 202, in run
    self.build_extension(ext)
  File "setup.py", line 274, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/home/rpg_students/mouritzen/.conda/envs/habitat/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--', '-j']' returned non-zero exit status 2.
```
</p></details>

Recompiling with `CMAKE_CXX_FLAGS="-fpic" python setup.py install --headless` gives the same result.

Any idea what is going wrong?

mosra, 2019-09-11:
This seems to be a similar case like in #179 (and another TODO for #121), it's apparently picking up gflags that are in `/home/rpg_students/mouritzen/.local/lib/` (it doesn't *need* them to work). Can you try passing `CMAKE_DISABLE_FIND_PACKAGE_gflags=TRUE` to CMake?

Danmou, 2019-09-11:
I removed `~/.local/lib` (don't need it anyway) and now it worked. Thanks for the hint!

closed 2019-09-11 by Danmou

mosra, 2019-09-25:
Opened #247 to fix properly on our side.