facebookresearch/habitat-sim #216: Question on the --with-cuda option

SenZHANG-GitHub, 2019-09-17:
I notice that this is an update of the installation documentation with an extra --with-cuda option. Wondering whether --with-cuda will include --headless? If we want both functionalities, should we install by "python setup.py install --headless --with-cuda"?

erikwijmans, 2019-09-17:
Yep, you should use both `--headless` and `--with-cuda`.  

`--with-cuda` doesn't enable `--headless` as you may want CUDA enabled features and the GUI viewer (currently there aren't any CUDA features that also work with the GUI viewer, but there may be some in the future!)

SenZHANG-GitHub, 2019-09-18:
> Yep, you should use both `--headless` and `--with-cuda`.
> 
> `--with-cuda` doesn't enable `--headless` as you may want CUDA enabled features and the GUI viewer (currently there aren't any CUDA features that also work with the GUI viewer, but there may be some in the future!)

Got it. Thanks！

closed 2019-09-18 by SenZHANG-GitHub