facebookresearch/habitat-sim #236: [Bug report] Navmesh not being respect in test-scenes

dhruvbatra, 2019-09-21:
## Steps to reproduce

1. Follow fresh install instructions. 
2. Follow testing instructions to open viewer. 

## Observed Results

Navmesh not respect. Agent can go through walls, floors, etc. See video. 

[image]

msbaines, 2019-09-23:
Ran git-bisect:

$ git bisect bad
9575dcd45fe6f55d2a44043833af08972a7895a9 is the first bad commit
commit 9575dcd45fe6f55d2a44043833af08972a7895a9
Author: Zhi-Yang He <hzyjerry@users.noreply.github.com>
Date:   Fri Aug 16 15:06:40 2019 -0700

    Bullet Physics Integration #1 (#91)

msbaines, 2019-09-23:
The navmesh handling code was commented out by the physics change:

  // connect controls to navmesh if loaded
  /*
  if (pathfinder_->isLoaded()) {
      controls_.setMoveFilterFunction([&](const vec3f& start, const vec3f& end)
  { vec3f currentPosition = pathfinder_->tryStep(start, end); LOG(INFO) <<
  "position=" << currentPosition.transpose() << " rotation="
                  << quatf(agentBodyNode_->rotation()).coeffs().transpose();
        LOG(INFO) << "Distance to closest obstacle: "
                  << pathfinder_->distanceToClosestObstacle(currentPosition);
        return currentPosition;
      });
    }
    */

closed 2019-09-24 by msbaines