facebookresearch/habitat-sim #241: NavMesh: unable to climb stairs in apartment_0 scene of replica_v1

msbaines, 2019-09-24:
## Steps to reproduce

  1. Try to walk up stairs
  2. Will only be able to climb the first few staris

## Observed Results

  * Can only climb the first few stairs

## Expected Results

  * Should be able to climb all the stairs.

dhruvbatra, 2019-09-24:
https://github.com/facebookresearch/Replica-Dataset/issues/20

aclegg3, 2019-11-08:
https://github.com/facebookresearch/Replica-Dataset/issues/20#issuecomment-551946323

aclegg3, 2020-01-08:
Can we close this, or do we want to modify the dataset with an updated navmesh using non-default params?

aclegg3, 2022-02-03:
Default agent parameters used to construct provided NavMesh files may not allow navigation through portions of some scenes with: narrow corridors/portals, low ceilings, steep stairs, etc... 

For more details and possible solutions, checkout the **Working with the NavMesh** section of the [Habitat-Sim Basics for Navigation](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/main/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb#scrollTo=iuzqsTfb1CCi) tutorial from our [ECCV 2020 tutorial series](https://aihabitat.org/tutorial/2020/).

closed 2022-02-03 by aclegg3