facebookresearch/habitat-sim #255: Check failed: eglDevId < numDevices [EGL] Could not find an EGL device for CUDA device 0

chenwydj, 2019-09-29:
## ❓ Questions and Help
I already integrated habitat env into the [A2C](https://github.com/ikostrikov/pytorch-a2c-ppo-acktr-gail). It works in `--num-processes 1`. However, when I set `--num-processes 2`, the habitat simulator raised the following errors.

Specifically, what does it mean by `Check failed: eglDevId < numDevices [EGL] Could not find an EGL device for CUDA device 0`? All my eight 1080Ti support opengl and work for a single process.

Thank you very much!

`
I0929 10:40:49.089522 29832 Simulator.cpp:42] Deconstructing Simulator
I0929 10:40:49.089660 29832 SemanticScene.h:40] Deconstructing SemanticScene
I0929 10:40:49.103410 29832 SceneManager.h:24] Deconstructing SceneManager
I0929 10:40:49.103449 29832 SceneGraph.h:20] Deconstructing SceneGraph
I0929 10:40:49.103891 29832 Sensor.h:82] Deconstructing Sensor
I0929 10:40:49.103901 29832 RenderTarget.h:51] Deconstructing RenderTarget
I0929 10:40:49.106288 29832 Sensor.h:82] Deconstructing Sensor
I0929 10:40:49.106312 29832 RenderTarget.h:51] Deconstructing RenderTarget
I0929 10:40:49.106452 29832 SceneGraph.h:20] Deconstructing SceneGraph
I0929 10:40:49.158921 29836 Simulator.cpp:42] Deconstructing Simulator
I0929 10:40:49.159044 29836 SemanticScene.h:40] Deconstructing SemanticScene
I0929 10:40:49.170749 29836 SceneManager.h:24] Deconstructing SceneManager
I0929 10:40:49.170784 29836 SceneGraph.h:20] Deconstructing SceneGraph
I0929 10:40:49.171011 29836 Sensor.h:82] Deconstructing Sensor
I0929 10:40:49.171020 29836 RenderTarget.h:51] Deconstructing RenderTarget
I0929 10:40:49.178689 29836 Sensor.h:82] Deconstructing Sensor
I0929 10:40:49.178714 29836 RenderTarget.h:51] Deconstructing RenderTarget
I0929 10:40:49.178874 29836 SceneGraph.h:20] Deconstructing SceneGraph
I0929 10:40:49.537631 29832 Renderer.cpp:33] Deconstructing Renderer
I0929 10:40:49.537721 29832 WindowlessContext.h:16] Deconstructing WindowlessContext
I0929 10:40:49.537730 29832 WindowlessContext.cpp:245] Deconstructing GL context
I0929 10:40:49.538054 29832 PathFinder.h:113] Deconstructing PathFinder
2019-09-29 10:40:49,563 initializing sim Sim-v0
I0929 10:40:49.570120 29832 WindowlessContext.cpp:99] [EGL] Detected 8 EGL devices
F0929 10:40:49.573048 29832 WindowlessContext.cpp:113] Check failed: eglDevId < numDevices [EGL] Could not find an EGL device for CUDA device 0
*** Check failure stack trace: ***
I0929 10:40:49.666532 29836 Renderer.cpp:33] Deconstructing Renderer
I0929 10:40:49.666630 29836 WindowlessContext.h:16] Deconstructing WindowlessContext
I0929 10:40:49.666640 29836 WindowlessContext.cpp:245] Deconstructing GL context
I0929 10:40:49.666983 29836 PathFinder.h:113] Deconstructing PathFinder
2019-09-29 10:40:49,694 initializing sim Sim-v0
I0929 10:40:49.704221 29836 WindowlessContext.cpp:99] [EGL] Detected 8 EGL devices
F0929 10:40:49.711859 29836 WindowlessContext.cpp:113] Check failed: eglDevId < numDevices [EGL] Could not find an EGL device for CUDA device 0
*** Check failure stack trace: ***
I0929 10:40:55.815344 29150 Simulator.cpp:42] Deconstructing Simulator
I0929 10:40:55.815410 29150 SemanticScene.h:40] Deconstructing SemanticScene
I0929 10:40:55.826141 29150 SceneManager.h:24] Deconstructing SceneManager
I0929 10:40:55.826181 29150 SceneGraph.h:20] Deconstructing SceneGraph
I0929 10:40:55.826371 29150 Sensor.h:82] Deconstructing Sensor
I0929 10:40:55.826378 29150 RenderTarget.h:51] Deconstructing RenderTarget
I0929 10:40:55.830835 29150 Sensor.h:82] Deconstructing Sensor
I0929 10:40:55.830860 29150 RenderTarget.h:51] Deconstructing RenderTarget
I0929 10:40:55.831213 29150 SceneGraph.h:20] Deconstructing SceneGraph
I0929 10:40:56.089581 29150 Renderer.cpp:33] Deconstructing Renderer
I0929 10:40:56.089658 29150 WindowlessContext.h:16] Deconstructing WindowlessContext
I0929 10:40:56.089666 29150 WindowlessContext.cpp:245] Deconstructing GL context
I0929 10:40:56.089870 29150 PathFinder.h:113] Deconstructing PathFinder
`

erikwijmans, 2019-09-30:
You will need to change the line here: https://github.com/ikostrikov/pytorch-a2c-ppo-acktr-gail/blob/master/a2c_ppo_acktr/envs.py#L91 to `context="forkserver"` as forked subprocesses don't work with habitat.

chenwydj, 2019-09-30:
> You will need to change the line here: https://github.com/ikostrikov/pytorch-a2c-ppo-acktr-gail/blob/master/a2c_ppo_acktr/envs.py#L91 to `context="forkserver"` as forked subprocesses don't work with habitat.

Thank you so much! It works!

closed 2019-09-30 by chenwydj