facebookresearch/habitat-sim #256: Compilation failed with error "EGL library, required by some contexts, was not found"

duggalrahul, 2019-09-30:
Hi,

I'm trying to install the headless version by calling `python setup.py install --headless`, however, I get the following error.

> -- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Could NOT find EGL (missing: EGL_INCLUDE_DIR) 
CMake Error at deps/magnum/src/Magnum/Platform/CMakeLists.txt:802 (message):
  EGL library, required by some contexts, was not found.  Set
  WITH_*APPLICATION and/or WITH_EGLCONTEXT to OFF to skip building them.

I figure that CMake is not able to locate EGL libraries. Couple of things I've tried after going through previous issues:

1. Installed `libegl1-mesa-dev`.

2. Added `-DEGL_LIBRARY=/usr/lib/x86_64-linux-gnu/libEGL.so` to the setup.py script at line 217 so that CMake gets called with this argument. 
I know that `DEGL_LIBRARY` is set because the error message also outputs the exact CMake command called from setup.py. The cmake command is mentioned below with the line containing `DEGL_LIBRARY` bolded. 

> subprocess.CalledProcessError: Command '['cmake', '-H/nethome/rduggal7/habitat/habitat-sim/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', **'-DEGL_LIBRARY=/usr/lib/x86_64-linux-gnu/libEGL.so'**,
 '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/nethome/rduggal7/habitat/habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/nethome/rduggal7/.conda/envs/habitat/bin/python',
 '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-GNinja', '-DBUILD_GUI_VIEWERS=OFF', '-DTARGET_HEADLESS=ON', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBU
ILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=OFF']' returned non-zero exit status 1.

Looking for pointers to resolve this.

Thanks!

erikwijmans, 2019-09-30:
Looks like `EGL_INCLUDE_DIR` also needs to be specified (in the same way as `EGL_LIBRARY`).  The EGL headers are probably in `/usr/include/EGL`.  Odd that cmake isn't picking things up.

mathfac, 2019-09-30:
@duggalrahul, most probably @erikwijmans's suggestion help. If not you can also check step by step [habitat_sim installation on fresh Ubuntu used for our CI testing](https://github.com/facebookresearch/habitat-sim/blob/master/.circleci/config.yml#L62-L201).

duggalrahul, 2019-10-01:
Had to install both `libgl1-mesa-dev` and `libegl1-mesa-dev`

The compilation issue is resolved. Thanks!

closed 2019-10-01 by duggalrahul

bigbike, 2020-01-29:
On Fedora:
sudo yum install mesa-libEGL-devel

KenaHemnani, 2020-12-12:
Hii all , I encountered similar kind of error:
-- Could NOT find EGL (missing: EGL_INCLUDE_DIR)
CMake Error at deps/magnum/src/Magnum/Platform/CMakeLists.txt:796 (message):
                             ........
Then , I installed **libgl1-mesa-dev** and **libegl1-mesa-dev** via. sudo apt-get install libgl1-mesa-dev and 
sudo apt-get install libegl1-mesa-dev. 

Running , **python examples/example.py** I am getting following error:
(habitat1) hemnani1@boss8:~/DIR1/habitat-lab$ python examples/example.py
2020-12-12 18:49:48,396 Initializing dataset PointNav-v1
Traceback (most recent call last):
  File "examples/example.py", line 28, in <module>
    example()
  File "examples/example.py", line 14, in example
    config=habitat.get_config("configs/tasks/pointnav.yaml")
  File "/home/hemnani1/DIR1/habitat-lab/habitat/core/env.py", line 78, in __init_                                                          _
    id_dataset=config.DATASET.TYPE, config=config.DATASET
  File "/home/hemnani1/DIR1/habitat-lab/habitat/datasets/registration.py", line 2                                                          0, in make_dataset
    return _dataset(**kwargs)
  File "/home/hemnani1/DIR1/habitat-lab/habitat/datasets/pointnav/__init__.py", l                                                          ine 30, in __init__
    raise pointnav_import_error
  File "/home/hemnani1/DIR1/habitat-lab/habitat/datasets/pointnav/__init__.py", l                                                          ine 20, in _try_register_pointnavdatasetv1
    from habitat.datasets.pointnav.pointnav_dataset import (  # noqa: F401
  File "/home/hemnani1/DIR1/habitat-lab/habitat/datasets/pointnav/pointnav_datase                                                          t.py", line 15, in <module>
    from habitat.tasks.nav.nav import (
  File "/home/hemnani1/DIR1/habitat-lab/habitat/tasks/nav/nav.py", line 31, in <m                                                          odule>
    from habitat.sims.habitat_simulator.actions import HabitatSimActions
  File "/home/hemnani1/DIR1/habitat-lab/habitat/sims/habitat_simulator/actions.py                                                          ", line 12, in <module>
    import habitat_sim
  File "/home/hemnani1/anaconda3/envs/habitat1/lib/python3.6/site-packages/habita                                                          t_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/__init__.py", line 13, in <module>
    from habitat_sim import (  # noqa: F401
  File "/home/hemnani1/anaconda3/envs/habitat1/lib/python3.6/site-packages/habita                                                          t_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/agent/__init__.py", line 7, in <mo                                                          dule>
    from .agent import *  # noqa: F401,F403
  File "/home/hemnani1/anaconda3/envs/habitat1/lib/python3.6/site-packages/habita                                                          t_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/agent/agent.py", line 10, in <modu                                                          le>
    import magnum as mn
  File "/home/hemnani1/anaconda3/envs/habitat1/lib/python3.6/site-packages/magnum                                                          /__init__.py", line 28, in <module>
    from _magnum import *
ImportError: /lib/x86_64-linux-gnu/libOpenGL.so.0: undefined symbol: _glapi_tls_C                                                          urrent

Can anyone help me to resolve this issue.