facebookresearch/habitat-sim #263: Category Labels for Semantics

chenwydj, 2019-10-02:
## ❓ Questions and Help

For mp3d there are 40 semantic categories. I can see that the color of the semantic segmentation is based on `semantic_img.putdata((semantic_obs.flatten() % 40).astype(np.uint8))` in [this demo](https://github.com/facebookresearch/habitat-api/blob/e0d4056180fea3238a51cec1b2fe47c09a8f08a2/docs/pages/habitat-sim-demo.rst).

Could you provide the meanings of the semantics, i.e. the category label for each semantic index? I found the indices loaded by the SEMANTIC_SENSOR do not match the [original mp3d labels](https://github.com/niessner/Matterport/blob/master/metadata/category_mapping.tsv).

Thank you very much!

erikwijmans, 2019-10-02:
The semantic observations corresponds to the _instance_ segmentation of the scene.  You can associate an instance ID to an object to get the semantic label.  https://aihabitat.org/docs/habitat-api/habitat-sim-demo.html#scene-semantic-annotations
https://aihabitat.org/docs/habitat-sim/habitat_sim.scene.SemanticObject.html

chenwydj, 2019-10-02:
> The semantic observations corresponds to the _instance_ segmentation of the scene. You can associate an instance ID to an object to get the semantic label.
> Pointers:
> https://aihabitat.org/docs/habitat-api/habitat-sim-demo.html#scene-semantic-annotations
> https://aihabitat.org/docs/habitat-sim/habitat_sim.scene.SemanticObject.html

Thank you Erik!

1. I'm currently using the habitat_api to create an env and having an agent walk in the room, the best possible solution would be to directly link the semantic observations to the category label. Would this be possible? (btw I found the range of the value of my semantic observation is from 0 to over 300)

2. I read your links, and basically they said to build a scene object by loading a room file (.glb). I think it will take much extra work to associate the agent's current state to the scene object each time.

3. Also, is the integer value from the semantic observations consistent across all rooms? how about the Object id stored in the scene?

Thank you!

erikwijmans, 2019-10-02:
Should have pointed you to this one: https://aihabitat.org/docs/habitat-api/habitat-api-demo.html#scene-semantic-annotations

The semantic scene is already loaded automatically.  Creating an id_to_label_id mapping is then just
```
instance_id_to_label_id = {int(obj.id.split("_")[-1]): obj.category.index() for obj in scene.objects}
```

> btw I found the range of the value of my semantic observation is from 0 to over 300

That is expected.  Once again, the semantic observations is the _instance_ segmentation.  There are a lot of object in MP3D environments.

chenwydj, 2019-10-02:
Thank you Erik!

This is great! However, things get a bit tricky, since I already wrapped the habitat env into the [openai baselines ShmemVecEnv](https://github.com/openai/baselines/blob/master/baselines/common/vec_env/shmem_vec_env.py#L20) and it seems hard to get the original env object back :(

erikwijmans, 2019-10-02:
The best way to do this in a vectorized way would be to add a new sensor that outputs semantic segmentation instead of instance segmentation.  Like the sensor here: https://github.com/facebookresearch/habitat-api/blob/master/habitat/sims/habitat_simulator/habitat_simulator.py#L120 but applies the mapping.

chenwydj, 2019-10-02:
I wrote a workaround for the instance2semantic mapping in the reset() of habitat/core/env.py:
```python
        # obtain mapping from instance id to semantic label id
        scene = self.sim.semantic_annotations()
        instance_id_to_label_id = {int(obj.id.split("_")[-1]): obj.category.index() for obj in scene.objects}
        self.mapping = np.array([ instance_id_to_label_id[i] for i in range(len(instance_id_to_label_id)) ])

        if 'semantic' in observations:
            observations['semantic'] = np.take(self.mapping, observations['semantic']
```
and in step():
```python
        if 'semantic' in observations:
            observations['semantic'] = np.take(self.mapping, observations['semantic']
```

This works for me now. Thank you Erik!

closed 2019-10-02 by chenwydj

BryanBo-Cao, 2020-03-01:
> instance_id_to_label_id = {int(obj.id.split("_")[-1]): obj.category.index() for obj in scene.objects}

Hi @erikwijmans, thanks for explaining this.

Does the label_id specifically refer to the object's category index?

BryanBo-Cao, 2020-03-01:
> > instance_id_to_label_id = {int(obj.id.split("_")[-1]): obj.category.index() for obj in scene.objects}
> 
> Hi @erikwijmans, thanks for explaining this.
> 
> Does the label_id specifically refer to the object's category index?

To me, yes.

erikwijmans, 2020-03-02:
Yep, I am using label_id and category index interchangably here.

AgentEXPL, 2022-07-06:
> 

@erikwijmans  Is it correct that one category index (e.g., 2) may indicate different categories in different scenes? In hm3d mini_val dataset, category index "2" means 'smoke detector' in one scene. After numbers of env.reset(), category index "2" means 'wall' in another scene.

francesco-taioli, 2023-03-21:
Thanks @chenwydj ! Your solution works perfectly