facebookresearch/habitat-sim #266: Issue with Building with bullet physics

BurakHocaoglu, 2019-10-02:
## ❓ Questions and Help
Hello,
I need help on building. I have used habitat-sim a while ago, but when I tried to build it with bullet physics after the latest commits, I got the error below:
```
[1/1] Linking CXX executable utils/viewer/viewer
FAILED: utils/viewer/viewer 
: && /usr/bin/c++  -DVERSION_INFO=\"0.1.2\" -fvisibility=hidden -O2 -g -DNDEBUG   utils/viewer/CMakeFiles/viewer.dir/viewer.cpp.o utils/viewer/CMakeFiles/viewer.dir/__/__/deps/magnum/src/MagnumPlugins/AnyImageImporter/importStaticPlugin.cpp.o utils/viewer/CMakeFiles/viewer.dir/__/__/deps/magnum/src/MagnumPlugins/AnySceneImporter/importStaticPlugin.cpp.o utils/viewer/CMakeFiles/viewer.dir/__/__/deps/magnum-plugins/src/MagnumPlugins/StbImageImporter/importStaticPlugin.cpp.o utils/viewer/CMakeFiles/viewer.dir/__/__/deps/magnum-plugins/src/MagnumPlugins/StbImageConverter/importStaticPlugin.cpp.o utils/viewer/CMakeFiles/viewer.dir/__/__/deps/magnum-plugins/src/MagnumPlugins/TinyGltfImporter/importStaticPlugin.cpp.o utils/viewer/CMakeFiles/viewer.dir/__/__/deps/magnum/src/MagnumPlugins/AnyImageConverter/importStaticPlugin.cpp.o utils/viewer/CMakeFiles/viewer.dir/__/__/deps/magnum-plugins/src/MagnumPlugins/AssimpImporter/importStaticPlugin.cpp.o  -o utils/viewer/viewer  -Wl,-rpath,/usr/local/lib esp/assets/libassets.a esp/gfx/libgfx.a esp/nav/libnav.a esp/scene/libscene.a deps/magnum/src/Magnum/DebugTools/libMagnumDebugTools.a esp/agent/libagent.a esp/sensor/libsensor.a esp/assets/libassets.a esp/gfx/libgfx.a esp/scene/libscene.a esp/physics/libphysics.a esp/physics/bullet/libbulletphysics.a esp/assets/libassets.a esp/gfx/libgfx.a esp/scene/libscene.a esp/physics/libphysics.a esp/physics/bullet/libbulletphysics.a deps/tinyply/libtinyply.a deps/magnum-plugins/src/MagnumPlugins/AssimpImporter/libAssimpImporter.a deps/assimp/code/libassimp.a /usr/lib/x86_64-linux-gnu/libz.so deps/assimp/contrib/irrXML/libIrrXML.a /usr/lib/gcc/x86_64-linux-gnu/5/libgomp.so -lpthread deps/magnum/src/MagnumPlugins/AnySceneImporter/libAnySceneImporter.a deps/magnum/src/MagnumPlugins/AnyImageConverter/libAnyImageConverter.a deps/magnum/src/Magnum/Platform/libMagnumGlfwApplication.a deps/glfw/src/libglfw3.a /usr/lib/x86_64-linux-gnu/librt.so -lm deps/magnum/src/Magnum/Platform/libMagnumWindowlessGlxApplication.a /usr/lib/x86_64-linux-gnu/libGLX.so /usr/lib/x86_64-linux-gnu/libSM.so /usr/lib/x86_64-linux-gnu/libICE.so /usr/lib/x86_64-linux-gnu/libX11.so /usr/lib/x86_64-linux-gnu/libXext.so esp/geo/libgeo.a esp/io/libio.a deps/magnum-plugins/src/MagnumPlugins/StbImageImporter/libStbImageImporter.a deps/magnum-plugins/src/MagnumPlugins/StbImageConverter/libStbImageConverter.a deps/magnum-plugins/src/MagnumPlugins/TinyGltfImporter/libTinyGltfImporter.a deps/magnum/src/MagnumPlugins/AnyImageImporter/libAnyImageImporter.a deps/magnum-integration/src/Magnum/BulletIntegration/libMagnumBulletIntegration.a /usr/local/lib/libBulletDynamics.so /usr/local/lib/libBulletCollision.so /usr/local/lib/libLinearMath.so /usr/local/lib/libBulletSoftBody.so esp/core/libcore.a deps/glog/libglog.a -lpthread deps/recastnavigation/Detour/libDetour.a deps/recastnavigation/Recast/libRecast.a deps/magnum/src/Magnum/MeshTools/libMagnumMeshTools.a deps/magnum/src/Magnum/SceneGraph/libMagnumSceneGraph.a deps/magnum/src/Magnum/Shaders/libMagnumShaders.a deps/magnum/src/Magnum/GL/libMagnumGL.a /usr/lib/x86_64-linux-gnu/libOpenGL.so deps/corrade/src/Corrade/TestSuite/libCorradeTestSuite.a deps/magnum/src/Magnum/Primitives/libMagnumPrimitives.a deps/magnum/src/Magnum/Trade/libMagnumTrade.a deps/magnum/src/Magnum/libMagnum.a deps/corrade/src/Corrade/PluginManager/libCorradePluginManager.a deps/corrade/src/Corrade/Utility/libCorradeUtility.a -ldl && :
esp/physics/bullet/libbulletphysics.a(BulletRigidObject.cpp.o): In function `esp::physics::BulletRigidObject::setInertiaVector(Magnum::Math::Vector3<float> const&)':
/home/rtx80/Desktop/bhocaoglu/habitat-sim/src/esp/physics/bullet/BulletRigidObject.cpp:356: undefined reference to `btRigidBody::setMassProps(float, btVector3 const&)'
esp/physics/bullet/libbulletphysics.a(BulletRigidObject.cpp.o): In function `esp::physics::BulletRigidObject::setMass(double)':
/home/rtx80/Desktop/bhocaoglu/habitat-sim/src/esp/physics/bullet/BulletRigidObject.cpp:340: undefined reference to `btRigidBody::setMassProps(float, btVector3 const&)'
esp/physics/bullet/libbulletphysics.a(BulletRigidObject.cpp.o): In function `std::_MakeUniq<btConvexHullShape>::__single_object std::make_unique<btConvexHullShape, float const*, unsigned long, unsigned long>(float const*&&, unsigned long&&, unsigned long&&)':
/usr/include/c++/5/bits/unique_ptr.h:765: undefined reference to `btConvexHullShape::btConvexHullShape(float const*, int, int)'
esp/physics/bullet/libbulletphysics.a(BulletRigidObject.cpp.o): In function `esp::physics::BulletRigidObject::setAngularDamping(double)':
/home/rtx80/Desktop/bhocaoglu/habitat-sim/src/esp/physics/bullet/BulletRigidObject.cpp:401: undefined reference to `btRigidBody::setDamping(float, float)'
esp/physics/bullet/libbulletphysics.a(BulletRigidObject.cpp.o): In function `esp::physics::BulletRigidObject::setLinearDamping(double)':
/home/rtx80/Desktop/bhocaoglu/habitat-sim/src/esp/physics/bullet/BulletRigidObject.cpp:393: undefined reference to `btRigidBody::setDamping(float, float)'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "setup.py", line 379, in <module>
    zip_safe=False,
  File "/home/rtx80/anaconda3/lib/python3.7/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/home/rtx80/anaconda3/lib/python3.7/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/rtx80/anaconda3/lib/python3.7/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/home/rtx80/anaconda3/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/rtx80/anaconda3/lib/python3.7/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/home/rtx80/anaconda3/lib/python3.7/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/home/rtx80/anaconda3/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/rtx80/anaconda3/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/rtx80/anaconda3/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/rtx80/anaconda3/lib/python3.7/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/home/rtx80/anaconda3/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/rtx80/anaconda3/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/rtx80/anaconda3/lib/python3.7/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/rtx80/anaconda3/lib/python3.7/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/home/rtx80/anaconda3/lib/python3.7/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/rtx80/anaconda3/lib/python3.7/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "setup.py", line 202, in run
    self.build_extension(ext)
  File "setup.py", line 276, in build_extension
    shlex.split("cmake --build {}".format(self.build_temp)) + build_args
  File "/home/rtx80/anaconda3/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'build', '--config', 'RelWithDebInfo', '--']' returned non-zero exit status 1.
```

I have built and installed the bullet physics as its repo tells and I did not get any errors (though importing ```pybullet``` gives an error in python 3.7 and ```pip install``` does not work as it gives a weird stack smash-like error), as well as the mesa drivers. If its about ```corrade```, building it from source did not improve the situation. As for ```magnum```, when I tried to build it from source to see if it helps, it actually made it worse by adding more undefined references for data types defined in ```magnum``` package. The weird thing is that this did not happen when I first used habitat-sim with bullet integration. 

Building without bullet physics works, btw.

Thanks in advance...

Burak

aclegg3, 2019-10-02:
Hey Burak, there was a bug in the Bullet build this week with setting the cmake variables. Be sure you are up to date with PR #261 and then do a clean install by deleting the `build/` directory. Typically, we find that Bullet is best picked up when installed with `apt-get install  libbullet-dev` on Linux or `brew install bullet` on MacOS.

BurakHocaoglu, 2019-10-03:
It did not change anything, unfortunately.

BurakHocaoglu, 2019-10-07:
I think I solved it for now. For some reason, I just could not set the version of the commit you specified. Then, I changed the cmake code by myself and it worked. Thanks...

BurakHocaoglu, 2019-10-09:
To be honest, I hate to re-open this issue, but when I succeeded with the ```loaded-mesh-bounding-boxes``` branch, I realized that the gravity did not work. The objects were hung in the air where they're spawned. I thought that maybe if I run ```python setup.py install --bullet```, it may be fixed. However, I got the same error in the beginning and I do not know why. 

Btw, my previous successful build is the result of ```./build.sh --bullet```, instead of python. Do these 2 have a different mechanism?

erikwijmans, 2019-10-09:
I can't comment on the other part (cc @aclegg3), but for

> Btw, my previous successful build is the result of ./build.sh --bullet, instead of python. Do these 2 have a different mechanism?

No, they don't have different mechanisms.  `build.sh` is a wrapper around `setup.py` that just causes the bindings to be built "in-tree" for development purposes.  Something like viewer isn't effected at all.

aclegg3, 2019-10-09:
Sorry to hear you are struggling with this, Burak. I'll need some more information to help you out. I assume you installed bullet with either `apt-get install libbullet-dev` on linux or `brew install bullet`. 

Then try the following(be sure you have activated your conda env): 

- `cd` into habitat-sim directory 
- clear cached build with `rm -r build`
- `git pull` any updates (best to debug with `master` branch now)
- then build with `./build.sh --bullet'
- then try the viewer with './build/viewer --enable-physics -- "NONE"'

You should see an empty window and terminal output like:
[image]
 
- Then press 'o' key to add objects and they should fall.

If this is not the case, post terminal output from the point you called './build/viewer --enable-physics -- "NONE"' and anything suspicious in build like your original post. Let's try to get this working! :grinning:

BurakHocaoglu, 2019-10-31:
Hello there,

Sorry for the long wait, I had to take care of many other things.

To make it completely clear, I started from scratch by setting up Ubuntu 16.04 on a new machine. Then, I followed what you have suggested and it worked. So, I moved on to my next task of projecting 3D bounding cubes onto 2D image plane to make bounding boxes. Thanks for all your help.

closed 2019-10-31 by BurakHocaoglu