facebookresearch/habitat-sim #274: Objects disappear after a few kinematic nudges

dhruvbatra, 2019-10-05:
## 🐛 Bug

In the viewer demo with physics enables, object disappear after a few (typically 3-4) kinematic nudges. See video.  

## Command
`build/viewer --enable-physics /path/to/data/scene_datasets/habitat-test-scenes/van-gogh-room.glb`

## Expected behavior

Object permanence. 

## Additional context

In the video, I am pressing 'k' repeatedly after introducing every new object. 

[image]

aclegg3, 2019-10-05:
Thanks for bringing this up, Dhruv. This is actually expected behavior. If you kinematically nudge a dynamic object through the floor mesh enough, collision response will push it through the other side. In these cases, the objects are actually falling forever after popping through the floor. This kind of behavior is why kinematic control is not recommended for MotionType::DYNAMIC objects. That said, this function could be changed to only create random nudges with positive Y direction in order to prevent pop through for the floor or tables as seen here:

[image]

dhruvbatra, 2019-10-05:
I see, thanks for the context. 

> That said, this function could be changed to only create random nudges with positive Y direction in order to prevent pop through for the floor or tables as seen here:

Yes, let's do that. Objects passing through the floor is unintuitive.

closed 2019-10-09 by aclegg3