facebookresearch/habitat-sim #283: Reconfigure does not work with objects if image types changes

dhuber23b, 2019-10-08:
## 🐛 Bug

You can add objects to the scene using the physics library capability in Habitat.  If you initially set the image types to none and configure the simulator and then later add new image types when you call reconfigure, the object is not added (or at least it is not displayed).

## To Reproduce

Attaching a zip file with code that reproduces the problem.  It uses the replica data, which is too large to attach, but the problem should be reproducible with any simple scene.

## System Info
ENVIRONMENT INFO:
Platform: Darwin-18.7.0-x86_64-i386-64bit
Machine: x86_64
Processor: i386
Libc version:
Mac version: 10.14.6
Python version: 3.6.2+
Architecture: 64bit
Win version:
System OS: Darwin
Release: 18.7.0
Version: Darwin Kernel Version 18.7.0: Tue Aug 20 16:57:14 PDT 2019; root:xnu-4903.271.2~2/RELEASE_X86_64
Operational System: darwin
GCC version: Apple clang version 11.0.0 (clang-1100.0.33.5)
CMAKE version: cmake version 3.15.4
NVIDIA-SMI:
Pip packages versions:

Conda packages versions:

[reconfigure_bug.zip](https://github.com/facebookresearch/habitat-sim/files/3704067/reconfigure_bug.zip)

aclegg3, 2019-10-08:
Hey Daniel, can you tell me what error output you get? It may be an issue with semantics for objects added through the physicsManager. If I initialize the simulator with no image types `{}` then ask for `{"color":True, "depth":True}` and reconfigure, it works fine.

dhuber23b, 2019-10-08:
Alex, 

There is no error message.  It just does not show the object.  Maybe I am not setting it up correctly.  You can see the steps I use in the sample code.  Below is the text of the main script that reproduces it:

from collections import namedtuple
import habitat_sim
from habitat_sim.utils import common
from habitat_sim.utils.common import d3_40_colors_rgb
import matplotlib.pyplot as plt
import numpy as np
import os
import random
import shutil
from PIL import Image
from typing import List

Pose = namedtuple('Pose', ['translation', 'quaternion'])

def make_habitat_cfg(settings, scene_file, image_types):
    sim_cfg = habitat_sim.SimulatorConfiguration()
    sim_cfg.gpu_device_id = 0
    sim_cfg.scene.id = scene_file
    sim_cfg.enable_physics = True
    sim_cfg.physics_config_file = settings["physics_config_file"]

    # Note: all sensors must have the same resolution
    sensors = {
        "color": {
            "sensor_type": habitat_sim.SensorType.COLOR,
            "resolution": [settings["image_height"], settings["image_width"]],
            "position": [0.0, 1.5, 0.0],  #::: fix y to be 0 later
        },
        "depth": {
            "sensor_type": habitat_sim.SensorType.DEPTH,
            "resolution": [settings["image_height"], settings["image_width"]],
            "position": [0.0, 1.5, 0.0],
        },
        "semantic": {
            "sensor_type": habitat_sim.SensorType.SEMANTIC,
            "resolution": [settings["image_height"], settings["image_width"]],
            "position": [0.0, 1.5, 0.0],
        },  
    }

    sensor_specs = []
    for sensor_uuid, sensor_params in sensors.items():
        if image_types.get(sensor_uuid, False):
            sensor_spec = habitat_sim.SensorSpec()
            sensor_spec.uuid = sensor_uuid
            sensor_spec.sensor_type = sensor_params["sensor_type"]
            sensor_spec.resolution = sensor_params["resolution"]
            sensor_spec.position = sensor_params["position"]

            sensor_specs.append(sensor_spec)

    agent_cfg = habitat_sim.agent.AgentConfiguration()
    agent_cfg.sensor_specifications = sensor_specs
    agent_cfg.action_space = { #::: fix: are all these needed?
        "move_forward": habitat_sim.agent.ActionSpec(
            "move_forward", habitat_sim.agent.ActuationSpec(amount=0.25)
        ),
        "turn_left": habitat_sim.agent.ActionSpec(
            "turn_left", habitat_sim.agent.ActuationSpec(amount=30.0)
        ),
        "turn_right": habitat_sim.agent.ActionSpec(
            "turn_right", habitat_sim.agent.ActuationSpec(amount=30.0)
        ),
    }

    return habitat_sim.Configuration(sim_cfg, [agent_cfg])

def set_camera_pose(sim, pose):
    # initialize agent_state
    agent_state = habitat_sim.AgentState()
    agent_state.position = pose.translation
    agent_state.rotation = pose.quaternion

    # update it
    agent = sim.get_agent(0)
    agent.set_state(agent_state)

def display_images(observations):
    # Convert raw images to something displayable (and store in processed_imgs).
    # At the same time, collect figure titles.

    titles = []
    processed_imgs = []

    if "color" in observations:
        rgb_img = Image.fromarray(observations["color"], mode="RGBA")
        processed_imgs.append(rgb_img)
        titles.append("RGB")

    if "semantic" in observations:
        raw_img = observations["semantic"]
        semantic_img = Image.new("P", (raw_img.shape[1], raw_img.shape[0]))
        semantic_img.putpalette(d3_40_colors_rgb.flatten())
        semantic_img.putdata((raw_img.flatten() % 40).astype(np.uint8))
        semantic_img = semantic_img.convert("RGBA")
        processed_imgs.append(semantic_img)
        titles.append("Semantic")

    if "depth" in observations:
        depth_img = Image.fromarray((observations["depth"] / 10 * 255).astype(np.uint8), mode="L")
        processed_imgs.append(depth_img)
        titles.append("Depth")

    # Actually create the figure
    fig, axes = plt.subplots(nrows=1, ncols=len(titles), figsize=(12, 8))
    for i, img in enumerate(processed_imgs):
        axes[i].axis("off")
        axes[i].set_title(titles[i])
        axes[i].imshow(img)

    plt.show()

scene_file = "replica-data/apartment_0/habitat/mesh_semantic.ply"
settings = {
    "image_width": 512,  # Spatial resolution of the cameras    
    "image_height": 512,
    "physics_config_file": "data/default.phys_scene_config.json",
    "dummy_scene_file": "replica-data/apartment_0/habitat/mesh_semantic.ply"
}
cfg = make_habitat_cfg(settings, settings["dummy_scene_file"], image_types={})  # broken
#cfg = make_habitat_cfg(settings, settings["dummy_scene_file"], image_types={"color":True, "semantic":True})   # works

sim = habitat_sim.Simulator(cfg)
random.seed(0)
sim.seed(0)
sim.initialize_agent(0)
        
cfg2 = make_habitat_cfg(settings, scene_file, image_types={"color":True, "semantic":True})
sim.reconfigure(cfg2)
id = sim.add_object(0)
sim.set_translation(np.array([-1.9, -0.9, 6]), id)

pose = Pose(np.array([0.5, -2.4, 5.0]), 
                    common.quat_from_coeffs(np.array([0, 0.77301, 0, 0.634393])))
set_camera_pose(sim, pose)
observations = sim.get_sensor_observations()
display_images(observations)

aclegg3, 2019-10-08:
Ok, I am able to reproduce this issue. I'll take a look, thanks for reporting this!

aclegg3, 2019-10-09:
So the tiny patch in PR #287 should fix this issue for you.

dhuber23b, 2019-10-09:
I tested this out, and also the other patch, and together, they both enable
my code to work again.  Thanks!

closed 2019-10-09 by aclegg3