facebookresearch/habitat-sim #284: How to create your own scene?

dhuber23b, 2019-10-08:
I need to create a blank scene for use in Habitat.  If that is not possible, I would like to create a scene with only a plane.

What are the requirements for this?  I tried replicating the information found in the replica header, but I get an error:
F1008 14:08:38.440702 53716416 Simulator.cpp:189] Check failed: activeSemanticSceneID_ >= 0 (-1 vs. 0)

Here is the ply file I am using:
ply
format ascii 1.0
comment Created by Blender 2.80 (sub 75) - www.blender.org, source file: 'cone.blend'
element vertex 3
property float x
property float y
property float z
property float nx
property float ny
property float nz
element face 1
property list uchar uint vertex_indices
property uint16 object_id
end_header
0.000000 0.031314 -1.000000 0.087754 0.445488 -0.890977 
0.000000 2.031314 -0.000000 0.087754 0.445488 -0.890977 
0.195090 0.031314 -0.980785 0.087754 0.445488 -0.890977 
3 0 1 2 0

aclegg3, 2019-10-09:
Hey Daniel, you can create an empty scene in habitat by setting the scene filepath (e.g. `sim_cfg.scene.id`) to `"NONE"`. This will not create a semantic scene in current master branch. You can see the branch in PR #285 for a fix and example of the results.

aclegg3, 2019-10-09:
> ... I get an error:
> F1008 14:08:38.440702 53716416 Simulator.cpp:189] Check failed: activeSemanticSceneID_ >= 0 (-1 vs. 0)

This should be fixed by the patch in PR #285.

dhuber23b, 2019-10-09:
Thank you , Alex.  I will give it a try.

closed 2019-11-08 by aclegg3