facebookresearch/habitat-sim #288: Having trouble installing headless

Toraudonn, 2019-10-09:
## ❓ Questions and Help

- I could install `habitat-sim` using `--headless` and `--with-cuda`.
- when I tried the example (`habitat-api/examples/example.py`), it comes up as:

```
❯ python habitat-sim/examples/example.py --scene habitat-sim/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
sim_cfg.physics_config_file = ./data/default.phys_scene_config.json
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1009 20:20:52.853154 23838 WindowlessContext.cpp:99] [EGL] Detected 4 EGL devices
F1009 20:20:52.853183 23838 WindowlessContext.cpp:113] Check failed: eglDevId < numDevices [EGL] Could not find an EGL device for CUDA device 0
*** Check failure stack trace: ***
[1]    23838 abort (core dumped)  python habitat-sim/examples/example.py --scene
```

Toraudonn, 2019-10-09:
Can someone explain to me what this line does?
https://github.com/facebookresearch/habitat-sim/blob/1d4bd7bc8c73f6f4674e6e2c553930eff1a0d606/src/esp/gfx/WindowlessContext.cpp#L113

erikwijmans, 2019-10-10:
That line is checking to make sure that we found a valid CUDA device to use.  In headless mode, habitat-sim will only target CUDA devices(as it needs Nvidia's libEGL to work.  Make sure `CUDA_VISIBLE_DEVICES` is not set to the empty string.  Given that it is detecting 4 devices, maybe the correct libEGL is not being picked up.  Make sure `libEGL_nvidia.so` can be found, i.e.  run `ldconfig -N -v | grep libEGL`.  If it doesn't show up, then try `locate libEGL_nvidia.so` and add the path via `export LD_LIBRARY_PATH=/path/to:${LD_LIBRARY_PATH}`

Toraudonn, 2019-10-11:
Thanks for the response!
I was using nvidia's docker image ([nvidia/cuda](https://hub.docker.com/r/nvidia/cuda/)) that [doesn't support EGL by default](https://github.com/NVIDIA/nvidia-docker/wiki/Frequently-Asked-Questions#is-opengl-supported), and I couldn't get the correct libEGL to pick up, so I've decided to follow your docker implementation and use [nvidia/cudagl](https://hub.docker.com/r/nvidia/cudagl) as my base and it worked flawlessly.

closed 2019-10-11 by Toraudonn

knzo25, 2019-11-06:
I know this issue is marked as closed, but I have the same problem in a non-docker environment

I have installed the `libglvnd` package and the nvidia drivers.
`libEGL_nvidia.so` exists in `/usr/lib/x86_64-linux-gnu/` and is found by `ldconfig`.

The package links against `libEGL.so.1.0.0` which according to what I understand, makes `libglvnd` pick the correct vendor implementarion

Any ideas would be highly appreciated ! :)

erikwijmans, 2019-11-06:
What does `nvidia-smi` report?

knzo25, 2019-11-09:
@erikwijmans, apologies for the delay.
`nvidia-smi` works as expected.

Since the op had the same issue, I think this may happen to more people than me.
After checking how `libglvnd` worked I realized that even if I had `libglvnd-dev` and `libnvidia-gl` packages, the file `10_nvidia.json` was not in the usual folder, making it impossible for `libglvnd` to pick the correct implementation.

A fix would be to paste the aforementioned file in the `/share` folder, or if you cannot modify that folder because of lack of permissions (as is my case), you can set the `__EGL_VENDOR_LIBRARY_FILENAMES` variable to point to the json file.

smorad, 2020-02-28:
A simpler description of the @knzo25 fix:

Download https://github.com/openai/mujoco-py/blob/master/vendor/10_nvidia.json into /usr/share/glvnd/egl_vendor.d/10_nvidia.json

jeffjunzhang, 2020-05-31:
> A simpler description of the @knzo25 fix:
> 
> Download https://github.com/openai/mujoco-py/blob/master/vendor/10_nvidia.json into `/usr/share/glvnd/egl_vendor.d/10_nvidia.json`

I tried this but still get the following: 

$ python examples/example.py
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0531 02:31:57.994230  4846 WindowlessContext.cpp:98] [EGL] Detected 5 EGL devices
F0531 02:31:57.994256  4846 WindowlessContext.cpp:112] Check failed: eglDevId < numDevices [EGL] Could not find an EGL device for CUDA device 0
*** Check failure stack trace: ***
Aborted (core dumped)

erikwijmans, 2020-05-31:
Hi, can you please also make sure `libEGL_nvidia.so` can be found by the dynamic linker?

jeffjunzhang, 2020-05-31:
Hi @erikwijmans, thanks for your reply.
Here are what I'm doing, could you let me know if I'm missing anything?

(1) I use ldconfig to check the libEGL:
```
$ ldconfig -N -v | grep libEGL
/sbin/ldconfig.real: Can't stat /usr/local/lib/i386-linux-gnu: No such file or directory
/sbin/ldconfig.real: Can't stat /usr/local/lib/i686-linux-gnu: No such file or directory
/sbin/ldconfig.real: Can't stat /lib/i686-linux-gnu: No such file or directory
/sbin/ldconfig.real: Can't stat /usr/lib/i686-linux-gnu: No such file or directory
/sbin/ldconfig.real: Can't stat /usr/local/lib/x86_64-linux-gnu: No such file or directory
/sbin/ldconfig.real: Path `/lib/x86_64-linux-gnu' given more than once
/sbin/ldconfig.real: Path `/usr/lib/x86_64-linux-gnu' given more than once
/sbin/ldconfig.real: /lib/i386-linux-gnu/ld-2.27.so is the dynamic linker, ignoring

        libEGL_nvidia.so.0 -> libEGL_nvidia.so.440.44
/sbin/ldconfig.real: /lib/x86_64-linux-gnu/ld-2.27.so is the dynamic linker, ignoring

/sbin/ldconfig.real: Cannot stat /usr/lib/x86_64-linux-gnu/libnvjpeg.so: No such file or directory
        libEGL.so.1 -> libEGL.so.1.0.0
        libEGL_mesa.so.0 -> libEGL_mesa.so.0.0.0
        libEGL_nvidia.so.0 -> libEGL_nvidia.so.440.44
```

(2) Also trying to locate libegl:
```
$ locate libEGL_nvidia.so
/usr/lib/i386-linux-gnu/libEGL_nvidia.so.0
/usr/lib/i386-linux-gnu/libEGL_nvidia.so.440.44
/usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.0
/usr/lib/x86_64-linux-gnu/libEGL_nvidia.so.440.44
```
(3) Then I set the LD_LIBRARY
`$ export LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu:/usr/lib/i386-linux-gnu:${LD_LIBRARY_PATH}`
```
$ echo $LD_LIBRARY_PATH
/usr/lib/x86_64-linux-gnu:/usr/lib/i386-linux-gnu:
```

erikwijmans, 2020-05-31:
Hmm, it is strange seeing that in `x86_64-linux-gnu`, with the standard install, it'll be in a folder like `/path/to/nvidia-<driver_version>`. Typically `/usr/lib/nvidia-<driver_version>/`.  If you didn't setup this machine yourself, can you check with however did to see if they modified where the standard driver install would put things?

Another question, it's showing `5 EGL devices`, that would make sense for a 4 or 5 GPU machine, so if this isn't a 4 or 5 GPU machine, that also points to something funky.

jeffjunzhang, 2020-06-02:
Hi @erikwijmans yes, we do have 4 GPUs on this machine, it's running Ubuntu 18.04.

I also checked with the system admin and get some info back:
The Nvidia driver version is 440.44 (via nvidia-smi command)
`root@lambda-quad:/usr# nvidia-smi`
Mon Jun  1 14:32:09 2020
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.44       Driver Version: 440.44       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+

The drivers are installed through the CUDA toolkit. 
Our sys admin did not find the `nvidia-<driver_version>` directory path.
**Is this the standard way to install?** 

```
root@lambda-quad:/usr# lsmod | grep -i nvidia
nvidia_uvm            929792  4
nvidia_drm             45056  6
nvidia_modeset       1110016  6 nvidia_drm
nvidia              20357120  744 nvidia_uvm,nvidia_modeset
drm_kms_helper        172032  1 nvidia_drm
drm                   401408  9 drm_kms_helper,nvidia_drm
ipmi_msghandler        53248  2 ipmi_devintf,nvidia
root@lambda-quad:/usr#

root@lambda-quad:/usr# dpkg -L nvidia-driver-440
/.
/usr
/usr/share
/usr/share/doc
/usr/share/doc/nvidia-driver-440
```

erikwijmans, 2020-06-02:
Hmm, I haven't had issue with cuda-drivers from apt.  Can you also check to see if libGLdispatch is being picked up correctly, i.e.

`ldd $(python -c "import habitat_sim; print(habitat_sim._ext.habitat_sim_bindings.__file__)")` should have something like `libGLdispatch.so.0 => /usr/lib/nvidia-440/libGLdispatch.so.0` in the output.

jeffjunzhang, 2020-06-02:
@erikwijmans 
Are you suggesting we should install the cuda-drivers directly from apt?

Also, It's strange to get errors even when I even try to `import habitat_sim` in python
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/jz2163/NeuralSLAM/habitat-sim/habitat_sim/__init__.py", line 12, in <module>
    from .nav import *
  File "/home/jz2163/NeuralSLAM/habitat-sim/habitat_sim/nav/__init__.py", line 1, in <module>
    from .greedy_geodesic_follower import GreedyGeodesicFollower
  File "/home/jz2163/NeuralSLAM/habitat-sim/habitat_sim/nav/greedy_geodesic_follower.py", line 6, in <module>
    import habitat_sim.agent
  File "/home/jz2163/NeuralSLAM/habitat-sim/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/home/jz2163/NeuralSLAM/habitat-sim/habitat_sim/agent/agent.py", line 14, in <module>
    import habitat_sim.bindings as hsim
  File "/home/jz2163/NeuralSLAM/habitat-sim/habitat_sim/bindings/__init__.py", line 24, in <module>
    from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
ModuleNotFoundError: No module named 'habitat_sim._ext'
```

I followed instructions to install the Habitat-Sim from [https://github.com/devendrachaplot/Neural-SLAM]]

```git clone https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim; git checkout 9575dcd45fe6f55d2a44043833af08972a7895a9; 
pip install -r requirements.txt; 
python setup.py install --headless
```

The logs from the installation are:

```
running install
running bdist_egg
running egg_info
creating habitat_sim.egg-info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-3.7
creating build/lib.linux-x86_64-3.7/habitat_sim
copying habitat_sim/utils.py -> build/lib.linux-x86_64-3.7/habitat_sim
copying habitat_sim/errors.py -> build/lib.linux-x86_64-3.7/habitat_sim
copying habitat_sim/__init__.py -> build/lib.linux-x86_64-3.7/habitat_sim
copying habitat_sim/simulator.py -> build/lib.linux-x86_64-3.7/habitat_sim
copying habitat_sim/logging.py -> build/lib.linux-x86_64-3.7/habitat_sim
creating build/lib.linux-x86_64-3.7/habitat_sim/nav
copying habitat_sim/nav/__init__.py -> build/lib.linux-x86_64-3.7/habitat_sim/nav
copying habitat_sim/nav/greedy_geodesic_follower.py -> build/lib.linux-x86_64-3.7/habitat_sim/nav
creating build/lib.linux-x86_64-3.7/habitat_sim/agent
copying habitat_sim/agent/__init__.py -> build/lib.linux-x86_64-3.7/habitat_sim/agent
copying habitat_sim/agent/agent.py -> build/lib.linux-x86_64-3.7/habitat_sim/agent
creating build/lib.linux-x86_64-3.7/habitat_sim/sensors
copying habitat_sim/sensors/__init__.py -> build/lib.linux-x86_64-3.7/habitat_sim/sensors
copying habitat_sim/sensors/sensor_suite.py -> build/lib.linux-x86_64-3.7/habitat_sim/sensors
creating build/lib.linux-x86_64-3.7/habitat_sim/bindings
copying habitat_sim/bindings/__init__.py -> build/lib.linux-x86_64-3.7/habitat_sim/bindings
creating build/lib.linux-x86_64-3.7/habitat_sim/agent/controls
copying habitat_sim/agent/controls/pyrobot_noisy_controls.py -> build/lib.linux-x86_64-3.7/habitat_sim/agent/controls
copying habitat_sim/agent/controls/__init__.py -> build/lib.linux-x86_64-3.7/habitat_sim/agent/controls
copying habitat_sim/agent/controls/controls.py -> build/lib.linux-x86_64-3.7/habitat_sim/agent/controls
[... 1029 lines cut ...]
Adding setuptools 41.2.0 to easy-install.pth file
Installing easy_install script to /home/jz2163/NeuralSLAM/py37-venv/bin
Installing easy_install-3.6 script to /home/jz2163/NeuralSLAM/py37-venv/bin

Using /home/jz2163/NeuralSLAM/py37-venv/lib/python3.7/site-packages
Searching for llvmlite==0.32.1
Best match: llvmlite 0.32.1
Adding llvmlite 0.32.1 to easy-install.pth file

Using /home/jz2163/NeuralSLAM/py37-venv/lib/python3.7/site-packages
Finished processing dependencies for habitat-sim==0.1.2
Processing ./build/deps/magnum-bindings/src/python
Building wheels for collected packages: magnum
  Building wheel for magnum (setup.py): started
  Building wheel for magnum (setup.py): finished with status 'done'
  Created wheel for magnum: filename=magnum-0.0.0-cp37-cp37m-linux_x86_64.whl size=1310218 sha256=86a96ed6ac96c3d0d5111c359192c63d80a683bd09eca5edbfa31b3d6bb1d8d1
  Stored in directory: /tmp/pip-ephem-wheel-cache-nucjj5md/wheels/df/b7/20/e3c7f890190a13a97a992848fbf819dcc7f06a306d30d46367
Successfully built magnum
Installing collected packages: magnum
  Found existing installation: magnum 0.0.0
    Uninstalling magnum-0.0.0:
      Successfully uninstalled magnum-0.0.0
Successfully installed magnum-0.0.0

```

erikwijmans, 2020-06-02:
You'll need to run that command _not_ in the repo path, i.e. `/path/to/habitat-sim`

jeffjunzhang, 2020-06-03:
@erikwijmans thanks. Now I'm able to get the info:

```
(py37-venv) jz2163@lambda-quad:~/NeuralSLAM$ ldd $(python -c "import habitat_sim; print(habitat_sim._ext.habitat_sim_bindings.__file__)")
	linux-vdso.so.1 (0x00007ffd1a5ae000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fda9fb37000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fda9f92f000)
	libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007fda9f6f8000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fda9f4d9000)
	libOpenGL.so.0 => /usr/lib/x86_64-linux-gnu/libOpenGL.so.0 (0x00007fda9f2ab000)
	libEGL.so.1 => /usr/lib/x86_64-linux-gnu/libEGL.so.1 (0x00007fda9f097000)
	libCorradePluginManager.so.2 => /home/jz2163/NeuralSLAM/habitat-sim/build/lib.linux-x86_64-3.7/habitat_sim/_ext/libCorradePluginManager.so.2 (0x00007fda9ee85000)
	libCorradeUtility.so.2 => /home/jz2163/NeuralSLAM/habitat-sim/build/lib.linux-x86_64-3.7/habitat_sim/_ext/libCorradeUtility.so.2 (0x00007fda9ec35000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fda9e855000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fda9e4b7000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fda9e29f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fda9deae000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fdaa09d1000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fda9dcaa000)
	libGLdispatch.so.0 => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007fda9d9f4000)

```

erikwijmans, 2020-06-03:
Unfortunately I am out of ideas currently.  You can try docker with a cudagl base as a last resort.

s-gupta, 2020-06-05:
I am facing the same issue as described here. Except the odd thing is that I have an older install of habitat, that works perfectly fine on this machine. So I guess, its a compile time issue? I will appreciate any hints on this.

erikwijmans, 2020-06-05:
@s-gupta For the version that works, what does the logging say about EGL/OpenGL?  You should see something like this in the outputs:

```
I0605 19:30:12.134407 41004 WindowlessContext.cpp:100] [EGL] Detected 8 EGL devices
I0605 19:30:12.496650 41004 WindowlessContext.cpp:121] [EGL] Selected EGL device 0 for CUDA device 0
I0605 19:30:12.508350 41004 WindowlessContext.cpp:135] [EGL] Version: 1.5
I0605 19:30:12.508415 41004 WindowlessContext.cpp:136] [EGL] Vendor: NVIDIA
Renderer: TITAN Xp/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 440.33.01
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
```

erikwijmans, 2020-06-07:
@jeffjunzhang What does `ls -l /usr/lib/x86_64-linux-gnu/libEGL.so.1` return?  I am thinking that since you also have `libEGL_mesa`, if the mesa install was after the Nvidia install, the mesa install may have overwritten Nvidia's `libEGL.so.1` -- there is an Nvidia version of `libEGL.so.1` that needs to be there for things to be picked up correctly.

jeffjunzhang, 2020-06-07:
@erikwijmans It return:
```
$ ls -l /usr/lib/x86_64-linux-gnu/libEGL.so.1
lrwxrwxrwx 1 root root 15 May 10  2019 /usr/lib/x86_64-linux-gnu/libEGL.so.1 -> libEGL.so.1.0.0
```
Should I reinstall the Nvidia driver through apt?

erikwijmans, 2020-06-07:
Yeah, that looks like it is linking to mesa's/nvidia's got overwritten.  You should re-install however you would install normally.

abhiskk, 2020-06-09:
Updating the nvidia drivers resolved this issue for me.

MeisterNewman, 2020-06-09:
I'm having the same problem: upon running, I get 
`I0609 14:59:26.919481  4712 WindowlessContext.cpp:100] [EGL] Detected 3 EGL devices
F0609 14:59:26.919498  4712 WindowlessContext.cpp:114] Check failed: eglDevId < numDevices [EGL] Could not find an EGL device for CUDA device 0`
Notably, the only graphics devices on this machine are one Nvidia GPU and an Intel iGPU (the Nvidia GPU is GPU 0 in nvidia-smi).

After I install the mesa packages, I run 
`ldd $(python3 -c "import habitat_sim; print(habitat_sim._ext.habitat_sim_bindings.__file__)")`
and get
`        linux-vdso.so.1 (0x00007ffcee991000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fc3a238b000)
        libgomp.so.1 => /lib/x86_64-linux-gnu/libgomp.so.1 (0x00007fc3a2349000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fc3a2326000)
        libOpenGL.so.0 => /lib/x86_64-linux-gnu/libOpenGL.so.0 (0x00007fc3a22fa000)
        libEGL.so.1 => /lib/x86_64-linux-gnu/libEGL.so.1 (0x00007fc3a22e5000)
        libcudart.so.10.2 => /usr/local/cuda/targets/x86_64-linux/lib/libcudart.so.10.2 (0x00007fc3a2065000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fc3a205f000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fc3a1e7e000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fc3a1d2f000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fc3a1d14000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fc3a1b22000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fc3a306e000)
        libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007fc3a1a6a000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fc3a1a5d000)`
which seems to suggest that the mesa drivers are getting overwritten in my case as well.
CUDA_VISIBLE_DEVICES was empty, but changing it to be equal to 0 made no difference. NV_GPU was empty. nvidia-smi is normal. All of my EGL .so files are in /usr/lib/x86_64-linux-gnu/, and are as follows:
`libEGL.so
libEGL.so.1
libEGL.so.1.1.0
libEGL_mesa.so.0
libEGL_mesa.so.0.0.0
libEGL_nvidia.so.0
libEGL_nvidia.so.440.82`

 Is there a natural way to solve this without updating the Nvidia drivers after installing the various mesa packages?

erikwijmans, 2020-06-09:
Is it possible to share a minimum version of the dockerfile?

MeisterNewman, 2020-06-09:
Unfortunately, I'm not permitted to do so. However, I'd be happy to collect any specific information from it and attach it.

erikwijmans, 2020-06-09:
What's the base container, what's the linux flavor/version, what CUDA version, what Nvidia driver version?

MeisterNewman, 2020-06-09:
The container is running Ubuntu 20.04. The Nvidia drivers are v440.82, with CUDA 10.2. What do you mean by "base container"? Sorry, I'm somewhat new to Docker.

erikwijmans, 2020-06-09:
Can you try adding in the install from here: https://gitlab.com/nvidia/container-images/opengl/blob/ubuntu18.04/glvnd/devel/Dockerfile

MeisterNewman, 2020-06-09:
apt-get was unwilling to install the i386 packages (I'm unclear on why, exactly) so I removed them (as my system is 64-bit). It still does not work, failing with the same error (`F0609 20:08:29.386997  1778 WindowlessContext.cpp:114] Check failed: eglDevId < numDevices [EGL] Could not find an EGL device for CUDA device 0`) as before. Thoughts?

erikwijmans, 2020-06-10:
For that base, I needed to mount my external Nvidia libs like `-v /usr/lib/nvidia-440:/usr/lib/nvidia-440 -v /usr/share/glvnd/egl_vendor.d:/usr/share/glvnd/egl_vendor.d`.

I have always had much better luck using `nvidia/cudagl` as the base, i.e. `nvidia/cudagl:10.2-devel-ubuntu18.04`.  On my host system (ubuntu 16.04, nvidia drivers are v440.33.01), that base works with issues and no weird mounting needed.

small-zeng, 2021-05-08:
> @erikwijmans, apologies for the delay.
> `nvidia-smi` works as expected.
>
> Since the op had the same issue, I think this may happen to more people than me.
> After checking how `libglvnd` worked I [...]

Thanks, I install libnvidia-gl  in  lxc container and succeed.

small-zeng, 2021-05-08:
> Thanks, I install libnvidia-gl in lxc container and succeed.

sudo apt-get install  libnvidia-gl-460

small-zeng, 2021-05-08:
> @erikwijmans, apologies for the delay.
> `nvidia-smi` works as expected.
>
> Since the op had the same issue, I think this may happen to more people than me.
> After checking how `libglvnd` worked I [...]

sudo apt-get install  libnvidia-gl-460

ruddradev, 2021-05-19:
Hi All,

I have been having this issue for the past 2 weeks and trying the suggestions in this page, I have made some progress, but would still need some help. 

I am using the docker image - nvcr.io/nvidia/pytorch:20.03-py3

And installing the below dependencies - 

`apt-get install -y libglvnd-dev libnvidia-gl-450 libgl1-mesa-dev libegl1-mesa-dev libglm-dev libjpeg-dev libpng-dev libglfw3-dev build-essential cmake`

When I left out libnvidia-gl-450, I would get the error mentioned at the start of the page, i.e. - `Check failed: eglDevId < numDevices [EGL] Could not find an EGL device for CUDA device 0`

After installing libnvidia-gl-450, I got the error - `Check failed: numDevices > 0 [EGL] No devices detected`

I saw that we needed to download the file `10_nvidia.json` and I tried putting it in the path mentioned `/usr/share/glvnd/egl_vendor.d` but got the same error. I could not try installing the i386 versions mentioned above, and I don't have access to run the docker with the mount instructions mentioned above. Is there something I can try that might fix this issue? Thank you all for helping.

Nvidia Driver version - 450.80.02, 
nvcc --version gives cuda 10.2
nvidia-smi gives cuda 11.0

erikwijmans, 2021-05-19:
I've only really had success with docker when using https://hub.docker.com/r/nvidia/cudagl as the base container so I'd start with that base if possible.

I'm not quite sure on this, but my understanding is that you need Nvidia's OpenGL libs to match perfectly with the driver version (even the minor version) so you should make sure you are pulling the 450.80.02 version of libnvidia-gl-450, which you wouldn't be by default as 450.119.04 is the most recent version of 450.  Ideally these libraries would just be provided to the container from the host system, if you can talk to the admins about why that aren't that would likely be good.

ruddradev, 2021-05-21:
Hello @erikwijmans ,

Thanks a lot for the docker image. I am now running the image you specified and the issue mentioned above is not appearing anymore. However a new issue that is part of another thread has come up so I will post there.

https://github.com/facebookresearch/habitat-lab/issues/303

jinxin-zhu, 2023-08-14:
> Here are what I'm doing, could you let me know if I'm missing anything?

`ldconfig -p | grep EGL`
`        libEGL_mesa.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_mesa.so.0
        libEGL.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so.1
        libEGL.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so
`
I seem to be missing the entry for nvidia, how can I add it?

Jaraxxus-Me, 2024-01-29:
> `ldconfig -p | grep EGL`： `
>
> ```
>     libEGL_mesa.so.0 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL_mesa.so.0
>
>     libEGL.so.1 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libEGL.so.1 [...]

I have the same issue, did you resolve this problem?

BlingblinggirlR, 2024-03-28:
> @erikwijmans It return:
> 
> ```
> $ ls -l /usr/lib/x86_64-linux-gnu/libEGL.so.1
> lrwxrwxrwx 1 root root 15 May 10  2019 /usr/lib/x86_64-linux-gnu/libEGL.so.1 -> libEGL.so.1.0.0
> ```
> 
> Should I reinstall the Nvidia driver through apt?

Sorry to bother you, but I had the same problem, how did you solve it in the end?
thanks for your reply!

francescotaioli, 2024-05-10:
@haruishi43 I was able to run habitat with docker using the official [Nvidia docker images](https://hub.docker.com/r/nvidia/cuda/tags), specifying at runtime these params:
```docker run --runtime=nvidia -e NVIDIA_DRIVER_CAPABILITIES=all  -e "DISPLAY=$DISPLAY" -v /tmp/.X11-unix:/tmp/.X11-unix <your_image_name>```

linukc, 2025-03-30:
I feel myself as a best hacker in the world after successfully solving this riddle 😆 , thanks to all the related comments.
For future folks my solution was:
1) download .so files as in https://github.com/facebookresearch/habitat-sim/issues/1511#issuecomment-1537221109
2) symlink **ONLY** libEGL_nvidia.so (i had others)
3) create vendor json https://github.com/facebookresearch/habitat-sim/issues/2424#issuecomment-2692332100
4) run `__EGL_VENDOR_LIBRARY_FILENAMES=<arbitry_path>/10_nvidia.json python your_script.py`