facebookresearch/habitat-sim #295: Cannot import habitat_sim from python shell

dhuber23b, 2019-10-11:
I am not sure if this is a bug or a feature. 

I am able to run the command "import habitat_sim" from a script or an ipython notebook, but if I just start python and import it manually, I get an error message (see below).  I think it has something to do with builtins and the command "builtins.__HSIM_SETUP__ = True".   I don't know enough about how __init__.py is supposed to work to be able to comment further.

venv) dhuber-mbp:habitat-sim dhuber$ python3
Python 3.6.2+ (heads/master-dirty:404de642c0, Sep 15 2017, 15:52:01)
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.37)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import habitat_sim
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/dhuber/Dropbox/projects/ar-occlusion/habitat-sim/habitat_sim/__init__.py", line 14, in <module>
    from habitat_sim.nav import *
  File "/Users/dhuber/Dropbox/projects/ar-occlusion/habitat-sim/habitat_sim/nav/__init__.py", line 1, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (
ModuleNotFoundError: No module named 'habitat_sim._ext'
>>> from collections import namedtuple
import habitat_sim>>>
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/dhuber/Dropbox/projects/ar-occlusion/habitat-sim/habitat_sim/__init__.py", line 14, in <module>
    from habitat_sim.nav import *
  File "/Users/dhuber/Dropbox/projects/ar-occlusion/habitat-sim/habitat_sim/nav/__init__.py", line 1, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (
ModuleNotFoundError: No module named 'habitat_sim._ext'

aclegg3, 2019-10-11:
This looks like Habitat-sim bindings are not on the python path. Try building with build.sh and check that the habitat-sim root directory is in the path for python.

dhuber23b, 2019-10-11:
I figured out the problem.  Strangely, it is just the opposite of not being in the python path.  Specifically, I think that running setup.py inside the virtual environment must set up habitat_sim so that it is in the path regardless of your location.  I didn't realize this and added a symlink to habitat_sim from a directory in my python path.  For some reason, that leads to this error.  

Problem is solved.

closed 2019-11-13 by mathfac

GELIELEO, 2019-11-28:
> I figured out the problem. Strangely, it is just the opposite of not being in the python path. Specifically, I think that running setup.py inside the virtual environment must set up habitat_sim so [...]

I can not understand what your silution is, can you explain specificlally, thx