facebookresearch/habitat-sim #300: Make sim.reset reset agent and simulation to after reconfigure state

mathfac, 2019-10-15:
## 🚀 Feature

Make sim.reset reset an agent position and Simulator to `after reconfigure state`.

## Motivation
Now [Simulator.reset()](https://github.com/facebookresearch/habitat-sim/blob/master/src/esp/gfx/Simulator.cpp#L165) does nothing, while user expect it to reset to initial state that the Simulator was configured with. 

## Pitch

## Alternatives
Similar as it is [done](https://github.com/facebookresearch/habitat-api/blob/master/habitat/sims/habitat_simulator/habitat_simulator.py#L420) in `habitat-api` now we can manually set the original agent state and get observations:
```
        for agent_id, _ in enumerate(self.config.AGENTS):
            agent_cfg = self._get_agent_config(agent_id)
            if agent_cfg.IS_SET_START_STATE:
                self.set_agent_state(
                    agent_cfg.START_POSITION,
                    agent_cfg.START_ROTATION,
                    agent_id,
                )

        return self._sim.get_sensor_observations()

```

## Additional context
Issue was surfaced by @piotr-bojanowski.

erikwijmans, 2019-10-15:
The agent config doesn't contain any initial position information  for habitat-sim: https://aihabitat.org/docs/habitat-sim/habitat_sim.agent.AgentConfiguration.html  (which is why this isn't done).

mathfac, 2020-01-09:
That makes sense for C++ part of sim, but Python part of Habitat-SIM has empty reset function while it should call [reconfigure](https://github.com/facebookresearch/habitat-sim/blob/823111e431df78bdb8c1224816912eb5aa74a08a/habitat_sim/simulator.py#L88) with original config.

erikwijmans, 2020-01-10:
A `reconfigure` also doesn't set the agent back to the initial state as the configuration doesn't contain that information (and unless the config changed, a reconfigure is a no-op).  We currently don't have a notion of an initial agent state and the agent is simply given a random state upon creation.

mathfac, 2020-01-10:
@erikwijmans, looks like `reconfigure ` is setting agent position [here](https://github.com/facebookresearch/habitat-sim/blob/823111e431df78bdb8c1224816912eb5aa74a08a/habitat_sim/simulator.py#L167).

mathfac, 2020-01-27:
Solved by @mpiseno

closed 2020-01-27 by mathfac