facebookresearch/habitat-sim #301: Bullet collision objects for rigid bodies with mesh hierarchies are incorrect.

aclegg3, 2019-10-15:
## 🐛 Bug

The current implementation creates rigid objects from loaded meshes by directly indexing into asset memory. Meshes are shifted to align object center of mass with origin (as required by Bullet physics) and then consumed directly to create collision objects. This works for single mesh objects (such as our test dataset). However, if the loaded object has a hierarchy of meshes including transforms (such as the Wayfair dataset), we should build a compound collision object encapsulating this hierarchy. 

Note: rendering and kinematics work, this bug only concerns the underlying physical collision mesh used for simulation.

## Command

`./build/viewer --enable-physics --debug-bullet -- "NONE" `
## To Reproduce

Steps to reproduce the behavior:

1. Pull the `bullet-debug-draw` branch (to get Bullet debug visualization) #298 
2. Add objects (such as wayfair objects) with a mesh transformation hierarchy to the `default.phys_scene_config.json` file.
2. Modify viewer.cpp to set objects to KINEMATIC upon creation:
```
void Viewer::addObject(std::string configFile) {
...
    physicsManager_->setObjectMotionType(physObjectID, physics::MotionType::KINEMATIC);
...
}
```
3. `./build/viewer --enable-physics --debug-bullet -- "NONE"`
4. Add objects with 'o' and remove with 'u'

Results show bullet collision meshes (white wireframes) do not match rendered objects:
[image]

## Expected behavior

Collision object meshes should align with rendered object meshes.

## Additional context

Relevant redesign and solution is in progress.

## System Info

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/master/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux):
 - How you installed PyTorch (`conda`, `pip`, source):
 - Build command you used (if compiling from source):
 - Python version:
 - GPU models and configuration:
 - CUDA version:
 - PIP/conda dependencies packages versions
 - Any other relevant information:

mosra, 2019-10-16:
> Meshes are shifted to align object center of mass with origin (as required by Bullet physics)

This is kinda sad (I couldn't believe it at first, had to double-check). All posts about this I found are from ~2011, one saying that Bullet 3 should have this fixed, others suggesting using `btCompoundShape` (and the overhead coming from it) to work around this. Did really no improvement happen to this since then? :D

aclegg3, 2019-10-16:
> This is kinda sad (I couldn't believe it at first, had to double-check). All posts about this I found are from ~2011, one saying that Bullet 3 should have this fixed, others suggesting using [...]

My understanding is that the collision object itself (the compound shape or single collision mesh) is expected to have `origin == COM`. Partly this is due to the relationship between linear and angular terms as the COM will be the point around which a rigid body rotates, so it should also be the center of rotation for the object's local frame. It is possible that Bullet could have provided a method for setting the COM and handling the coordinate frame transforms internally, but instead that construction falls on the user (maybe this could be considered an opportunity to take advantage of pre-processing speedups?). For us, this means either shifting the mesh or applying transforms in the collision object hierarchy. There will be some overhead (compared to baking the meshes into a single collision shape), but it should not be much compared to the already costly mesh/mesh collision.

closed 2020-01-08 by aclegg3