facebookresearch/habitat-sim #302: Question about texture of inserted objects

dhuber23b, 2019-10-15:
I have a model of a chair, and I am using the object insertion capability to add the model to a test scene.  The texture of the synthetic object does not look very good.  Basically, there is almost no texture at all.  I am wondering if I am doing something wrong.  Maybe I need to add a lighting setting or something?  Attached is a screenshot of the glb file in an online viewer and the image generated by Habitat for the same object.

[image]
[image]

dhuber23b, 2019-10-15:
I also noticed that the feet on the chair are not being rendered.

aclegg3, 2019-10-15:
Hey Daniel, you are correct that there is a missing shader issue here. We have a feature in the works to address this. PR #152 shows the current state of that effort, but we intend to improve on this to support additional shaders and settings in the coming months. For now, if you need non-trivial shading via illumination you can check out some of the changes in the `clutter_nav` branch. 

Specifically: src/esp/gfx/GenericDrawable.cpp shows lighting setup in the shader draw.

Happy to chat more about this and help you get a working solution while we get this feature locked down.

aclegg3, 2019-10-22:
Related to Issue #319

aclegg3, 2019-12-05:
I have a new prototype branch to address this issue: [phong_shaded_physics_objects](https://github.com/facebookresearch/habitat-sim/tree/phong_shaded_physics_objects). With this branch all scenes are flat shaded via textures (as in master), but objects you add with PhysicsManager are rendered with a Phong illumination shader. The default setup is 8 lights placed at the scene bounding box corners. You can also set the position, color, and intensity of the lights in python (see [example of this functionality](https://github.com/facebookresearch/habitat-sim/blob/4cac4a88d71eddc998d1c0da24f960dc6410e484/examples/demo_runner.py#L226)). As a note, adjusting light intensities [0,1] for all 8 lights is likely the best course of action to immediately use this functionality for new scenes (default is 0.2).

A more complete implementation to come, but this should do what you need presently. Let me know how it goes!

AwokeKnowing, 2020-01-28:
I think this makes a lot of sense. It seems the general static 'environment'  should be either fully flat or fully shaded, regardless of if it's loaded 'in pieces' (adding chairs).  But at the physics layer, with dynamic rotation, having shading makes sense, regardless of whether the environment is flat or shaded.

I wonder if some optimization could be done where the object's physics are updated more often than the shading.  This could allow a bit more advanced shading but I don't know what it would look like.

PR #500 (2020-02-26): Expose lighting setup and individual object lighting manipulation
## Motivation and Context

This PR builds on #489 and allows manipulation of light setups through the simulator.

Fixes #302.
Contributes to #332, #361, #319, #315.

Through the simulator, the user can now:
* select a light setup to use when adding an object
* change the light setup a single object uses
* create new light setups
* change an existing light setup, updating all current objects using that setup

For changing the setup for an individual entity, we currently have to traverse the subtree of that object and dynamic cast [... cut]
closed 2020-02-26 by matthewjmay