facebookresearch/habitat-sim #315: PLY import fails to load texture.

tlalexander, 2019-10-18:
## 🐛 Bug

When loading a PLY file as a scene, I expect the texture from the PLY file to load. Instead I get a view of only black or white pixels with no texture. The file shows texture in Meshlab and CloudCompare.

## Command

./build/viewer /path/to/scene.ply

## To Reproduce

Download the file I am using:
https://drive.google.com/open?id=1vEeRPTJunJ_ck0W1aFvzfh4tHoWP4Ecr

Run it in the viewer with the command above. You should see a viewer window with only black and white pixels and no texture.

(Note about the file: This file looks weird. It is a hastily resampled version of a larger mesh I am using. However the larger version and this resampled version exhibit the same behavior, and its easier to share the small version.)

## Expected behavior

I expect the mesh to load with the texture.

## Additional context

I originally generated these PLY files using photogrammetry with COLMAP. Fixing this bug means enabling photogrammetry scene capture to simulation. :)

Also the command line shows some warnings which I've found originate in Magnum. See the full terminal output below:

```
taylor@taylor-desktop:~/Software/habitat-sim$ ./build/viewer examples/colmap/depth/mesh_colmap.ply 
Renderer: Quadro P5000/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 430.26
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-forward-compatible-core-context
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
Trade::AssimpImporter::material(): white ambient detected, forcing back to black
```

aclegg3, 2019-10-22:
Hey, I don't have permissions for the shared file. I've send a request. 😄

aclegg3, 2019-10-22:
Ok, so the problem here is that PLY format with vertex color does not have a predetermined route to create the correct shader  (i.e. we don't expect this use case). Our default shader when expected types are exhausted is flat color shading (white in this case without one provided). 

You can see this [here](https://github.com/facebookresearch/habitat-sim/blob/a72ba43593a57995972ba1521f6f7a20d122761c/src/esp/assets/ResourceManager.cpp#L1113):
```
...
} else {
      // Color-only material
      createDrawable(COLORED_SHADER, mesh, node, drawables, texture,
                     componentID, materials_[materialID]->diffuseColor());
    }
...
```

You can change this to `VERTEX_COLORED_SHADER`:
```
...
} else {
      // Color-only material
      createDrawable(VERTEX_COLORED_SHADER, mesh, node, drawables, texture,
                     componentID, materials_[materialID]->diffuseColor());
    }
...
```

and you will see:
[image]

Upside down, but the color is coming through correctly now. 👍

aclegg3, 2019-10-22:
Related to: Issue #319.

tlalexander, 2019-10-23:
Wonderful, thank you! Yes the original file was upside-down. :)

aclegg3, 2019-10-23:
We are always happy to hear feedback from users! Any thoughts you have about how we can improve the Habitat experience, use cases we may not have considered, feature requests, bug reports, etc... We also welcome contribution. Feel free to read the contributors guidelines and open a PR to help Habitat grow.

We'll be building out our shader configuration support in the coming months and will keep cases like this in mind. Thanks for bringing this issue to our attention! 😄

aclegg3, 2022-03-01:
Per-asset configuration files can now be used to specify the shader type, lighting, and coordinate space transform for stages, scenes, and objects ([doc page](https://aihabitat.org/docs/habitat-sim/attributesJSON.html)).

closed 2022-03-01 by aclegg3