facebookresearch/habitat-sim #316: ModuleNotFoundError: No module named 'habitat_sim._ext' when running example.py

rrwiyatn, 2019-10-21:
## ❓ Questions and Help

Running `python examples/example.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`gives me the following error: https://pastebin.com/3ysgCfGV

mathfac, 2019-10-21:
Hi Rey,

Try to unset or remove habitat_sim folder from PYHTONPATH.

rrwiyatn, 2019-10-22:
Hi Oleksandr,

Thanks for the quick response! I did what you suggested, now I am getting different error:

`ImportError: generic_type: type "SceneNode" referenced unknown base type "Magnum::SceneGraph::Object<Magnum::SceneGraph::BasicTranslationRotationScalingTransformation3D<float> >"`

Full error message: https://pastebin.com/gYrLKu4u

erikwijmans, 2019-10-22:
Interesting.  Seems like it isn't picking the correct version of magnum python bindings.

What does `python -c "import sys; print(sys.path)"` give?

rrwiyatn, 2019-10-23:
`['', '/home/rey/catkin_ws/devel/lib/python2.7/dist-packages', '/opt/ros/melodic/lib/python2.7/dist-packages', '/home/rey/Github/magnum-bindings/build/src/python', '/home/rey/anaconda3/envs/habitat/lib/python36.zip', '/home/rey/anaconda3/envs/habitat/lib/python3.6', '/home/rey/anaconda3/envs/habitat/lib/python3.6/lib-dynload', '/home/rey/.local/lib/python3.6/site-packages', '/home/rey/anaconda3/envs/habitat/lib/python3.6/site-packages', '/home/rey/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.3-py3.6-linux-x86_64.egg']`

erikwijmans, 2019-10-23:
Hmmm, I haven't tried building with magnum manually and it seems likely that things will fall out of sync (new features are getting built out in magnum to support habitat constantly).  Make sure your magnum bindings are build with a commit not too different to the one habitat is expecting.

rrwiyatn, 2019-11-04:
Ah okay thanks!

closed 2019-11-04 by rrwiyatn

JeremyLinky, 2020-11-23:
Hi, I  meet this problem now, can you provide a solution for me?? The error message is as below,
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/chenpeihao/anaconda3/envs/SS/lib/python3.7/site-packages/habitat_sim/__init__.py", line 13, in <module>
    from habitat_sim import (  # noqa: F401
  File "/home/chenpeihao/anaconda3/envs/SS/lib/python3.7/site-packages/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *  # noqa: F401,F403
  File "/home/chenpeihao/anaconda3/envs/SS/lib/python3.7/site-packages/habitat_sim/agent/agent.py", line 14, in <module>
    from habitat_sim import bindings as hsim
  File "/home/chenpeihao/anaconda3/envs/SS/lib/python3.7/site-packages/habitat_sim/bindings/__init__.py", line 29, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (  # noqa: F401 isort:skip
ImportError: generic_type: type "SceneNode" referenced unknown base type "Magnum::SceneGraph::Object<Magnum::SceneGraph::BasicTranslationRotationScalingTransformation3D<float> >"

JeremyLinky, 2020-11-23:
Hi, I meet this problem now! Could you please provide a solution for me? The error message is as belows
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/cc/anaconda3/envs/SS/lib/python3.7/site-packages/habitat_sim/__init__.py", line 13, in <module>
    from habitat_sim import (  # noqa: F401
  File "/home/chenpeihao/anaconda3/envs/SS/lib/python3.7/site-packages/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *  # noqa: F401,F403
  File "/home/cc/anaconda3/envs/SS/lib/python3.7/site-packages/habitat_sim/agent/agent.py", line 14, in <module>
    from habitat_sim import bindings as hsim
  File "/home/cc/anaconda3/envs/SS/lib/python3.7/site-packages/habitat_sim/bindings/__init__.py", line 29, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (  # noqa: F401 isort:skip
ImportError: generic_type: type "SceneNode" referenced unknown base type "Magnum::SceneGraph::Object<Magnum::SceneGraph::BasicTranslationRotationScalingTransformation3D<float> >"

rrwiyatn, 2020-11-24:
@JeremyLinky i ended up just using Docker instead (following the instruction in https://github.com/facebookresearch/habitat-sim#installation)