facebookresearch/habitat-sim #321: inconsistency between semantic scene meshes and the annotation

yz-cnsdqz, 2019-10-29:
Thanks for this fancy platform! 
I encounter two issues that are really confusing to me, and hence need your help!

**(1)**
I am confused by the definition of _centers and dims_ in the scene annotation. For example, after running the demo for habitat-sim, the function print_scene_recur() gives me:

```
/mp3d/17DRP5sb8fy/17DRP5sb8fy_semantic.ply
House has 1 levels, 10 regions and 187 objects
House center:[-2.7928102 1.3372793 -1.5051247] dims:[17.57338 2.9023628 -8.8595495]
Level id:0, center:[-3.157365 1.3372804 -1.5051247], dims:[16.69967 2.9023607 -8.8595495]
Region id:0_0, category:bedroom, center:[-8.821845 1.259409 -2.6915383], dims:[ 4.1633096 2.5356617 -4.207343 ]
Object id:0_0_0, category:wall, center:[-8.86568 1.2817702 -2.73879 ], dims:[2.58148 4.5891 4.59182]
Object id:0_0_1, category:ceiling, center:[-8.91329 2.20326 -2.80575], dims:[4.4761996 4.46008 0.7124357]
Object id:0_0_2, category:misc, center:[-8.69572 1.1633401 -4.2134695], dims:[2.5021195 0.61951023 2.34074 ]
```
According to the docs, the ''dim'' has the type of "habitat_sim.geo.BBox.sizes", so why the BBox size can be negative?

**(2)** When I use open3d to read the entire scene file, e.g. 17DRP5sb8fy_semantic.ply, the vertices of the entire scene are actually bounded within verts_max=[4.88612032 5.46445322 2.78845835] and verts_min=[-11.50337124  -2.92465186  -0.11390185].  This is inconsistent with the room center, which is [-2.7928102 1.3372793 -1.5051247] as above. 

In my project, I would like to extract one room from the entire scene, according to the annotated bounding box. If I am wrong, could you please correct me and give me some hints of how to do it? Many thanks!

erikwijmans, 2019-10-30:
The mp3d meshes having -Z as gravity and habitat uses -Y, so we rotate the mp3d meshes and the coordinates in annotations on load.

This rotation can sometimes causes the dims to be negative, you can either work with the negative values or just just do `.abs()`

yz-cnsdqz, 2019-10-30:
Many thanks! So, the annotation information from print_scene_recur(), and the sensor poses are actually the one after rotation from -Z to -Y?  

> The mp3d meshes having -Z as gravity and habitat uses -Y, so we rotate the mp3d meshes and the coordinates in annotations on load.
> 
> This rotation can sometimes causes the dims to be negative, you can either work with the negative values or just just do `.abs()`

Many thanks! So, the annotation information from print_scene_recur(), and the sensor poses are actually the one after rotation from -Z to -Y?

yz-cnsdqz, 2019-10-30:
> Many thanks! So, the annotation information from print_scene_recur(), and the sensor poses are actually the one after rotation from -Z to -Y?
>
>
> Many thanks! So, the annotation information from [...]

When I read the file _17DRP5sb8fy_semantic.ply_  using Open3D and then rotate it around x axis by 90 degree, so as to change the gravity direction from -Z to -Y

yz-cnsdqz, 2019-10-30:
> Many thanks! So, the annotation information from print_scene_recur(), and the sensor poses are actually the one after rotation from -Z to -Y?
>
>
> Many thanks! So, the annotation information from [...]

I did an experiment to verify it. The solution is first rotating the mesh around x-axis by -90 degree, and then take abs() for the bounding box size.

closed 2019-10-30 by yz-cnsdqz

erikwijmans, 2019-10-31:
Perfect.  The rotation we apply to mesh (`habitat_sim.utils.common.quat_from_two_vectors(np.array([0, 0, -1]), np.array([0, -1, 0]))`) is identical to rotating about x by -90 degrees (`habitat_sim.utils.common.quat_from_angle_axis(-np.pi/2, np.array([1, 0, 0]))`).  So glad that all worked out.