facebookresearch/habitat-sim #328: About RGB Images Quality

erick84mm, 2019-10-31:
## ❓ Questions and Help

I have been working with Habitat and the Matterport3DSimulator. And It seems to me that there is a significant difference in terms of the quality of the RGB images. As far as I understand the images are being generated from the meshes in habitat. 
I was wondering if there is any way to extract high quality RGB images as in Matterport3DSimulators.

[image]

On top you will see the image extracted from the Matterport3DSimulator, the bottom one is from Habitat. 

Thank you in advance

erikwijmans, 2019-10-31:
There is not a way to render the panoramas from the Matterport3D dataset in habitat-sim.

dhruvbatra, 2019-11-02:
It's a nice domain-adaptation pix2pix project waiting to happen.

smorad, 2020-02-05:
It would be nice to have an option to utilize the `gibson` "hole-filler" network in the rendering pipeline:
https://github.com/StanfordVL/GibsonEnv/blob/8e9e23c2dcc44bb5442a25d12928103f05b51d8f/gibson/core/render/pcrender.py#L169