facebookresearch/habitat-sim #330: isometric projection of scen

msbaines, 2019-10-31:
## 🚀 Feature

In some cases where the current top-down map is used it may be helpful to use an isometric projection of the scene instead.

## Motivation

An isometric projection of the scene is useful for seeing the whole scene at once and for visually where the agent, or object, is in the scene.

## Pitch

## Alternatives

## Additional context

dhruvbatra, 2019-11-01:
Example output: http://crlab.cs.columbia.edu/learning_your_way/

gianscarpe, 2020-12-09:
Hi, I'm still interested in a similar feature. Is someone working on it? I could implement it myself, maybe with some suggestions from the authors! Thank you :)

jturner65, 2020-12-09:
Hello

Yes, orthographic projection cameras have recently been added to Habitat-Sim with [this PR](https://github.com/facebookresearch/habitat-sim/pull/933) which can be used to provide an isometric projection.  You can experiment with an orthographic camera by running viewer.cpp - the camera can be toggled between orthographic and pinhole/perspective by pressing '5'. @Skylion007 has been working on [implementing hab-lab support here.](https://github.com/facebookresearch/habitat-lab/pull/533).

closed 2022-03-01 by aclegg3