facebookresearch/habitat-sim #347: Looping over multiple environments

anoopsonar30, 2019-11-12:
I'm trying to simulate in multiple environments sequentially by iterating through them. The source code is below. `numSim = 5` and `L` Is the number of controllers. 

```
128     for glb in envGlbs:
129         sim_settings["scene"] = glb
130         cfg = make_cfg(sim_settings)
131         sim = habitat_sim.Simulator(cfg)
132         costEnv = np.zeros((1, L))
133         sim.seed(seed)
134         
135         agent = sim.initialize_agent(sim_settings["default_agent"])
136         initState = agent.get_state()
137 
138         obs = sim.reset()
139         depth = obs["depth_sensor"]
140 
141         print("Environment Number : ", (envNum + 1))
142         for l in range(0, L):
143             
144             agent.set_state(initState)
145             for t in range(0, T_horizon):       
146 
147                 
148                 print("Timestep : ", t)
149                 depth = np.reshape(obs["depth_sensor"], [1920, 1080])
150                 indices = np.round(np.linspace(0, np.shape(depth)[1] - 1, numRays)).astype(int)
151                 y = depth[(int)(np.shape(depth)[0]/2)][indices]
152                 y[y == 0] = 0.05
153 
154                 # print("Depth input : ", y)
155                 u = compute_control(y, K[l])       
156                 obs = robot_update_state(sim, agent, u)
157                 # print("Control output :", u)
158 
159                 rgb = obs["color_sensor"]
160                 # semantic = obs["semantic_sensor"]
161                 # display_sample(rgb, depth)
162                     
163                 # Add code to act and detect collision  
164                 if obs["collided"]:
165                     print("Collided")
166                     costEnv[0][l] = 1
167                     break
168 
169         allCosts[envNum] = costEnv
170         envNum = envNum + 1
171                     
172         if (envNum == numSim):
173             print(limits)
174             break
175 
176         sim.close() 
177         del sim
178 
179     sim.close()
180     return allCosts
```

The first environment is loaded correctly and the simulation runs just fine. However, I get the following error before the simulations in the second environment can begin - 

```
I1111 19:25:31.772603  2061 Simulator.cpp:35] Deconstructing Simulator
I1111 19:25:31.772620  2061 SemanticScene.h:40] Deconstructing SemanticScene
I1111 19:25:31.772625  2061 SceneManager.h:24] Deconstructing SceneManager
I1111 19:25:31.772629  2061 SceneGraph.h:20] Deconstructing SceneGraph
I1111 19:25:31.811895  2061 Renderer.cpp:33] Deconstructing Renderer
I1111 19:25:31.811921  2061 WindowlessContext.h:16] Deconstructing WindowlessContext
I1111 19:25:31.811928  2061 WindowlessContext.cpp:245] Deconstructing GL context
Renderer: TITAN Xp/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 418.87.00
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
W1111 19:25:34.744913  2061 Simulator.cpp:133] :
---
 The active scene does not contain semantic annotations. 
---
I1111 19:25:34.745260 2061 simulator.py:127] Loaded navmesh ../../data/scene_datasets/allData/Elton.navmesh
I1111 19:25:34.747261  2061 Sensor.h:82] Deconstructing Sensor
I1111 19:25:34.747273  2061 RenderTarget.h:51] Deconstructing RenderTarget
I1111 19:25:34.747690  2061 Sensor.h:82] Deconstructing Sensor
I1111 19:25:34.747695  2061 RenderTarget.h:51] Deconstructing RenderTarget
I1111 19:25:34.747942  2061 PathFinder.h:113] Deconstructing PathFinder
Assertion framebuffer_.checkStatus(GL::FramebufferTarget::Draw) == GL::Framebuffer::Status::Complete failed in /home/ugrad/Documents/Habitat/habitat-sim/src/esp/gfx/RenderTarget.cpp on line 90
[1]    2061 abort (core dumped)  python run_pacbayes.py 5
```
Let me know if you need any more system info. Also I'd love to know if there's an alternate solution to randomly spawning in different positions in random indoor environments if there is one. I'm not sure if Habitat-API already does this.

Also is there a way to parallelize this process? or do you recommend it? I've done that in Go but not sure how this is built under the hood and how it would work in Python. Could look into that if you haven't already implemented it. The computer this is running on has a lot of cores so that would be pretty helpful to speed everything up.

erikwijmans, 2019-11-12:
For the crash, can you see if you building after this PR https://github.com/facebookresearch/habitat-sim/pull/313 ? 

The `agent = sim.initialize_agent(sim_settings["default_agent"])` interfered with the shutdown logic before that PR.  You can also add a `agent = None` before the `sim.close()` and things should work.

erikwijmans, 2019-11-12:
For parallelizing things, I use a python multiprocessing pool (https://docs.python.org/3/library/multiprocessing.html) to do it.

i.e.

```python
import multiprocessing

def my_habitat_compute_fn(...):
    ...

mp_ctx = multiprocessing.get_context("forkserver")
with mp_ctx.Pool(num_workers) as pool:
    results = pool.map(my_habitat_compute_fn, work_items)
```

anoopsonar30, 2019-11-12:
> For the crash, can you see if you building after this PR #313 ?
> 
> The `agent = sim.initialize_agent(sim_settings["default_agent"])` interfered with the shutdown logic before that PR. You can also add a `agent = None` before the `sim.close()` and things should work.

The current build of habitat-sim is with a new clone that I made just around a week ago. Although the `agent = None` fixed worked. Thank you :)

mathfac, 2019-11-12:
If the issue is happening, let's keep issue open till we validate using your code.

erikwijmans, 2019-11-13:
The test here should be checking to see if this is stable: https://github.com/facebookresearch/habitat-sim/blob/master/tests/test_simulator.py#L46 Likely that this is just on a commit before that PR landed.

aclegg3, 2021-04-15:
Is this issue resolved? @erikwijmans @mathfac @anoopsonar30

closed 2021-04-18 by anoopsonar30