facebookresearch/habitat-sim #351: I can't build habitat-sim

Codeczh, 2019-11-14:
## ❓ Questions and Help
**This is the error  **
(habitat) czh@dirac:~/project/habitat/habitat-sim$ python setup.py install --headless --with-cuda --user
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
-- The CUDA compiler identification is unknown
CMake Error at CMakeLists.txt:10 (project):
  No CMAKE_CUDA_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CUDACXX" or the CMake cache entry CMAKE_CUDA_COMPILER to the full
  path to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/farm/czh/project/habitat/habitat-sim/build/CMakeFiles/CMakeOutput.log".
See also "/farm/czh/project/habitat/habitat-sim/build/CMakeFiles/CMakeError.log".
Traceback (most recent call last):
  File "setup.py", line 380, in <module>
    include_package_data=True,
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 172, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command
    self.run_command(cmdname)
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 202, in run
    self.build_extension(ext)
  File "setup.py", line 269, in build_extension
    env=env,
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-H/farm/czh/project/habitat/habitat-sim/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/farm/czh/project/habitat/habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/farm/czh/anaconda3/envs/habitat/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=OFF', '-DTARGET_HEADLESS=ON', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=ON']' returned non-zero exit status 1.
-------------------------------
so anyone can help me? thanks a lot.

Codeczh, 2019-11-14:
It also occurs that CMake Error at CMakeLists.txt:10 (project):
  No CMAKE_CUDA_COMPILER could be found.
And I insert "set(CMAKE_CUDA_COMPILER "/usr/local/cuda-10.0/bin/nvcc")" into  /habitat-sim/src/CMakeLists.txt.
Then the new problem——
(habitat) czh@dirac:~/project/habitat/habitat-sim$ python setup.py install --headless --with-cuda --user
running install
running bdist_egg
running egg_info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
copying habitat_sim/errors.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/gfx.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/logging.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/scene.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/geo.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/registry.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/physics.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/sensor.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/simulator.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/nav/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/nav
copying habitat_sim/nav/greedy_geodesic_follower.py -> build/lib.linux-x86_64-3.6/habitat_sim/nav
copying habitat_sim/utils/common.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/collect_env.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/bindings/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/bindings
copying habitat_sim/sensors/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors
copying habitat_sim/sensors/sensor_suite.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors
copying habitat_sim/agent/agent.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent
copying habitat_sim/agent/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent
copying habitat_sim/sensors/noise_models/redwood_depth_noise_model.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/sensor_noise_model.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/sensors/noise_models/no_noise_model.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models
copying habitat_sim/agent/controls/controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/object_controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/default_controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/pyrobot_noisy_controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/sensors/noise_models/data/redwood-depth-dist-model.npy -> build/lib.linux-x86_64-3.6/habitat_sim/sensors/noise_models/data
running build_ext
error: [Errno 2] No such file or directory: 'build/compile_commands.json'

Codeczh, 2019-11-14:
I delete all the code and follow the readme.md insturction, to now, I  have not come to any problem, but the process is so slow.   I feel almost 1 hour passed...

###this is report###
(habitat) czh@dirac:~/project/habitat/habitat-sim$ python setup.py install --headless --with-cuda --user
running install
running bdist_egg
running egg_info
creating habitat_sim.egg-info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-3.6
creating build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/errors.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/gfx.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/logging.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/scene.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/geo.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/registry.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/physics.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/sensor.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/simulator.py -> build/lib.linux-x86_64-3.6/habitat_sim
creating build/lib.linux-x86_64-3.6/habitat_sim/nav
copying habitat_sim/nav/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/nav
copying habitat_sim/nav/greedy_geodesic_follower.py -> build/lib.linux-x86_64-3.6/habitat_sim/nav
creating build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/common.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/collect_env.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
copying habitat_sim/utils/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/utils
creating build/lib.linux-x86_64-3.6/habitat_sim/bindings
copying habitat_sim/bindings/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/bindings
creating build/lib.linux-x86_64-3.6/habitat_sim/sensors
copying habitat_sim/sensors/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors
copying habitat_sim/sensors/sensor_suite.py -> build/lib.linux-x86_64-3.6/habitat_sim/sensors
creating build/lib.linux-x86_64-3.6/habitat_sim/agent
copying habitat_sim/agent/agent.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent
copying habitat_sim/agent/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent
creating build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/object_controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/__init__.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/default_controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
copying habitat_sim/agent/controls/pyrobot_noisy_controls.py -> build/lib.linux-x86_64-3.6/habitat_sim/agent/controls
running build_ext
Submodule 'docs/m.css' (https://github.com/mosra/m.css) registered for path 'docs/m.css'
Submodule 'src/deps/Sophus' (https://github.com/strasdat/Sophus) registered for path 'src/deps/Sophus'
Submodule 'src/deps/assimp' (https://github.com/assimp/assimp) registered for path 'src/deps/assimp'
Submodule 'src/deps/corrade' (https://github.com/mosra/corrade.git) registered for path 'src/deps/corrade'
Submodule 'src/deps/eigen-git-mirror' (https://github.com/eigenteam/eigen-git-mirror) registered for path 'src/deps/eigen-git-mirror'
Submodule 'src/deps/glfw' (https://github.com/glfw/glfw) registered for path 'src/deps/glfw'
Submodule 'src/deps/glog' (https://github.com/google/glog.git) registered for path 'src/deps/glog'
Submodule 'src/deps/googletest' (https://github.com/google/googletest) registered for path 'src/deps/googletest'
Submodule 'src/deps/magnum' (https://github.com/mosra/magnum.git) registered for path 'src/deps/magnum'
Submodule 'src/deps/magnum-bindings' (https://github.com/mosra/magnum-bindings.git) registered for path 'src/deps/magnum-bindings'
Submodule 'src/deps/magnum-integration' (https://github.com/mosra/magnum-integration.git) registered for path 'src/deps/magnum-integration'
Submodule 'src/deps/magnum-plugins' (https://github.com/mosra/magnum-plugins.git) registered for path 'src/deps/magnum-plugins'
Submodule 'src/deps/pybind11' (https://github.com/pybind/pybind11.git) registered for path 'src/deps/pybind11'
Submodule 'src/deps/rapidjson' (https://github.com/Tencent/rapidjson.git) registered for path 'src/deps/rapidjson'
Submodule 'src/deps/recastnavigation' (https://github.com/erikwijmans/recastnavigation.git) registered for path 'src/deps/recastnavigation'
Submodule 'src/deps/tinyply' (https://github.com/ddiakopoulos/tinyply) registered for path 'src/deps/tinyply'
Cloning into '/farm/czh/project/habitat/habitat-sim/docs/m.css'...
Cloning into '/farm/czh/project/habitat/habitat-sim/src/deps/Sophus'...
Cloning into '/farm/czh/project/habitat/habitat-sim/src/deps/assimp'...

Codeczh, 2019-11-14:
Fortunately， I compile the habitat-sim successfully. My new question is my machine is headless, so if I need carry out the instruction of WebGL?

Codeczh, 2019-11-14:
When I run the  `python setup.py test`  following the 4th step of habitat-api install, this error occurs.

>(habitat) czh@dirac:~/project/habitat/habitat-api$ python setup.py test
running pytest
Searching for pytest-mock
Best match: pytest-mock 1.11.2
Processing pytest_mock-1.11.2-py3.6.egg

Using /farm/czh/project/habitat/habitat-api/.eggs/pytest_mock-1.11.2-py3.6.egg
running egg_info
writing habitat.egg-info/PKG-INFO
writing dependency_links to habitat.egg-info/dependency_links.txt
writing requirements to habitat.egg-info/requires.txt
writing top-level names to habitat.egg-info/top_level.txt
reading manifest file 'habitat.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat.egg-info/SOURCES.txt'
running build_ext
========================================================================= test session starts =========================================================================
platform linux -- Python 3.6.9, pytest-5.2.2, py-1.8.0, pluggy-0.13.0
rootdir: /farm/czh/project/habitat/habitat-api, inifile: setup.cfg, testpaths: test
plugins: mock-1.11.2, hypothesis-4.44.2
collected 77 items                                                                                                                                                    

test/test_baseline_agents.py F.                                                                                                                                 [  2%]
test/test_baseline_trainers.py FFFF..                                                                                                                           [ 10%]
test/test_config.py ..                                                                                                                                          [ 12%]
test/test_dataset.py ....................                                                                                                                       [ 38%]
test/test_demo_notebook.py s                                                                                                                                    [ 40%]
test/test_habitat_env.py F.F.s....FFatal Python error: Aborted

Current thread 0x00007fc4c8e60740 (most recent call first):
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/json/decoder.py", line 355 in raw_decode
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/json/decoder.py", line 339 in decode
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/json/__init__.py", line 354 in loads
  File "/farm/czh/project/habitat/habitat-api/habitat/datasets/pointnav/pointnav_dataset.py", line 104 in from_json
  File "/farm/czh/project/habitat/habitat-api/habitat/datasets/pointnav/pointnav_dataset.py", line 83 in __init__
  File "/farm/czh/project/habitat/habitat-api/habitat/datasets/registration.py", line 18 in make_dataset
  File "/farm/czh/project/habitat/habitat-api/test/test_habitat_env.py", line 62 in _load_test_data
  File "/farm/czh/project/habitat/habitat-api/test/test_habitat_env.py", line 221 in test_rl_vectorized_envs
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/python.py", line 170 in pytest_pyfunc_call
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 86 in <lambda>
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 92 in _hookexec
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/python.py", line 1423 in runtest
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/runner.py", line 125 in pytest_runtest_call
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 86 in <lambda>
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 92 in _hookexec
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/runner.py", line 201 in <lambda>
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/runner.py", line 229 in from_call
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/runner.py", line 201 in call_runtest_hook
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/runner.py", line 176 in call_and_report
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/runner.py", line 95 in runtestprotocol
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/_pytest/runner.py", line 80 in pytest_runtest_protocol
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 86 in <lambda>
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/manager.py", line 92 in _hookexec
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/pluggy/hooks.py", line 286 in __call__
[... cut ...]
  File "/farm/czh/project/habitat/habitat-api/.eggs/pytest_runner-5.2-py3.6.egg/ptr.py", line 209 in run
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974 in run_command
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 955 in run_commands
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/core.py", line 148 in setup
  File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/__init__.py", line 145 in setup
  File "setup.py", line 99 in <module>
Aborted (core dumped)

---

erikwijmans, 2019-11-14:
>  My new question is my machine is headless

You need to add `--headless` to your build command.  Tests are likely failing because it can't get a rendering context.

Codeczh, 2019-11-14:
> > My new question is my machine is headless
> 
> You need to add `--headless` to your build command. Tests are likely failing because it can't get a rendering context.

Thank you for reply,   I  just use `python  setup.py test --headless`, but it prompts error as below.
>(habitat) czh@dirac:~/project/habitat/habitat-api$ python setup.py test --headless
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --headless not recognized

erikwijmans, 2019-11-14:
This is for the build command for habitat-sim, not habitat-api.  Although based on your first post, it seems like you already ran that.

What does running `python example/example.py` say?

Codeczh, 2019-11-14:
@erikwijmans   I really appreciate for your help，running the command python example/example.py just shows that "Episode finished after 3 steps"(almost this) in the end. Its successful as the expected results.  But, when it turns to python setup.py test, errors above occur._(:з」∠)_

erikwijmans, 2019-11-14:
What does `pytest -xs` report?  That'll exit on the first failure and print things as it goes.  Multiple failures can cause OpenGL context errors sometimes.

Codeczh, 2019-11-15:
> What does `pytest -xs` report? That'll exit on the first failure and print things as it goes. Multiple failures can cause OpenGL context errors sometimes.

Error occurs in the end, and this is report——
>test/test_mp3d_eqa.py ssss
test/test_pointnav_dataset.py 2019-11-15 09:41:35,252 JSON conversion finished. 0.12128925323486328 sec
.ss2019-11-15 09:41:35,694 Initializing dataset PointNav-v1
2019-11-15 09:41:35,697 initializing sim Sim-v0
I1115 09:41:35.699921  4036 WindowlessContext.cpp:99] [EGL] Detected 11 EGL devices
I1115 09:41:35.700793  4036 WindowlessContext.cpp:120] [EGL] Selected EGL device 2 for CUDA device 0
I1115 09:41:35.700837  4036 WindowlessContext.cpp:134] [EGL] Version: 1.5
I1115 09:41:35.700843  4036 WindowlessContext.cpp:135] [EGL] Vendor: NVIDIA
Renderer: Tesla V100-PCIE-16GB/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 410.48
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
W1115 09:41:37.013196  4036 Simulator.cpp:133] :

 The active scene does not contain semantic annotations. 

I1115 09:41:37.013591 4036 simulator.py:127] Loaded navmesh data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
2019-11-15 09:41:37,032 Initializing task Nav-v0
.
test/test_pyrobot.py E

=========================================================== ERRORS ============================================================
_______________________________________________ ERROR at setup of test_pyrobot ________________________________________________
file /farm/czh/project/habitat/habitat-api/test/test_pyrobot.py, line 43
  def test_pyrobot(mocker):
E       fixture 'mocker' not found
       available fixtures: cache, capfd, capfdbinary, caplog, capsys, capsysbinary, doctest_namespace, monkeypatch, pytestconfig, record_property, record_testsuite_property, record_xml_attribute, recwarn, tmp_path, tmp_path_factory, tmpdir, tmpdir_factory
       use 'pytest --fixtures [testpath]' for help on them.

/farm/czh/project/habitat/habitat-api/test/test_pyrobot.py:43
====================================================== warnings summary =======================================================
/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/past/builtins/misc.py:45
  /farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/past/builtins/misc.py:45: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
    from imp import reload

-- Docs: https://docs.pytest.org/en/latest/warnings.html
=================================================== short test summary info ===================================================
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_demo_notebook.py:22: Please download the Matterport3D PointNav val dataset and Matterport3D val scenes
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_habitat_env.py:111: Fork does not support gpu2gpu
SKIPPED [1] test/test_habitat_env.py:371: unconditional skip
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_habitat_task.py:85: Please download dataset to data folder data/datasets/eqa/mp3d/v1/{split}/{split}.json.gz.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_mp3d_eqa.py:97: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_mp3d_eqa.py:115: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_mp3d_eqa.py:152: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_mp3d_eqa.py:241: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_pointnav_dataset.py:68: Test skipped as dataset files are missing.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_pointnav_dataset.py:96: Test skipped as dataset files are missing.
=============================== 59 passed, 10 skipped, 1 warnings, 1 error in 326.77s (0:05:26) ===============================
I1115 09:42:26.099572  4036 Simulator.cpp:35] Deconstructing Simulator
I1115 09:42:26.099627  4036 SemanticScene.h:40] Deconstructing SemanticScene
I1115 09:42:26.099633  4036 SceneManager.h:24] Deconstructing SceneManager
I1115 09:42:26.099637  4036 SceneGraph.h:20] Deconstructing SceneGraph
I1115 09:42:26.099649  4036 Sensor.h:82] Deconstructing Sensor
I1115 09:42:26.099653  4036 RenderTarget.h:51] Deconstructing RenderTarget
I1115 09:42:26.102501  4036 Sensor.h:82] Deconstructing Sensor
I1115 09:42:26.102510  4036 RenderTarget.h:51] Deconstructing RenderTarget
I1115 09:42:26.119688  4036 Renderer.cpp:33] Deconstructing Renderer
I1115 09:42:26.119726  4036 WindowlessContext.h:16] Deconstructing WindowlessContext
I1115 09:42:26.119730  4036 WindowlessContext.cpp:245] Deconstructing GL context
I1115 09:42:26.119956  4036 PathFinder.h:113] Deconstructing PathFinder

erikwijmans, 2019-11-15:
Looks like we need up to update the requirements.  `pip install mock pytest-mock` should fix that.

Codeczh, 2019-11-15:
> Looks like we need up to update the requirements. `pip install pytest-mock` should fix that.

Thank you， after `pip install pytest-mock`, the `pytest -xs`  runs well:
======================================================== warnings summary ========================================================
/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/past/builtins/misc.py:45
  /farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/past/builtins/misc.py:45: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
    from imp import reload

-- Docs: https://docs.pytest.org/en/latest/warnings.html
==================================================== short test summary info =====================================================
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_demo_notebook.py:22: Please download the Matterport3D PointNav val dataset and Matterport3D val scenes
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_habitat_env.py:111: Fork does not support gpu2gpu
SKIPPED [1] test/test_habitat_env.py:371: unconditional skip
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_habitat_task.py:85: Please download dataset to data folder data/datasets/eqa/mp3d/v1/{split}/{split}.json.gz.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_mp3d_eqa.py:97: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_mp3d_eqa.py:115: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_mp3d_eqa.py:152: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_mp3d_eqa.py:241: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_pointnav_dataset.py:68: Test skipped as dataset files are missing.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_pointnav_dataset.py:96: Test skipped as dataset files are missing.
===================================== 67 passed, 10 skipped, 1 warnings in 344.47s (0:05:44) ====================================

##however, running `python setup.py test` still  come across errors stated above.  

test/test_demo_notebook.py   s              [ 40%]
test/test_habitat_env.py  F.F.s....F               Fatal Python error: Aborted

erikwijmans, 2019-11-15:
`pytest -xs` passing but `python setup.py test` not is very odd.  My guess as to the test that's failing first (`test/test_habitat_env.py F`) is one of the tests that uses GPU2GPU transfer (one of those is the first to run in that file).  If you build habitat-sim without `--with-cuda` do things pass?

Codeczh, 2019-11-15:
> `pytest -xs` passing but `python setup.py test` not is very odd. My guess as to the test that's failing first (`test/test_habitat_env.py F`) is one of the tests that uses GPU2GPU transfer (one of those is the first to run in that file). If you build habitat-sim without `--with-cuda` do things pass?

I remember that I run `python setup.py install --headless --with-cuda --user` to build habitat-sim and it succeeded. Should I try `python setup.py install --headless  --user` again?

Codeczh, 2019-11-15:
> This is for the build command for habitat-sim, not habitat-api. Although based on your first post, it seems like you already ran that.
> 
> What does running `python example/example.py` in the habitat-sim repo say?

Hi, I finally fixed it, just add `CUDA_VISIBLE_DEVICES=2  python setup.py test`,
then it runs well.

==================== test session starts ==============================
platform linux -- Python 3.6.9, pytest-5.2.2, py-1.8.0, pluggy-0.13.0
rootdir: /farm/czh/project/habitat/habitat-api, inifile: setup.cfg, testpaths: test
plugins: mock-1.11.2, hypothesis-4.44.2
collected 77 items                                                                                                                                          

test/test_baseline_agents.py ..                                                                                                                       [  2%]
test/test_baseline_trainers.py ......                                                                                                                 [ 10%]
test/test_config.py ..                                                                                                                                [ 12%]
test/test_dataset.py ....................                                                                                                             [ 38%]
test/test_demo_notebook.py s                                                                                                                          [ 40%]
test/test_habitat_env.py ....s...........s                                                                                                            [ 62%]
test/test_habitat_example.py .....                                                                                                                    [ 68%]
test/test_habitat_sim.py ..                                                                                                                           [ 71%]
test/test_habitat_task.py ....s                                                                                                                       [ 77%]
test/test_install.py .                                                                                                                                [ 79%]
test/test_mp3d_eqa.py ssss                                                                                                                            [ 84%]
test/test_pointnav_dataset.py .ss.                                                                                                                    [ 89%]
test/test_pyrobot.py .                                                                                                                                [ 90%]
test/test_relative_camera.py .                                                                                                                        [ 92%]
test/test_sensors.py ......                                                                                                                           [100%]

================ warnings summary =============================
/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/past/builtins/misc.py:45
  /farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/past/builtins/misc.py:45: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
    from imp import reload

-- Docs: https://docs.pytest.org/en/latest/warnings.html
========== short test summary info ==============
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_demo_notebook.py:22: Please download the Matterport3D PointNav val dataset and Matterport3D val scenes
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_habitat_env.py:111: Fork does not support gpu2gpu
SKIPPED [1] test/test_habitat_env.py:371: unconditional skip
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_habitat_task.py:85: Please download dataset to data folder data/datasets/eqa/mp3d/v1/{split}/{split}.json.gz.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_mp3d_eqa.py:97: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_mp3d_eqa.py:115: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_mp3d_eqa.py:152: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_mp3d_eqa.py:241: Please download Matterport3D EQA dataset to data folder.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_pointnav_dataset.py:68: Test skipped as dataset files are missing.
SKIPPED [1] /farm/czh/project/habitat/habitat-api/test/test_pointnav_dataset.py:96: Test skipped as dataset files are missing.
======= 67 passed, 10 skipped, 1 warnings in 328.04s (0:05:28) =========

closed 2020-08-24 by Skylion007

TianyuCao, 2024-11-06:
> > ## ❓ Questions and Help
> > This is the error (habitat) czh@dirac:~/project/habitat/habitat-sim$ python setup.py install --headless --with-cuda --user running install running bdist_egg running egg_info writing habitat_sim.egg-info/PKG-INFO writing dependency_links to habitat_sim.egg-info/dependency_links.txt writing requirements to habitat_sim.egg-info/requires.txt writing top-level names to habitat_sim.egg-info/top_level.txt reading manifest file 'habitat_sim.egg-info/SOURCES.txt' reading manifest template 'MANIFEST.in' writing manifest file 'habitat_sim.egg-info/SOURCES.txt' installing library code to build/bdist.linux-x86_64/egg running install_lib running build_py running build_ext -- The CUDA compiler identification is unknown CMake Error at CMakeLists.txt:10 (project): No CMAKE_CUDA_COMPILER could be found. Tell CMake where to find the compiler by setting either the environment variable "CUDACXX" or the CMake cache entry CMAKE_CUDA_COMPILER to the full path to the compiler, or to the compiler name if it is in the PATH. --Configuring incomplete, errors occurred! See also "/farm/czh/project/habitat/habitat-sim/build/CMakeFiles/CMakeOutput.log". See also "/farm/czh/project/habitat/habitat-sim/build/CMakeFiles/CMakeError.log". Traceback (most recent call last): File "setup.py", line 380, in include_package_data=True, File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/**init**.py", line 145, in setup return distutils.core.setup(**attrs) File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/core.py", line 148, in setup dist.run_commands() File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 955, in run_commands self.run_command(cmd) File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command cmd_obj.run() File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run self.do_egg_install() File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install self.run_command('bdist_egg') File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command self.distribution.run_command(command) File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command cmd_obj.run() File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 172, in run cmd = self.call_command('install_lib', warn_dir=0) File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 158, in call_command self.run_command(cmdname) File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command self.distribution.run_command(command) File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command cmd_obj.run() File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run self.build() File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/command/install_lib.py", line 107, in build self.run_command('build_ext') File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command self.distribution.run_command(command) File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command cmd_obj.run() File "setup.py", line 202, in run self.build_extension(ext) File "setup.py", line 269, in build_extension env=env, File "/farm/czh/anaconda3/envs/habitat/lib/python3.6/subprocess.py", line 311, in check_call raise CalledProcessError(retcode, cmd) subprocess.CalledProcessError: Command '['cmake', '-H/farm/czh/project/habitat/habitat-sim/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/farm/czh/project/habitat/habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/farm/czh/anaconda3/envs/habitat/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=OFF', '-DTARGET_HEADLESS=ON', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=ON']' returned non-zero exit status 1.
> > so anyone can help me? thanks a lot.
> 
> Hi I got the same error " No CMAKE_CUDA_COMPILER could be found." . Could you kindly inform me how did you solve this problem ? Thanks

Hi all, I got the same error here. Could you plz inform me how you solved the problem as well? Many thanks in advance!

Codeczh, 2024-11-06:
你好，邮件已收到。