facebookresearch/habitat-sim #354: AssertionError: Could not find dataset PointNav-v1

BryanBo-Cao, 2019-11-15:
## ❓ Questions and Help

I cloned and installed habitat-sim, downloaded and extracted the test dataset in the ```habitat-sim/data/```.

But when I run ```python examples/example.py``` in ```habitat-sim```

I got
```
2019-11-15 17:07:58,934 Initializing dataset PointNav-v1
Traceback (most recent call last):
  File "examples/example.py", line 25, in <module>
    example()
  File "examples/example.py", line 11, in example
    env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))
  File "/home/bryanbc/Repos/habitat-api/habitat/core/env.py", line 74, in __init__
    id_dataset=config.DATASET.TYPE, config=config.DATASET
  File "/home/bryanbc/Repos/habitat-api/habitat/datasets/registration.py", line 16, in make_dataset
    assert _dataset is not None, "Could not find dataset {}".format(id_dataset)
AssertionError: Could not find dataset PointNav-v1
```
Any help would be appreciated!

erikwijmans, 2019-11-16:
You seem to be running habitat-api's example script, not habitat-sim's.

BryanBo-Cao, 2019-11-16:
Ahh, yes. Sorry I was installing both habitat-sim and habitat-api simultaneously and supposed to create this issue on habitat-api repo :(

closed 2019-11-16 by BryanBo-Cao