facebookresearch/habitat-sim #358: Check failed: std::getenv("DISPLAY") != nullptr DISPLAY not detected in Habitat Sim Demo

BryanBo-Cao, 2019-11-20:
## 🐛 Bug
When I run the [demo code](https://aihabitat.org/docs/habitat-api/habitat-sim-demo.html#simulator-config) until
```
sim = habitat_sim.Simulator(cfg)
```
My notebook's Kernel died and restarted.

Error message from my Terminal:
```
[I 20:42:14.992 NotebookApp] KernelRestarter: restarting kernel (1/5), keep random ports
kernel d28c62f2-014b-41a4-85bc-3dfb1d72cde5 restarted
WARNING: Logging before InitGoogleLogging() is written to STDERR
F1119 20:42:19.850073 19734 WindowlessContext.cpp:235] Check failed: std::getenv("DISPLAY") != nullptr DISPLAY not detected. For headless systems, compile with --headless for EGL support
*** Check failure stack trace: ***
```

## To Reproduce

Steps to reproduce the behavior:

Built the habitat-api code by

1. ```python setup.py install --headless```

2. ```export PYTHONPATH=$PYTHONPATH:/path/to/habitat-sim/```
Then I create my own repo outside the habitat-sim and run ```test_Habitat_Sim_Demo.ipynb``` , which is the code from [demo code](https://aihabitat.org/docs/habitat-api/habitat-sim-demo.html#simulator-config).

## Expected behavior
Simply run the demo in a remote server without attaching to a physical monitor, be able to have an agent taking actions in the environment and display frames when needed from my Jupyter Notebook.

## Additional context
I run the code in Jupyter Notebook on a remote server that does not have attached monitors.

## System Info
```
ENVIRONMENT INFO:
Platform: Linux-4.15.0-65-generic-x86_64-with-debian-buster-sid
Machine: x86_64
Processor: x86_64
Libc version: glibc 2.10
Mac version:
Python version: 3.7.4
Architecture: 64bit
Win version:
System OS: Linux
Release: 4.15.0-65-generic
Version: #74-Ubuntu SMP Tue Sep 17 17:06:04 UTC 2019
Operational System: linux
GCC version: gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
CMAKE version: cmake version 3.10.2
NVIDIA-SMI: Tue Nov 19 20:50:52 2019
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 430.40       Driver Version: 430.40       CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  TITAN X (Pascal)    Off  | 00000000:01:00.0 Off |                  N/A |
| 23%   26C    P8     8W / 250W |      2MiB / 12196MiB |      0%      Default |
...
```

erikwijmans, 2019-11-20:
`export PYTHONPATH=$PYTHONPATH:/path/to/habitat-sim/` should be done when using `python setup.py install --headless`.

BryanBo-Cao, 2019-11-20:
One thing to note is that I can run the demo code without error in ```python test-habitat-sim-demo.py```, but I was unable to run the corresponding ```.ipynb``` code using Jupyter Notebook remotely.

erikwijmans, 2019-11-20:
Are you running `python test-habitat-sim-demo.py` remotely also?

BryanBo-Cao, 2019-11-20:
Separately, not at the same time.

erikwijmans, 2019-11-20:
Are you doing `export PYTHONPATH=$PYTHONPATH:/path/to/habitat-sim/` when you run `python test-habitat-sim-demo.py` on the remote server?

BryanBo-Cao, 2019-11-20:
Yes, I did
```
export PYTHONPATH=$PYTHONPATH:/path/to/habitat-sim/
```
and
```
python test-habitat-sim-demo.py
```
Previously without doing ```export PYTHONPATH=$PYTHONPATH:/path/to/habitat-sim/```, my code raised an issue saying it couldn't import habitat-sim. But after ```export ...``` it was fixed.

erikwijmans, 2019-11-20:
If you ran `python setup.py install --headless`, that shouldn't happen.  Do you still have the correct conda env activated?

BryanBo-Cao, 2019-11-20:
I see! I follow the instructions again:
```
conda activate habitat
```
and then
```
python setup.py install --headless
```
Now I can run the demo code!

Probably previously I opened a couple of tabs on my Terminal but didn't realize that in some tabs the conda env of habitat was not activated(sorry about that) when installing ```python setup.py install --headless```.

Thank you for your time and appreciate your help! @erikwijmans

BryanBo-Cao, 2019-11-30:
Not sure why my Jupyter Notebook kernel crashed again so I reopen this issue.

I downloaded the repo and followed the instructions again when my env is activated:
```
pip install -r requirements.txt
python setup.py install --headless
export PYTHONPATH=$PYTHONPATH:/path/to/habitat-sim/
```
and for sure ```/path/to/habitat-sim/``` is specified in my server.

erikwijmans, 2019-11-30:
Do not do `export PYTHONPATH=$PYTHONPATH:/path/to/habitat-sim/`

closed 2019-11-30 by BryanBo-Cao