facebookresearch/habitat-sim #364: How to set the position of added object

picksh, 2019-11-26:
## ❓ Questions and Help
Hello, I am wondering  how can i set a certain position and orientation of the object added through  '--enable-physics'. 

Also, I noticed that there is the function of drawing bounding box of physics but I have no idea how to use it (i have  tried to press 'B' in 'Physical interactions testing' but nothing happened) .

I built Habitat-Sim with 'Default build' and use the master branch.

Every answer would be appreciated!

aclegg3, 2019-12-02:
Hey, thanks for reaching out. If you are intending to use python, physics objects can be manipulated [through the Simulator class](https://github.com/facebookresearch/habitat-sim/blob/8d5d51e9019da061f561c1cc07a85908a5f62ef2/habitat_sim/simulator.py#L246). You can see an example of this in the [demo_runner script](https://github.com/facebookresearch/habitat-sim/blob/8d5d51e9019da061f561c1cc07a85908a5f62ef2/examples/demo_runner.py#L103).

Regarding the C++ viewer example: you can add an object with the `'o'` key and then toggle display of its bounding box with the `'b'` key. Having downloaded the [test object dataset](http://dl.fbaipublicfiles.com/habitat/objects_v0.1.zip) and extracted it to `data/objects/`, running `build/viewer --enable-physics data/scene_datasets/habitat-test-scenes/van-gogh-room.glb` from the `habitat-sim` directory results in (agent position/orientation will vary):

[image]

Then pressing the `'o'` key (exact object will vary):

[image]

Then pressing the `'b'` key:

[image]

If this is not the case for you, please check your terminal output for possible error messages which may indicate the issue.

picksh, 2019-12-03:
Thank you for your reply! It works for me.
Also, is it possible for me to set the position of added object when using the C++ viewer? I think all added objects' positions are set to be same and have random(?) orientations in the  C++ viewer example and I would like to know how to  change this setting.

aclegg3, 2019-12-04:
Sure, take a look at the [`addObject`](https://github.com/facebookresearch/habitat-sim/blob/dedd5541dc7123068bbf65cde93326c528a2dff5/src/utils/viewer/viewer.cpp#L233) function in the viewer code. Here we add a new object to the scene and then set the object's position and orientation using `PhysicsManager` functions. You can call these directly given an object's id which is pushed into the viewer's `objectIDs_` vector when the object is added. 

For an example of setting object position this way, see the viewer's [`wiggleLastObject`](https://github.com/facebookresearch/habitat-sim/blob/dedd5541dc7123068bbf65cde93326c528a2dff5/src/utils/viewer/viewer.cpp#L323) function.

picksh, 2019-12-05:
Thank you very much!  It solved my problem.

closed 2019-12-05 by picksh