facebookresearch/habitat-sim #367: How to insert objects and or humans

ronjamoller, 2019-12-02:
Hi, i am working on a navigation task that includes maneuvering around humans and since many of the frameworks for human interaction are very simplistic I would really love to use habitat. However, I have found very little material on how to insert third party static models (basically only one github issue on the shading of inserted objects), never mind animated ones. Is there material or tutorials I have missed that cover this customization option or maybe an example that i can follow ?
Thanks

aclegg3, 2019-12-02:
Thanks for this feature request! Animated models are not currently supported by Habitat-sim. There is, however, a framework in place to add/remove and set transformations for rigid/static objects through the `PhysicsManager` class. Examples of this functionality exist for python and C++ as described by the [README:Testing](https://github.com/facebookresearch/habitat-sim#testing) section.

You can also add your own objects to the scene very simply by creating a `object.phys_properties.json` file which references your mesh and adding it to the `default.phys_scene_config.json` file. You can follow the example of the [test object dataset](http://dl.fbaipublicfiles.com/habitat/objects_v0.1.zip) we provide.

If you would like to expand on your desired use case and features you would appreciate, please feel free to do so in this thread. We will happily take these into consideration when planning our development efforts. 👍

ronjamoller, 2020-01-08:
Hi again, thank you for the reading material and examples, i have successfully inserted my desired objects into the simulation. My next step is to get the exact pixel positions of my inserted object or parts of it (like a human face/eyes), my current understanding is that i could add this information as a sensor but I am a bit unsure about where exactly this should be added. I have looked at the register new_sensors_and measures code and the examples and I admit I am still very unsure on such a basic level that i dont know if it is smarter to implement the change in the python part or the c++ part. It is my understanding that the c++ core could provide the actual information (a point to screen coordinate functionality) but the only way to actually add a sensor seems to be a python script. I have posted a question about that in the habitat-api issues but received now answer, maybe it is not correct there. In addition to implementing the sensor I would of course like to test it in an interactive environment - is this only possible in c++ ? The testing sections seems to imply that but in the habitat docs there is a quickstart example that provides the w a s d interaction with only a python script. Im sorry if these questions are very basic, maybe the screen coordinate functionality already exists and I have not found it yet. 
Thank you
ronja

aclegg3, 2020-01-08:
Hey @ronjamoller, I'm glad you've gotten things working for your use case! I'd be interested to see the results if you'd like to share at some point. 😄 

I see your issue in Habitat-api: https://github.com/facebookresearch/habitat-api/issues/265#issue-537569899

Regarding the interactive python quick-start script: this is not versioned in codebase and could be more complete. Good point. I've opened a specific feature request issue for this. 👍 #418 

Regarding your sensor needs. You could create a new sensor for this and we are in the process of improving the overall sensor API to make this more obvious. While we do not yet have any built-in functions for 3D point to screen point project, all the necessary transformation and projection matrices for the camera should be exposed both in C++ and python, so where you develop is up to you. You might first start by implementing the point projection for your desired features and then looking into a custom sensor. For a quick test you could modify the image produced by `simulator.get_observation` by drawing your projected 2D points on the image before displaying or saving it (either with opencv or PIL).

You might look into [`RenderCamera` python bindings](https://github.com/facebookresearch/habitat-sim/blob/823111e431df78bdb8c1224816912eb5aa74a08a/src/esp/bindings/GfxBindings.cpp#L38)
You can get access to the RenderCamera for the scene with:
```
render_camera = simulator._sim.get_active_scene_graph().get_default_render_camera()
```

ronjamoller, 2020-01-13:
Hi again, I am plodding along (my biggest difficulty is the level of abstraction) and I have decided that for simplicity's sake I would like to develop in Python so I tried to insert objects into the scene via Python but it seems to me like this only works with the physics option on ? I would like to avoid installing bullet physics, i was not successful the last time i tried and i don't really need any physics for my objects anyway.  If I understand correctly my issue is that the object library is not populated (checked with sim.get_physics_object_library_size()) when the physics option is deactivated but I have run the c++ simulation successfully with inserting objects - is it possible to avoid pybullet and still insert objects via python ?
A more general question - while going through the code, both the python sources and the c++ sources repeatedly mention that  python does not get ownership ob the semantic scene graph and that there is some memory handling issue, will i get in trouble if i access the scene graph and manipulate the positions of my agent/object ?
Thank you

aclegg3, 2020-01-13:
You can absolutely run the simulator with physics enabled, but bullet not installed. This will default to kinematic control (transform setting) only. You won't get contact checking once that feature lands either. To clarify: you must run with physics enabled to load and insert objects, but it does not need to be built with bullet.

 For portability with dynamics (using bullet physics) down the line, you can directly call the `simulator.set_translation()` and `simulator.set_rotation()` functions to manipulate the object state, instead of transforming the nodes directly. 

For example: 
```
sim = habitat_sim.Simulator(cfg)
object_lib_size = sim.get_physics_object_library_size()
rand_obj_index = random.randint(0, object_lib_size - 1)
object_id = sim.add_object(rand_obj_index)
old_object_position = sim.get_translation(object_id)
new_object_position = np.array([0.0,0.0,0.0])
self._sim.set_translation(new_object_position, object_id)
```

That said, you can also directly manipulate the SceneGraph if you choose, no memory issue here. What we mean is that if the `SceneGraph` is deleted the `SceneNode`s will be also, you don't get to keep them. 😉

ronjamoller, 2020-01-15:
Thanks for all the helpful feedback, I have inserted my objects as needed and the screen projection matches. Unfortunately i was not successful so far in getting the setup to run from an environment class (like in the habitat api demo), only when i create and manipulate a simulation like you suggested. Since the long term is to train an agent my guess is that i should use the environment class that manages the simulation internally - problem is i have not found a way to enable physics with the way the environment configs are handled. In addition to being just generally different to the simulation configs it seems there are no physics flags ? I have tried to reinitialize the simulation by accessing it via env._sim._sim but it seems this simulation takes a different type of config from "vanilla" simulation configs as well so i`m unsure how to proceed. 
tldr: how can i enable physics during the setup of the environment and if that's impossible how can i correctly reinitialize the internal simulation ?
thank you

ronjamoller, 2020-01-16:
Addition to the previous post: my long term goal is to manage the pixel positions that i compute with a sensor and the only examples for adding sensors that i have found so far is using the environment class. In summary i feel like i have no choice but to use the environment class but i cant configure it the way i want re: physics.
thank you

ronjamoller, 2020-01-22:
Update: there was a naming mismatch in the names from different configs which i mistook for general incompatibility. I have corrected everything and now i can reinitialize the env._sim._sim to enable physics and all is well. Sorry for the confusion, i have removed my issue on habitat_api as well

aclegg3, 2020-01-22:
Glad to hear that you have things up and running. 👍  
Any other issues, or should we close this thread?

ronjamoller, 2020-01-22:
the only thing that is left for me to look at would be how to implement "animations" but i guess for now i will just manipulate translations of the individual components with a script and hope it doesn't slow me down too much but that is out of scope for this issue i think so it can be closed. Thank you for all the help !

closed 2020-01-23 by ronjamoller