facebookresearch/habitat-sim #369: How to get Map information in python?

Kyridiculous2, 2019-12-04:
## ❓ Questions and Help
Hi, your projects are pretty useful and amazing!
I plan to use your environments to do some work, where i need to sample a few views in the scene randomly/under controlled, so the ground truth map may need to be known. Specifically i want to sample a feasible(not occupied by objects) camera pose in the scene and visualize the location in the (top-down) map , any ideas on how I can achieve that? I've read your readme, its all about agents, and I can't find some related things.  I'll appreciate it very much if you can give me some advice!

aclegg3, 2019-12-04:
If you want to sample locations in the scene which are not occupied by objects, you may be interested in our Bullet physics integration. While not in master branch yet, we have a prototype branch discussed in #345 (see [this comment](https://github.com/facebookresearch/habitat-sim/issues/345#issuecomment-555280363)) which approximates a quad-rotor with kinematic control and discrete collision checking against a bounding box primitive shape.

This may be exactly what you are looking for. By loading the scene with physics enabled in this branch, you can call the `Simulator.contact_test` function to see if any scene geometry or other objects are colliding with an imported shape. You could also generate a mesh (or meshes) which approximates the camera frustum you want to check and use it to cull camera placements which result in scene collision with this frustum.

Hopefully this helps, let us know if you have more questions or feature requests.

erikwijmans, 2019-12-04:
If the current navigability checking suites your needs, habitat-api has functionality to generate a top-down occupancy grid.  This example will make a video of what that looks like (there are additional things drawn on the top-down occupancy grid): https://github.com/facebookresearch/habitat-api/blob/master/examples/shortest_path_follower_example.py

Kyridiculous2, 2019-12-05:
Thank you for your being willing to help! 
After more considerations i decide to start from the 4 Dof case, which equals navigation plus a camera  pitch. So how can i pitch the camera?
@erikwijmans  I check the map generation codes and find that you build the map by sampling. Does sample_navigable_point only sample a navigable point on the floor(navmesh)? so what's is_navigable used for? I notice a height is added on the point,does it also take into account the agent collision capsule ?

So I think I can just use this to sample point in the scene, random set camera poses,then get images?Also, any ideas on how to control the sampled points to be equally distributed in the scene ?

erikwijmans, 2019-12-05:
>  I check the map generation codes and find that you build the map by sampling. Does sample_navigable_point only sample a navigable point on the floor(navmesh)? so what's is_navigable used for?

1. Yes `sample_navigable_point` only returns navigable points, but those won't be strictly on the floor.  Things like the top of a table may get marked as navigable.

2.  The sampling is done to get a bounding box of the environment, the map itself is computed via a grid sweep in that box, hence `is_navigable`.  The height is given to points as environments can be multi-level, the map is just a single slice.

> So I think I can just use this to sample point in the scene, random set camera poses,then get images?

Yep, that'll work

> Also, any ideas on how to control the sampled points to be equally distributed in the scene ?

Not sure what you mean here, but `sample_navigable_point` samples points uniformly at random. (the underlying function in the simulator is https://aihabitat.org/docs/habitat-sim/habitat_sim.nav.PathFinder.html#get_random_navigable_point)

Kyridiculous2, 2019-12-05:
Thanks a lot, I'll try on that! 
So how can i change the camera pitch? I find the apis  in the demo can only turn left/right.

erikwijmans, 2019-12-05:
You can add the loop_up and look_down actions (all actions: https://aihabitat.org/docs/habitat-sim/habitat_sim.agent.html) example spec using them: https://github.com/facebookresearch/habitat-sim/blob/master/tests/test_controls.py#L154

You can also set the camera orientation directly.

```
state = agent.state
state.sensor_states[<sensor_id>].rotation = np.quaternion(.....)
agent.state = state
```
Quaternion helper functions: https://aihabitat.org/docs/habitat-sim/habitat_sim.utils.common.html

Kyridiculous2, 2019-12-05:
Thank u!! 
I think I should check the docs more thoroughly next time I have questions lol

Kyridiculous2, 2019-12-05:
Now I'm trying to generate top-down map from the habitat-api.utils, why the coordinate_min/max should be constant?  Shall they vary with different scenes?
[image]

I do get bugs when try on a scene, I got weird x_range , y_range and no valid points in them

closed 2022-02-10 by aclegg3