facebookresearch/habitat-sim #373: Dockerfile Version 1.3: Aborted (core dumped)

AAdeleye, 2019-12-06:
## ❓ Questions and Help

I am using the Dockerfile given, with git checkout version "9129fb2d35569cbdc07ac784fd58d6287240ad8b" for HabitatSim and "55fc9353a81d9a2a899dcd7bc08c8be39d228b8c" for Habitat-api

When I try running the basic example, it seems to fail. 
Before updating to version 1.3 of the github checkout, everything worked well? 

[image]

There was a [similar issue here](https://github.com/facebookresearch/habitat-sim/issues/135#issue-477741653) but not super helpful.

erikwijmans, 2019-12-06:
Can you turn back on logging for the simulator? i.e.
```
unset MAGNUM_LOG
unset GLOG_minloglevel
```

AAdeleye, 2019-12-10:
[image]

I spent some trying to debug it based on this error message. It looks like it is an Nvida issue. I will comment again if I figure it out but please let me know if you have any suggestions.

erikwijmans, 2019-12-11:
I didn’t notice this earlier, using python3 maybe be cause the wrong python exe to be picked up. Make sure you are in the correct conda env (the same on you installed the simulator in) and use just python (if you are in the correct conda env, that’ll be python 3)

AAdeleye, 2019-12-13:
Thank you for your help, that doesn't seem to be the issue though. 

I traced it to the method "get_observations", line 417 in simulator.py. 
It seems to be an issue with my conda enviroment set up here is the path to the file: "/opt/conda/envs/habitat/lib/python3.6/site-packages/habitat_sim-0.1.3-py3.6-linux-x86_64.egg/habitat_sim/simulator.py" 

I am going to revert back to the challenge specific version on the dockerfile.

closed 2019-12-13 by AAdeleye