facebookresearch/habitat-sim #374: Add support for importing semantics from https://3dscenegraph.stanford.edu for use with gibson dataset

msbaines, 2019-12-10:
## 🚀 Feature

We want to be able to use the semantic dataset from https://3dscenegraph.stanford.edu/ with the scene dataset from http://gibsonenv.stanford.edu/

msbaines, 2019-12-10:
The 3dscenegraph semantic dataset is currently limited to gibson_tiny. However, semantics for gibson_medium are expected to be release soon.

msbaines, 2019-12-10:
The mesh used for semantics is different from the mesh used for Habitat. The coordinate system is also different but both meshes. The Y and Z axis are switched but the origin is the same. We should be able to generate a semantic ply mesh from the origin .obj mesh by transformed vertex coordinates.

msbaines, 2019-12-11:
The semantic data from 3DSceneGraph is available via an [npz](https://docs.scipy.org/doc/numpy/reference/generated/numpy.savez.html) file.

To access the data:

`data = np.load(npz_path, allow_pickle=True)['output'].item()`

This will return the following dictionary:

`dict_keys(['building', 'room', 'object', 'camera', 'panorama'])`

### building

```
>>> pprint.pprint(data['building'])
{'floor_area': 35.04333970662052,
 'function': 'residential',
 'gibson_split': 'tiny',
 'id': 2,
 'name': 'Allensville',
 'num_cameras': 26,
 'num_floors': 1,
 'num_objects': 33,
 'num_rooms': 11,
 'object_inst_segmentation': array([[0.],
       [0.],
       [0.],
       ...,
       [0.],
       [0.],
       [0.]]),
 'object_voxel_occupancy': array([[0.],
       [0.],
       [0.],
       ...,
       [0.],
       [0.],
       [0.]]),
 'reference_point': array([0., 0., 0.]),
 'room_inst_segmentation': array([[10.],
       [ 8.],
       [ 8.],
       ...,
       [11.],
       [11.],
       [11.]]),
 'room_voxel_occupancy': array([[0.],
       [0.],
       [0.],
       ...,
       [0.],
       [0.],
       [0.]]),
 'size': array([9.76260114, 8.85856447, 2.50682107]),
 'volume': 201.63530725199752,
 'voxel_centers': array([[-0.98990329, -0.97658977, -0.02014603],
       [-0.98990329, -0.97658977,  0.07985397],
       [-0.98990329, -0.97658977,  0.17985397],
       ...,
       [ 8.71009671,  7.82341023,  2.27985397],
       [ 8.71009671,  7.82341023,  2.37985397],
       [ 8.71009671,  7.82341023,  2.47985397]]),
 'voxel_resolution': array([98, 89, 26]),
 'voxel_size': 0.1}
```

### room

```
>>> pprint.pprint(data['room'])
{1: {'floor_area': 8.73437848991798,
[... 156 lines cut ...]
             'object_instance': array([[0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       ...,
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0]], dtype=int16)},

 ...

 'p000026': {'object_class': array([[0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       ...,
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0]], dtype=int16),
             'object_instance': array([[0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       ...,
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0]], dtype=int16)}}
```

msbaines, 2019-12-11:
Was hoping to load the .npz file in C++ using [cnpy](https://github.com/rogersce/cnpy), but the .npz contains pickled data which cnpy can't handle. I could potentially handle the pickled data using [http://www.picklingtools.com/](http://www.picklingtools.com/).

Alternatively, I could do all the processing in python but the python tools for writing a mesh are cumbersome and likely slow. So for now,  I think I will write a python script to convert the data I need into a format that can be easily loaded in C++ and then do the processing in C++.

msbaines, 2019-12-11:
The semantic mask information is located in:

`data['building']['object_inst_segmentation']`

which an array that is the same size as the number of faces. Each element in the array contains the semantic object_id for that face. If we don't have semantic information for that face, the id is 0.

The format of the array:

```
>>> type(data['building']['object_inst_segmentation'])
<class 'numpy.ndarray'>
>>> type(data['building']['object_inst_segmentation'][0])
<class 'numpy.ndarray'>
>>> data['building']['object_inst_segmentation'][0]
array([0.])
>>> data['building']['object_inst_segmentation'][0][0]
0.0
>>> type(data['building']['object_inst_segmentation'][0][0])
<class 'numpy.float64'>
```

msbaines, 2019-12-11:
We should be able to write out the object ids using the following code:

```
f = open("out.bin", "wb")
object_ids = data['building']['object_inst_segmentation']
f.write(object_ids.astype(np.int16).tobytes())
f.close()
```

msbaines, 2019-12-16:
Bounding box information:

Looking at the object schema:

```
 33: {'action_affordance': ['sit at',
                            'lay on',
                            'pick up',
                            'move',
                            'clean',
                            'set',
                            'decorate'],
      'class_': 'dining table',
      'floor_area': 3.473003668596787,
      'id': 33,
      'location': array([4.48357247, 6.70686119, 0.5614044 ]),
      'material': ['wood', None],
      'parent_room': 8,
      'size': array([1.14685995, 0.68447991, 0.6124021 ]),
      'surface_coverage': 1.3836484369355602,
      'tactile_texture': None,
      'visual_texture': None,
      'volume': 0.27629888509653355}}
```

location and size may be the axis-aligned bounding box. This will have to be verified.

msbaines, 2019-12-16:
Semantics seem to be working with the following transformation:

x1 = x0
y1 = -z0
z1 = y1

ybgdgh, 2020-09-15:
Hello, I want to know can we get the room's centers and bounding boxes from habitat in Gibson dataset? I used the 3Dscenegraph as gibson semantics but only get the SemanticObject class. Thanks!