facebookresearch/habitat-sim #375: Cartoonized simulated RGB image

BryanBo-Cao, 2019-12-10:
## ❓ Questions and Help

How to get the Cartoonized simulated RGB image (top left in red circle) shown in the repo home page? Is it just from a specific dataset?

[image]

I tried to search the config in the source code but with no luck. Only found COLOR, DEPTH and SEMANTICS. Any help would be appreciated.

aclegg3, 2019-12-10:
Yes, this is just a model from a particular dataset. There is no "cartoonized" render mode.

BryanBo-Cao, 2019-12-10:
@aclegg3 Did you happen to know which dataset? Matterport3D and Gibson use real images.

aclegg3, 2019-12-11:
@msavva ?

dhruvbatra, 2019-12-11:
It's from a dataset that is no longer publicly available. We are unable to comment further because of an ongoing litigation.

closed 2019-12-13 by BryanBo-Cao