facebookresearch/habitat-sim #376: from habitat_sim.utils.common import d3_40_colors_rgb

BryanBo-Cao, 2019-12-10:
## 🐛 Bug

This might be a trivial bug, the method ```d3_40_colors_rgb()``` shown in the [demo](https://aihabitat.org/docs/habitat-api/habitat-sim-demo.html) seems to have been moved from
```
from habitat_sim.utils import d3_40_colors_rgb
```
to
```
from habitat_sim.utils.common import d3_40_colors_rgb
```
From my end, I have to import it from ```habitat_sim.utils.common```.
Maybe the document has not been updated yet.

aclegg3, 2019-12-11:
Thanks, this is just the result of some refactoring. Likely need to update the demo to reflect this. @erikwijmans?

mosra, 2019-12-17:
:thinking: are we compiling / running the demo code on the CI? Is it possible to do that? Would be great to ensure all documentation code snippets are always up-to-date.

This *might* be a result of the module reorganization we did for the new documentation.

erikwijmans, 2019-12-17:
Yeah... these are the demo notebooks and because of how poorly git works with jupyter notebooks, we've put those elsewhere, but they will fall out of sync :/

mosra, 2019-12-18:
Did a quick google search:

- https://towardsdatascience.com/version-control-with-jupyter-notebooks-f096f4d7035a
- https://nextjournal.com/schmudde/how-to-version-control-jupyter
- https://mg.readthedocs.io/git-jupyter.html

The easiest from these seems to be to strip the output from all notebooks, saving only the code. A downside is that the output will not be shown in GitHub's Jupyter renderer, but that's not done right now anyway :) Looks like it has to be done manually tho, so maybe a check could be added to the CI runner to ensure the output is stripped again after an ipynb file gets updated.

wojga, 2020-03-18:
For the record the fix is to add `.common`:
`from habitat_sim.utils.common import d3_40_colors_rgb`

closed 2022-08-23 by aclegg3