facebookresearch/habitat-sim #388: ModuleNotFoundError: No module named 'magnum.scenegraph'

MarchingVoxels, 2019-12-18:
## ❓ Questions and Help
import habitat_sim
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.3-py3.6-linux-x86_64.egg/habitat_sim/__init__.py", line 14, in <module>
    from habitat_sim.nav import *
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.3-py3.6-linux-x86_64.egg/habitat_sim/nav/__init__.py", line 1, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (
ImportError: ModuleNotFoundError: No module named 'magnum.scenegraph'

erikwijmans, 2019-12-20:
This happens when magnum falls out of sync with habitat-sim.  The system-wide install of habitat-sim isn't well tested and sometimes it can do odd things (we highly recommend using a conda environment).  Doing `pip3 install build/deps/magnum-bindings/src/python/` should a force a reinstall of magnum.

closed 2019-12-20 by erikwijmans

fraazor, 2020-10-14:
I am dealing with the same problem when running the gt-generation command:

python occant_baselines/generate_topdown_maps/generate_occant_gt_maps.py     --config-path occant_baselines/generate_topdown_maps/configs/occant_gt/gibson_${SPLIT}.yaml     --save-dir data/datasets/exploration/gibson/v1/${SPLIT}/occant_gt_maps     --global-map-size 961

Somehow the magnum.scenegraph is not found despite being installed/reinstalled. I'm using a new conda environment.

Error:
...
  File "/home/usr/miniconda3/envs/OccAnt/lib/python3.6/site-packages/habitat_sim-0.1.5-py3.6-linux-x86_64.egg/habitat_sim/errors.py", line 2, in <module>
    import magnum.scenegraph
ModuleNotFoundError: No module named 'magnum.scenegraph'

Unfortunately I could not found much advice on this elsewhere.

erikwijmans, 2020-10-14:
Can you try the command above to force a reinstall of magnum?

fraazor, 2020-10-14:
Thank you for your quick response! I had to uninstall habitat_sim before reinstalling magnum before re-installing magnum which made it work.

KenaHemnani, 2020-11-18:
Hii, I am finding the same issue. @erikwijmans I also tried the solution is suggested . But, for be magnum-bindings folder is in "/home/hemnani1/DIR/habitat-sim/src/deps/magnum-bindings/"   . That is not helping me what shall I do? Can someone please help me?