facebookresearch/habitat-sim #390: Compute absolute AABB for mesh components in a static scene.

bigbike, 2019-12-18:
## 🚀 Feature
The scene consists of many mesh components. Compute a AABB  for each component in the world coordinate frame.

Note this AABB is different from the AABB computed in the component's local coordinate frame.

Also, if you rotate the local AABB to the world coordinate frame, which becomes an OBB, then you build an AABB of this OBB. This is an "AABB approximation", NOT what we request here.

## Motivation
To accelerate frustum culling.

More details, see discussion: #380 

## Pitch

## Alternatives

## Additional context

Skylion007, 2020-10-05:
@bigbike Can we close this issue or add more detail?

bigbike, 2020-10-05:
@Skylion007 Yes, we can. It has been resolved.

closed 2020-10-05 by bigbike