facebookresearch/habitat-sim #396: Magnum-bindings does not install correctly

smorad, 2019-12-19:
## 🐛 Bug

Magnum package does not always get installed

## Command
`sudo python3 setup.py install`

## To Reproduce

This occurred during an install/reinstall. I initially built `stable`, then built`master`

1. sudo python3 setup.py install
```
...
Processing ./build/deps/magnum-bindings/src/python
  Requirement already satisfied (use --upgrade to upgrade): magnum==0.0.0 from file:///home/smorad/code/cambridge/habitat-sim/build/deps/magnum-bindings/src/python in /usr/local/lib/python2.7/dist-packages
...
```
2. python3 examples/example.py
```
(habitat) smorad@smorad-ThinkPad-T490:~/code/cambridge/habitat-sim$ python3 examples/example.py -scen
e data/scene_datasets/habitat-test-scene<!-- A clear and concise description of what you expected to happen. -->
s/skokloster-castle.glb                       
Traceback (most recent call last):                        
  File "examples/example.py", line 10, in <module>                                
    import demo_runner as dr
  File "/home/smorad/code/cambridge/habitat-sim/examples/demo_runner.py", line 15, in <module>
    from settings import default_sim_settings, make_cfg                                              
  File "/home/smorad/code/cambridge/habitat-sim/examples/settings.py", line 5, in <module>
    import habitat_sim                                  
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.3-py3.6-linux-x86_64.egg/habitat_sim/$
_init__.py", line 14, in <module>                 
    from habitat_sim.nav import *
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.3-py3.6-linux-x86_64.egg/habitat_sim/$
av/__init__.py", line 1, in <module>                                           
    from habitat_sim._ext.habitat_sim_bindings import (                                   
ImportError: ModuleNotFoundError: No module named 'magnum'
```
3. `pip3 install build/deps/magnum-bindings/src/python/ # This fixes the error`

## Expected behavior

The example code to run without crashing

## Additional context
Followed install guide, running in conda

## System Info
ENVIRONMENT INFO:
Platform: Linux-5.0.0-36-generic-x86_64-with-Ubuntu-18.04-bionic
Machine: x86_64
Processor: x86_64
Libc version: glibc 2.25
Mac version: 
Python version: 3.6.9
Architecture: 64bit ELF
Win version:    
System OS: Linux
Release: 5.0.0-36-generic
Version: #39~18.04.1-Ubuntu SMP Tue Nov 12 11:09:50 UTC 2019
Operational System: linux
GCC version: gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
CMAKE version: cmake version 3.10.2
NVIDIA-SMI: 
Pip packages versions:
numba==0.46.0
numpy==1.17.4
numpy-quaternion==2019.12.11.22.25.52
tqdm==4.40.2
Conda packages versions:

erikwijmans, 2019-12-20:
> Followed install guide, running in conda
When running things in conda, you should not needs to use `sudo` nor should you need to do `python3` to get the correct version of python (i.e. just `python` should get you python 3.6.9)

smorad, 2020-01-10:
I'm going to chalk this up to user error (ie somehow installing packages outside of the conda virtualenv). I think it's fair to close this.

erikwijmans, 2020-01-10:
Sounds good, feel free to re-open if needed.

closed 2020-01-10 by erikwijmans