facebookresearch/habitat-sim #402: how to get/set camera focal length, pixel size etc'

davidkarl, 2019-12-24:
hello,
can anyone please help me understand how do i get/set the effective focal length, pixel size, and number of pixels in the simulation?
i'm using the default pinhole camera and am trying to get stereo images with depth AND CONVERT THEM TO DISPARITY. in order to do that i need know (and maybe change) the default "camera" configurations.

can anybody help?
thank you,
david karl

BryanBo-Cao, 2019-12-24:
You may check this demo:https://facebookmicrosites.github.io/habitat-website/docs/habitat-api/habitat-sim-demo.html

You can set the number of pixels by
```
sim_settings = {
    "width": 256,  # Spatial resolution of the observations
    "height": 256,
    "scene": test_scene,  # Scene path
    ...
}
```
and sensor positions by
```
def make_cfg(settings):
    sim_cfg = habitat_sim.SimulatorConfiguration()
    sim_cfg.gpu_device_id = 0
    sim_cfg.scene.id = settings["scene"]

    # Note: all sensors must have the same resolution
    sensors = {
        "color_sensor": {
            "sensor_type": habitat_sim.SensorType.COLOR,
            "resolution": [settings["height"], settings["width"]],
            "position": [0.0, settings["sensor_height"], 0.0],
        },
        "depth_sensor": {
            "sensor_type": habitat_sim.SensorType.DEPTH,
            "resolution": [settings["height"], settings["width"]],
            "position": [0.0, settings["sensor_height"], 0.0],
        },
        "semantic_sensor": {
            "sensor_type": habitat_sim.SensorType.SEMANTIC,
            "resolution": [settings["height"], settings["width"]],
            "position": [0.0, settings["sensor_height"], 0.0],
        },
    }
```

erikwijmans, 2019-12-24:
For any SensorSpec in the `AgentConfiguration.sensor_specifications` list, the FOV will get picked up from the `.parameters['hfov']` field.

davidkarl, 2019-12-27:
hello everybody and thank you for your responses.
however as far as i understand it the question is still not answered.

as i understand it:
1. width and height are the number of pixels used the represent your FOV
2. hfov is your horizontal field of view

as i said - my main mission is to translate precisely from depth to disparity (shift in terms of pixels) from a left pixel to a right pixel.
the equation for that is: disparity=(focal_length * baseline / pixel_size) / depth
and since assuming hfov is in degrees, N is number of horizontal pixels, and baseline is known then ->
hfov/2 = arctan(half_screen_size/distance_from_pinhole_to_screen) ->
hfov/2 = arctan((N/2)*pixel_size/focal_length)
->
pixel_size/focal_length = 2/N * arctan(hfov/2)

and so if i'm right i might be able to calculate disparity from depth as it's enough to know pixels_size/focal_length for that....
however i would still love it if anybody know what it the actual focal_length / distance from pinhole to "screen".

erikwijmans, 2019-12-27:
Yeah, `hfov` is the horizontal field of view.  Given the hfov and width, you can calculate the focal length.  This blog goes over it reasonably well: http://paulbourke.net/miscellaneous/lens/ 

Since these are virtual cameras, everything will be in units of pixels instead of mm, thus the pixel size is just 1.  So you already had the equation for focal length! `1/focal_length = 2/N * tan(hfov/2)` (set pixel size to 1 and correct the `arctan` typo).

aclegg3, 2021-04-15:
Issue is inactive and the question appears to be answered.

closed 2021-04-15 by aclegg3