facebookresearch/habitat-sim #404: Observations not changed after resetting agent's position

BryanBo-Cao, 2019-12-24:
## ❓ Questions and Help
Based on this [demo](https://facebookmicrosites.github.io/habitat-website/docs/habitat-api/habitat-sim-demo.html), I tried to display the observations after resetting an agent's position directly, but it looks like it does not change.

### Start Position
```
this_agent = sim.get_agent(0)
pathfinder = sim.pathfinder

path = habitat_sim.ShortestPath()
path.requested_start = this_agent.get_state().position
print("path.requested_start: %s " % path.requested_start)
print("this_agent.get_state().position: %s" % this_agent.get_state().position)

print("Start image:")
observations = sim.get_sensor_observations()
rgb = observations["color_sensor"]
semantic = observations["semantic_sensor"]
depth = observations["depth_sensor"]

display_sample(rgb, semantic, depth)
```
### Observation from the Start Position
[image]

### End Position
```
random_navigable_point = pathfinder.get_random_navigable_point()
path.requested_end = random_navigable_point

pathfinder.find_path(path)
while path.geodesic_distance == np.inf or path.geodesic_distance < 20:
    random_navigable_point = pathfinder.get_random_navigable_point()
    path.requested_end = random_navigable_point
    pathfinder.find_path(path)
    
pathfinder.find_path(path)
print()
print("path.requested_end: %s " % path.requested_end)
print("path.geodesic_distance: %s" % path.geodesic_distance)
this_agent_state = this_agent.get_state()
this_agent_state.position = path.requested_end
this_agent.set_state(this_agent_state)
print("this_agent.get_state().position: %s" % this_agent.get_state().position)

print("End image:")
# observations = sim.step("stop")
observations = sim.get_sensor_observations()
rgb = observations["color_sensor"]
semantic = observations["semantic_sensor"]
depth = observations["depth_sensor"]

display_sample(rgb, semantic, depth)
```

### Observation from the End Position
[image]

Any better way to do that? Any help would be appreciated.

BryanBo-Cao, 2019-12-24:
I ended up with initializing goal_agent and check the observation, then I intialize the agent back to the original one:
```
goal_agent = sim.initialize_agent(sim_settings["default_agent"])
goal_agent_state = habitat_sim.AgentState()
goal_agent_state.position = path.requested_end
goal_agent.set_state(goal_agent_state)
```
Not sure if there's a elegent way to do that.

erikwijmans, 2019-12-24:
The sensors have their own states that can be individually manipulated.  If you'd like the sensor states to be recalculated wrt. to the agent's body, you can unset their states in the agent state, i.e. `this_agent_state.sensor_states = {}`

closed 2019-12-24 by erikwijmans