facebookresearch/habitat-sim #414: can not run habitat on web

picksh, 2020-01-05:
## ❓ Questions and Help
Hello, I have followed the instruction of webGl in README but when I open http://127.0.0.1:8000/build_js/esp/bindings_js/bindings.html after running webserver with `python -m http.server 8000 --bind 127.0.0.1`, I get the following results:

#### with google-chrome:
website:
 "Downloading... 0/4".
and
`wasm streaming compile failed: TypeError: Failed to execute 'compile' on 'WebAssembly': Incorrect response MIME type. Expected 'application/wasm'.
falling back to ArrayBuffer instantiation`

terminal:
`Serving HTTP on 127.0.0.1 port 8000 (http://127.0.0.1:8000/) ...`

####  with firefox:
website:
 "Downloading... 0/4".
and
`wasm streaming compile failed: TypeError: Response has unsupported MIME type
falling back to ArrayBuffer instantiation`

terminal:
`Serving HTTP on 127.0.0.1 port 8000 (http://127.0.0.1:8000/) ...
127.0.0.1 - - [05/Jan/2020 15:00:38] "GET /build_js/esp/bindings_js/bindings.html HTTP/1.1" 200 -
127.0.0.1 - - [05/Jan/2020 15:00:38] "GET /build_js/esp/bindings_js/WebApplication.css HTTP/1.1" 200 -
127.0.0.1 - - [05/Jan/2020 15:00:38] "GET /build_js/esp/bindings_js/WindowlessEmscriptenApplication.js HTTP/1.1" 200 -
127.0.0.1 - - [05/Jan/2020 15:00:38] "GET /build_js/esp/bindings_js/hsim_bindings.js HTTP/1.1" 200 -
127.0.0.1 - - [05/Jan/2020 15:00:38] "GET /build_js/esp/bindings_js/bundle.js?c44b04faecd6eb3caae5 HTTP/1.1" 200 -
127.0.0.1 - - [05/Jan/2020 15:00:38] "GET /build_js/esp/bindings_js/hsim_bindings.wasm HTTP/1.1" 200 -
`

The downloading process seems to fail and is there any solution?  Thank you!

apsdehal, 2020-01-06:
Hi,
Can you try `http://localhost:8000/build_js/esp/bindings_js/bindings.html` and make sure that you have Habitat test data installed?

picksh, 2020-01-07:
It works, thank you very much!

closed 2020-01-07 by picksh