facebookresearch/habitat-sim #416: The cmake error when installing Habitat

liuxinzhu0353150307, 2020-01-08:
## ❓ Questions and Help

I was trying to install it in windows10 system.  When I run the command "python setup.py install", I get the error message "CMake Error: The source directory "E:habitat-simsrc" does not exist." 
The path of the habitat-sim project is in E:/habitat-sim. There does exist the file "E:habitat-simsrc" actually.
So I want to know how can I solve the cmake error. Thanks a lot.

wannabeOG, 2020-01-08:
Running into the same error, 

I ran the command "python setup.py install --with-cuda" and the process terminated with a similar error

Here's the console output for your reference: https://pastebin.com/dteadcH1

erikwijmans, 2020-01-08:
See #84

closed 2020-01-08 by erikwijmans