facebookresearch/habitat-sim #417: Running into an error with both the interactive and non-interactive viewer

wannabeOG, 2020-01-08:
## ❓ Questions and Help

### Device Specifications

OS: Ubuntu 18.04
GPU: NVIDIA Geforce 1050ti, Driver Version: 390.129
CUDA 9.0

### Steps to reproduce the error

1. Followed the instructions detailed in the README using the ``python setup.py install --with-cuda`` to build in a virtual environment
2. To test the interactive viewer I ran the command ``./build/viewer /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`` which resulted in this error message: 
Platform: can't get Xft.dpi property for virtual DPI scaling, falling back to physical DPI
GLX: GLX extension not found
Platform::GlfwApplication::tryCreate(): cannot create a window with core OpenGL context, falling back to compatibility context
GLX: No GLXFBConfigs returned
GLX: Failed to find a suitable GLXFBConfig
Platform::GlfwApplication::tryCreate(): cannot create a window with OpenGL context

3. Tried to test the non-interactive viewer as well by running the command ``python examples/example.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb``  which resulted in this error message:
sim_cfg.physics_config_file = ./data/default.phys_scene_config.json
Initialized Sensor Spec: 
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
Platform::WindowlessGlxContext: no supported framebuffer configuration found
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0108 21:38:01.951243 15016 WindowlessContext.cpp:201] Check failed: glxCtx_.isCreated() [GLX] Failed to created headless glX context
*** Check failure stack trace: ***
[1]    15016 abort (core dumped)  python examples/example.py --scene 

### Other notes

Issue #32 seemed to be facing a similar problem, however, I am not employing any port forwarding in my case and am testing this out on a local machine. I have not opened 2 separate issues since I feel they are related to each other however I may be wrong in assuming this. 

Any help would be greatly appreciated :)

erikwijmans, 2020-01-08:
Assuming your compute does have an attached display, then non headless mode should work.  What does `echo $DISPLAY` return?  It should almost certainly be set `:0`

wannabeOG, 2020-01-08:
Yeah, it does return ``:0``

erikwijmans, 2020-01-08:
Not sure what is going on then, you can try building with `--headless`.  The viewer won't work, but everything else should.

erikwijmans, 2020-01-08:
One other idea:  Maybe you are using Wayland not X11, what does `echo $XDG_SESSION_TYPE` show?

wannabeOG, 2020-01-09:
Huh this is pretty weird, it doesn't return anything. It just outputs a blank space. 
Running ``env | grep XDG_`` returns the following: 

XDG_SEAT=seat0
XDG_SESSION_ID=3
XDG_CURRENT_DESKTOP=KDE
XDG_RUNTIME_DIR=/run/user/1000
XDG_DATA_DIRS=/usr/share:/usr/share:/usr/local/share
XDG_VTNR=1

This does seem abnormal

wannabeOG, 2020-01-09:
A bit more context:

While booting into Ubuntu, I manually start the xorg server (``exec startx``). This was made necessary since I was running into an endless login loop when I first installed Ubuntu on my system.

mathfac, 2020-01-10:
@wannabeOG you can also check [a step by step installation on a fresh Ubuntu in our CI setup](https://github.com/facebookresearch/habitat-sim/blob/master/.circleci/config.yml#L62) and find missing pieces or repeat steps yourself. Ignore Javascript, emscripten, chrome and other unrelated dependencies.

closed 2020-08-09 by dhruvbatra