facebookresearch/habitat-sim #418: Interactive Python simulator script demo

aclegg3, 2020-01-08:
## 🚀 Feature

An interactive python demo of the simulator using python sim bindings and utilities directly, similar to viewer.cpp for C++ side.

## Motivation

Our [quickstart documentation](https://aihabitat.org/docs/habitat-api/quickstart.html) provides an example of a simple interactive Habitat demo in python using opencv to display observations and waiting for user input to command simulator actions.

This demo is purely single discrete actions (i.e. can't capture dynamics without a notion of time passing) and is not versioned with the codebase or mentioned outside of this docs page.

It would be nice to have a testbed for python functionality similar to viewer.cpp for C++ functionality.

## Pitch

We should develop and version an interactive python simulator script and accompanying tutorial documentation and link it in the [Testing section](https://github.com/facebookresearch/habitat-sim#testing) of the github page.

## Alternatives

We could bind the C++ viewer similar to our webGL approach, but this would preclude direct access to other bound simulator functions.

## Additional context

Additionally requested by others including #367

ronjamoller, 2020-01-09:
I am unsure if this is the place to add this information and if you will even develop the existing quickstart guide further instead of rebuilding from scratch but since i am currently working with it i might as well share: to get the quickstart script running one has to replace pointgoal with pointgoal_with_gps_compass and the habitat.SimulatorActions do not work, instead one can just insert MOVE_FORWARD, TURN_LEFT etc. although i'm sure this is just a hotfix and it is better practice to find the place where the actions are defined in the first place and access them. Again feel free to remove this information if it is not of interest.

aclegg3, 2020-01-09:
Thanks for sharing your fix, @ronjamoller. Should help others adapt the existing example until we get it properly incorporated into the codebase.

Skylion007, 2020-10-13:
#727 should be linked to this issue.

aclegg3, 2022-03-01:
This is addressed by the following PRs: #1538 #1519 #1631 #1551
The python viewer is functional and ready for use! 🎉

closed 2022-03-01 by aclegg3