facebookresearch/habitat-sim #420: How can i set the initial state of agent in viewer

picksh, 2020-01-09:
## ❓ Questions and Help
Hi, I am wondering how to set the initial state for agent when I use `./build/viewer`.

I tried to change code in `initialize_agent(self, agent_id, initial_state=None)`  in `simulator.py`:
`
        agent = self.get_agent(agent_id=agent_id)
        if initial_state is None:
            initial_state = AgentState()
        initial_state.position=np.ndarray=np.array([12.563317, 5.60849 ,-0.16552])
        initial_state.rotation=np.quaternion(0,0.0980171,0,-0.995185)
        initial_state.sensor_states = {}
        agent.set_state(initial_state)
        self._last_state = agent.state
        return agent
`
and rebuilt it, but it did not work. The initial position of agent did not change at all.

aclegg3, 2020-01-09:
./build/viewer runs the code in viewer.cpp and does not depend on any python functionality. You'll want to look into the C++ code to modify that application.

picksh, 2020-01-09:
Ahh I see. Thanks a lot！

closed 2020-01-09 by picksh