facebookresearch/habitat-sim #421: Make habitat-sim conda installable

mathfac, 2020-01-09:
## 🚀 Feature

## Motivation
That involve multiple steps to clone, build and install habitat-sim and is limiting adoption of the framework by community dramatically.

## Pitch
Produce and publish pre-built binaries of habitat-sim for Ubuntu and make it conda installable: `conda install habitat_sim -c habitat`
[image]
That will decrease framework usage threshold significantly and make our Continues Integration setup simpler. 

## Additional context
As reference point [PyTorch night builds CI system](https://github.com/pytorch/pytorch/tree/master/.circleci) can be checked and partially replicated.

dhruvbatra, 2020-01-10:
Yes! This will be a high-impact feature!

mathfac, 2020-01-13:
Reference point to previous @erikwijmans effort to make it done: [https://github.com/facebookresearch/habitat-sim/tree/conda-build](https://github.com/facebookresearch/habitat-sim/compare/conda-build?expand=1).

erikwijmans, 2020-01-15:
I took another looked a pytorch's conda builder to see what differences are there.  I noticed 2 things:

1.  They include the path to the conda env's lib folder in their rpath manipulations: https://github.com/pytorch/builder/blob/master/conda/pytorch-nightly/build.sh#L118
2. They do the build of the binaries in a docker container.  My guess is that their choice of base container and OS is what helps make the binaries compatible with other linux variants: https://github.com/pytorch/builder/blob/master/conda/Dockerfile.  We should try building the binaries in a centos7 cudagl container (https://hub.docker.com/r/nvidia/cudagl)

dhruvbatra, 2020-03-26:
@mpiseno -- now that #445 has landed, can we add instructions in the README on how to use the conda build?

shuishida, 2020-05-04:
@dhruvbatra Hello, I was trying to install Habitat sim onto my linux but for some reason it didn't build properly, so I was looking for an alternative solution when I came across this thread. I was wondering if you could point me to the conda-installable habitat sim solution. I tried

`conda install -c aihabitat -c conda-forge habitat-sim`

which I found in the README, but it produced the following errors:

```Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: \ 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                                                      

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Package xorg-libx11 conflicts for:
habitat-sim -> xorg-libx11
Package bzip2 conflicts for:
python=3.6 -> bzip2[version='>=1.0.6,<2.0a0']
Package ld_impl_linux-64 conflicts for:
python=3.6 -> ld_impl_linux-64
Package python_abi conflicts for:
python=3.6 -> python_abi[version='*|3.6',build='*_pypy36_pp73|*_cp36m']
wheel -> python_abi=[build='*_cp38|*_cp36m|*_cp35m|*_cp27mu|*_cp34m|*_cp37m']
pip -> python_abi=[build='*_cp38|*_cp36m|*_cp35m|*_cp27mu|*_cp34m|*_cp37m']
certifi -> python_abi[version='2.7.*|3.6.*|3.7.*|3.8.*|3.6',build='*_pypy36_pp73|*_cp37m|*_cp36m|*_cp27mu|*_cp38']
habitat-sim -> python_abi=3.6[build=*_cp36m]
setuptools -> python_abi[version='3.6|3.6.*|3.7.*|3.8.*',build='*_pypy36_pp73|*_cp38|*_cp37m|*_cp36m']
Package ncurses conflicts for:
readline -> ncurses[version='5.9.*|6.0.*|>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.1,<7.0a0']
sqlite -> ncurses[version='5.9|5.9.*|>=6.1,<6.2.0a0|>=6.2,<7.0a0']
libedit -> ncurses[version='5.9|6.0.*|>=6.1,<6.2.0a0|>=6.1,<7.0a0']
python=3.6 -> ncurses[version='5.9|5.9.*|6.0.*|>=6.0,<7.0a0|>=6.1,<6.2.0a0|>=6.1,<7.0a0|>=6.2,<7.0a0']
Package libstdcxx-ng conflicts for:
python=3.6 -> libstdcxx-ng[version='>=4.9|>=7.2.0|>=7.3.0']
libffi -> libstdcxx-ng[version='>=4.9|>=7.2.0|>=7.3.0']
sqlite -> libstdcxx-ng[version='>=4.9|>=7.3.0']
ncurses -> libstdcxx-ng[version='>=4.9|>=7.2.0|>=7.3.0']
Package scipy conflicts for:
habitat-sim -> scipy[version='>=1.3.0']
Package libffi conflicts for:
python=3.6 -> libffi[version='3.2.*|>=3.2.1,<3.3.0a0|>=3.2.1,<3.3a0']
Package quaternion conflicts for:
habitat-sim -> quaternion
Package attrs conflicts for:
habitat-sim -> attrs[version='>=19.1.0']
Package openssl conflicts for:
python=3.6 -> openssl[version='1.0.*|1.0.*,>=1.0.2l,<1.0.3a|>=1.0.2m,<1.0.3a|>=1.0.2n,<1.0.3a|>=1.0.2o,<1.0.3a|>=1.0.2p,<1.0.3a|>=1.1.1a,<1.1.2a|>=1.1.1c,<1.1.2a|>=1.1.1d,<1.1.2a|>=1.1.1e,<1.1.2a|>=1.1.1f,<1.1.2a|>=1.1.1g,<1.1.2a']
Package libgcc-ng conflicts for:
libedit -> libgcc-ng[version='>=4.9|>=7.2.0|>=7.3.0']
readline -> libgcc-ng[version='>=4.9|>=7.2.0|>=7.3.0']
tk -> libgcc-ng[version='>=4.9|>=7.2.0|>=7.3.0']
sqlite -> libgcc-ng[version='>=4.9|>=7.2.0|>=7.3.0']
openssl -> libgcc-ng[version='>=4.9|>=7.2.0|>=7.3.0']
zlib -> libgcc-ng[version='>=4.9|>=7.2.0|>=7.3.0']
xz -> libgcc-ng[version='>=4.9|>=7.2.0|>=7.3.0']
libffi -> libgcc-ng[version='>=4.9|>=7.2.0|>=7.3.0']
ncurses -> libgcc-ng[version='>=4.9|>=7.2.0|>=7.3.0']
python=3.6 -> libgcc-ng[version='>=4.9|>=7.2.0|>=7.3.0']
Package xorg-libxcursor conflicts for:
habitat-sim -> xorg-libxcursor
Package zlib conflicts for:
python=3.6 -> zlib[version='1.2.*|1.2.11|1.2.8|>=1.2.11,<1.3.0a0']
tk -> zlib[version='>=1.2.11,<1.3.0a0']
sqlite -> zlib[version='>=1.2.11,<1.3.0a0']
Package setuptools conflicts for:
pip -> setuptools
[... 30 lines cut ...]
habitat-sim -> pillow
Package pypy3.6 conflicts for:
python=3.6 -> pypy3.6[version='7.3.0.*|7.3.1.*']
Package readline conflicts for:
python=3.6 -> readline[version='6.2.*|7.*|7.0|>=7.0,<8.0a0|>=8.0,<9.0a0']
sqlite -> readline[version='6.2.*|7.0.*|7.0|>=7.0,<8.0a0|>=8.0,<9.0a0']
Package xorg-libxinerama conflicts for:
habitat-sim -> xorg-libxinerama[version='>=1.1.4,<1.2.0a0']
Package pip conflicts for:
python=3.6 -> pip
Package xorg-libxau conflicts for:
habitat-sim -> xorg-libxau
Package numpy conflicts for:
habitat-sim -> numpy[version='>=1.13']
Package xorg-libxfixes conflicts for:
habitat-sim -> xorg-libxfixes
Package ca-certificates conflicts for:
openssl -> ca-certificates
Package libedit conflicts for:
sqlite -> libedit[version='>=3.1.20170329,<3.2.0a0|>=3.1.20181209,<3.2.0a0']
Package tqdm conflicts for:
habitat-sim -> tqdm
Package libgomp conflicts for:
libgcc-ng -> libgomp[version='7.3.0|9.2.0',build='h24d8f2e_4|h24d8f2e_5|h24d8f2e_2']
```

mpiseno, 2020-05-04:
@shuishida Does your system have a display? The conda binaries for linux currently are for headless systems.

Does this work?
`conda install -c aihabitat -c conda-forge habitat-sim headless`

shuishida, 2020-05-04:
@mpiseno Ah thank you so much! Yes I have a display, but the headless version seems to work!! This is beautiful!

Skylion007, 2020-10-13:
Now closed.

closed 2020-10-13 by Skylion007

KenaHemnani, 2020-12-13:
Hii, Can we install specific commit for habitat-sim via. conda command  **conda install -c aihabitat -c conda-forge habitat-sim headless**?

erikwijmans, 2020-12-13:
No.  We only have pre-compiled binaries for tagged versions or latest master through the nightly channel.

KenaHemnani, 2020-12-15:
Thank you very much for the response, Sir. I really appreciate your help. Is there any way to install older version on google colab?

erikwijmans, 2020-12-15:
You can build from source on collab.  @Skylion007 @aclegg3 IIRC, one of you had scripts and/or a collab notebook that did at some point.  If that is correct, does it still exist?

Skylion007, 2020-12-15:
This cell will install habitat-sim:
```%cd /content
!wget -qO - https://apt.kitware.com/keys/kitware-archive-latest.asc | sudo apt-key add -
!apt-add-repository 'deb https://apt.kitware.com/ubuntu/ bionic main' #Add latest CMake version
# clang-9 llvm-9 lld-9
!apt update && apt install -y --no-install-recommends  libjpeg-dev libglm-dev libgl1-mesa-glx libegl1-mesa-dev mesa-utils  libbullet-dev cmake libomp-dev ninja-build libassimp-dev libeigen3-dev clang-9 lld-9  libclang-9-dev llvm-9-dev
#libxrandr-dev libxinerama-dev
# The default in Colab CMake 3.12 has a bug that prevents Habitat-Sim from compiling properly with CUDA
!rm -f /usr/local/bin/cmake #Delete Old CMake

%env HEADLESS=True
%cd /content/

!git clone https://github.com/facebookresearch/habitat-sim --depth 1 --recursive --jobs 16
%cd '/content/habitat-sim'
#CLANG'S LINKER IS A LOT FASTER WHICH SAVES A LOT OF TIME IN COLAB
%env CC=/usr/bin/clang-9
%env CXX=/usr/bin/clang++-9
%env LD=/usr/bin/lld-9
#We use a system wide precompiled version of ASSIMP to save compile time.
!python setup.py install --headless --bullet --cmake-args='-DUSE_SYSTEM_ASSIMP=ON'
```

KenaHemnani, 2020-12-15:
Thank you for your quick response!! @erikwijmans and @Skylion007 . Habitat is really providing awesome platform for researcher working in embodied-AI.

Hey, I installed habitat-sim as given in above code and habitat-lab as given in https://github.com/facebookresearch/habitat-lab

And running following:
**!python examples/example.py**
But got following error:
`2020-12-15 18:46:27,029 Initializing dataset PointNav-v1
2020-12-15 18:46:27,298 initializing sim Sim-v0
Traceback (most recent call last):
  File "examples/example.py", line 28, in <module>
    example()
  File "examples/example.py", line 14, in example
    config=habitat.get_config("configs/tasks/pointnav.yaml")
  File "/content/habitat-lab/habitat/core/env.py", line 105, in __init__
    id_sim=self._config.SIMULATOR.TYPE, config=self._config.SIMULATOR
  File "/content/habitat-lab/habitat/sims/registration.py", line 19, in make_sim
    return _sim(**kwargs)
  File "/content/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 182, in __init__
    self.sim_config = self.create_sim_config(self._sensor_suite)
  File "/content/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 198, in create_sim_config
    sim_config.scene.id = self.habitat_config.SCENE
AttributeError: 'habitat_sim._ext.habitat_sim_bindings.SimulatorCon' object has no attribute 'scene'
Exception ignored in: <bound method Simulator.__del__ of HabitatSim(config=NOTHING, agents=NOTHING, _num_total_frames=NOTHING, _default_agent_id=NOTHING, _Simulator__sensors=NOTHING, _initialized=NOTHING, _previous_step_time=NOTHING, _Simulator__last_state=NOTHING)>
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/simulator.py", line 467, in __del__
    self.close()
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.6-py3.6-linux-x86_64.egg/habitat_sim/simulator.py", line 118, in close
    for agent_sensorsuite in self.__sensors:
AttributeError: 'HabitatSim' object has no attribute '_Simulator__sensors'`

Is there anything I am missing?? Thank you in advance.

erikwijmans, 2020-12-15:
You will also need to be on a commit of habitat-lab that is compatible with that version of the simulator.  You need to be on a newer one it looks like.  I believe current master in habitat-lab is fine.

KenaHemnani, 2020-12-15:
> You will also need to be on a commit of habitat-lab that is compatible with that version of the simulator. You need to be on a newer one it looks like. I believe current master in habitat-lab is fine.

Hii Thanks for the support. I used current version of habitat-lab only.

Also, tried compatible git checkouts taking this as reference:https://github.com/devendrachaplot/Neural-SLAM

Then got following error:

2020-12-15 20:03:33,218 initializing sim Sim-v0
Traceback (most recent call last):
  File "examples/example.py", line 25, in <module>
    example()
  File "examples/example.py", line 11, in example
    env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))
  File "/content/habitat-api/habitat/core/env.py", line 93, in __init__
    id_sim=self._config.SIMULATOR.TYPE, config=self._config.SIMULATOR
  File "/content/habitat-api/habitat/sims/registration.py", line 19, in make_sim
    return _sim(**kwargs)
  File "/content/habitat-api/habitat/sims/habitat_simulator/habitat_simulator.py", line 158, in __init__
    self.sim_config = self.create_sim_config(self._sensor_suite)
  File "/content/habitat-api/habitat/sims/habitat_simulator/habitat_simulator.py", line 171, in create_sim_config
    sim_config.scene.id = self.config.SCENE
AttributeError: 'habitat_sim._ext.habitat_sim_bindings.SimulatorCon' object has no attribute 'scene'

Am I doing any mistake?  Is the code is only for some perticular habitat-sim git checkout??

erikwijmans, 2020-12-15:
Yes, that commit of habitat-lab requires an older commit of habitat-sim.  If you absolutely must use those two commits, this PR is what is missing from your habitat-lab version: https://github.com/facebookresearch/habitat-lab/pull/516

KenaHemnani, 2020-12-16:
Hii thank you for this suggestion. I changed the code manually in to scene_id from scene.id where the code was breaking.

Now it is showing me following error:
2020-12-16 05:52:13,430 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1216 05:52:13.433264  7440 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : default changed to proposed JSON configuration filename : default.scene_dataset_config.json
I1216 05:52:13.433437  7440 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I1216 05:52:13.433506  7440 AssetAttributesManager.cpp:117] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I1216 05:52:13.433570  7440 AssetAttributesManager.cpp:117] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I1216 05:52:13.433611  7440 AssetAttributesManager.cpp:117] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I1216 05:52:13.433634  7440 AssetAttributesManager.cpp:117] Asset attributes (cubeSolid : cubeSolid) created and registered.
I1216 05:52:13.433653  7440 AssetAttributesManager.cpp:117] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I1216 05:52:13.433706  7440 AssetAttributesManager.cpp:117] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I1216 05:52:13.433756  7440 AssetAttributesManager.cpp:117] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I1216 05:52:13.433782  7440 AssetAttributesManager.cpp:117] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I1216 05:52:13.433805  7440 AssetAttributesManager.cpp:117] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I1216 05:52:13.433838  7440 AssetAttributesManager.cpp:117] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I1216 05:52:13.433873  7440 AssetAttributesManager.cpp:117] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I1216 05:52:13.433889  7440 AssetAttributesManager.cpp:105] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I1216 05:52:13.434511  7440 SceneDatasetAttributesManager.cpp:23] File (default) not found, so new default dataset attributes created and registered.
I1216 05:52:13.434525  7440 MetadataMediator.cpp:47] MetadataMediator::createDataset : Dataset default successfully created.
I1216 05:52:13.438854  7440 PhysicsAttributesManager.cpp:27] File (./data/default.physics_config.json) not found, so new default physics manager attributes created and registered.
I1216 05:52:13.438887  7440 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : data/scene_datasets/habitat-test-scenes/skokloster-castle.glb changed to proposed JSON configuration filename : data/scene_datasets/habitat-test-scenes/skokloster-castle.stage_config.json
I1216 05:52:13.438977  7440 AbstractObjectAttributesManagerBase.h:180] File (data/scene_datasets/habitat-test-scenes/skokloster-castle.glb) exists but is not a recognized config filename extension, so new default Stage attributes created and registered.
I1216 05:52:13.439000  7440 Simulator.cpp:153] Loading navmesh from data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
I1216 05:52:13.439163  7440 Simulator.cpp:155] Loaded.
I1216 05:52:13.439195  7440 SceneGraph.h:93] Created DrawableGroup: 
Platform::WindowlessEglApplication::tryCreateContext(): no EGL devices found, likely a driver issue; enable --magnum-gpu-validation to see additional info
WindowlessContext: Unable to create windowless context

How to solve this driver issue? Thank you in advance!!

erikwijmans, 2020-12-16:
If this is still on colab, you will need a GPU colab instance to run habitat-sim.

KenaHemnani, 2020-12-17:
Thank you very much @erikwijmans , it is working now.

ML4Life, 2021-05-20:
Hi, 
I tried to install the habitat-sim using all the methods as mentioned in the respected website and even in this thread but nothing works for me. Either no response or package error as noted by @shuishida. Please guide me in resolving this issue. 
Thanks

ML4Life, 2021-05-20:
Hi, 
I am having the issue of **WindowlessContext: Unable to create windowless context** (the same as mentioned in[ link](https://github.com/facebookresearch/habitat-sim/issues/421#issuecomment-745789814). I want to use this simulator on my server PC which enables GPU in it. Please try to resolve this. 

Thanks in advance.

dhruvbatra, 2021-05-20:
Please open a fresh issue, answering the questions in the template.