facebookresearch/habitat-sim #426: The error ''No such file or directory: 'build/compile_commands.json' "

liuxinzhu0353150307, 2020-01-14:
## ❓ Questions and Help

I use the command   python setup.py build_ext --parallel 1 install  to build he habitat-sim in my virture  machine Ubuntu18.04， but when I built it I get the following error:

running build_ext
error: [Errno 2] No such file or directory: 'build/compile_commands.json'

The file compile_commands.json does not exist surely, so how can I solve the problem?

bigbike, 2020-01-14:
`rm -rf build/`
run `python setup.py install` again.

liuxinzhu0353150307, 2020-01-14:
> `rm -rf build/`
> run `python setup.py install` again.

Thanks a lot, I have solved the error.

closed 2020-01-14 by mathfac

201814476, 2021-10-10:
> > `rm -rf build/`
> > run `python setup.py install` again.
> 
> Thanks a lot, I have solved the error.

Can you tell me how to do that? Thanks a lot!