facebookresearch/habitat-sim #427: The error "GLSL 4.10 is not supported."

liuxinzhu0353150307, 2020-01-14:
## ❓ Questions and Help

When I run the habitat-test code in my virtual machine ubuntu18.04, I get the following error.

When I run  ./build/viewer /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb, the viewer windows cannot display the scene, the display window is all black.

Then when I run non-interactive testing using command  python examples/example.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb, I get the error:
"GLSL 4.10 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.50, 3.30, 1.00 ES, and 3.00 ES".

I build the environment in the virtual machine, and I search the opengl version is "OpenGL version: 3.3 (Core Profile) Mesa 19.0.2".

So I want to know how to solve the problem? Is it because the version of opengl is not compatible?

erikwijmans, 2020-01-15:
Yes, Open GL 3.3 is not compatible/supported.

closed 2020-01-15 by erikwijmans

mhaboali, 2020-02-24:
Hi @erikwijmans 
I'm facing the same error, do you know how to avoid this error?

When I ran `glxinfo | grep "OpenGL version"` I got -> OpenGL version string: 3.1 Mesa 19.2.8 but when I ran python examples/example.py I got the same error for GLSL 4.10 is not supported.

I tried to use `MESA_GL_VERSION_OVERRIDE=3.1 python examples/example.py` but I got the same error.

Any help?

erikwijmans, 2020-02-24:
I don't have a linux box with software rendering to test, but may it's suppose to `MESA_GL_VERSION_OVERRIDE=4.1`?

mosra, 2020-02-24:
The GL driver commonly used in virtual machines (SVGA3D) is stuck on GL 3.1 /  3.3 and AFAIK that's not going to change in the foreseeable future, I'm afraid. So there's really no way around this except for not using a VM.

mhaboali, 2020-02-25:
@mosra 
Actually I'm trying to install it on my local machine(Ubuntu 18.04) with GTX 1080 Ti but I didn't install the Nvidia driver yet.

When I tried on a remote VM with --headless it worked properly for me.

mosra, 2020-02-25:
It should work with the binary NVidia driver. I have to admit I never tried with Mesa/Nouveau myself, so I have no idea about the quirks there (though I'm running on Mesa daily on both Intel and AMD GPUs).

mhaboali, 2020-02-25:
I'm using now the built-in Graphics card(llvmpipe (LLVM 9.0, 256 bits)), so should I install the NVidia driver or it should work normally with the built-in one?

mosra, 2020-02-25:
`llvmpipe` is a software driver, which is slow and doesn't make use of your beefy 1080 Ti at all :) Yes, install the NVidia driver.

mhaboali, 2020-02-25:
Okay, I'll do and will come back if I face further issues :)

Thanks for your help, I appreciate your time!